introlab/rtabmap_ros #851: Problems configuring stereo odometry with rtabmap in ROS Galactic.

brayanpa, 2022-12-01:
Hi @matlabbe, hope all is going well.

I am doing some tests with rtabmap and visual odometry in ROS Galactic. For this I am using a D435i camera.

I have not achieved a good odometry performance. Using RGBD odometry (with camera depth) I can get the odom-base_footprint transform but it is easily lost (quality=0).
For this reason I opted to use the stereo odometry with the infrared cameras of the D435. In this way the odometry is not easily lost, but the odom-base_footprint transform is practically static.

If you could guide me I would really appreciate it. These are the parameters that I am using for visual odometry:
```
"qos": 2,
"publish_tf": True,
"wait_imu_to_init": True,
"approx_sync": True,
"Stereo/Eps": '0.01',
"Stereo/Iterations": '30',
"Stereo/MaxDisparity": '512',
"Stereo/MaxLevel": '3',
"Stereo/MinDisparity": '1',
"Stereo/OpticalFlow": 'False',
"Stereo/SSD": 'True',
"Stereo/WinHeigh": '3',
"Stereo/WinWidth": '15',
'frame_id': 'base_footprint',
'odom_frame_id': 'odom',
'map_frame_id': 'map', 
'use_sim_time': False,
```

If I set the Stereo/OpticalFlow parameter to True, I never get a quality greater than 0.

matlabbe, 2022-12-01:
Did you try this launch? https://github.com/introlab/rtabmap_ros/blob/ros2/launch/ros2/realsense_d435i_infra.launch.py

brayanpa, 2022-12-13:
I'm testing it and it's going well, it seems that the problem was that the emitter and the alignment of the camera made it too slow for the odometry. But now I am having an error in rtabmap the odometry in rtabmapviz is not lost but rtabmap dies after sometime:

```
[rtabmap-2] [INFO] [1670946616.319927787] [rtabmap]: rtabmap (63): Rate=1.00s, Limit=0.000s, Conversion=0.0026s, RTAB-Map=0.2553s, Maps update=0.0199s pub=0.0000s (local map=44, WM=44)

[rtabmap-2] [INFO] [1670946617.714599585] [rtabmap]: rtabmap (64): Rate=1.00s, Limit=0.000s, Conversion=0.0021s, RTAB-Map=0.4732s, Maps update=0.1309s pub=0.0003s (local map=45, WM=45)

[rtabmap-2] [INFO] [1670946618.735201700] [rtabmap]: rtabmap (65): Rate=1.00s, Limit=0.000s, Conversion=0.0027s, RTAB-Map=0.3883s, Maps update=0.0859s pub=0.0003s (local map=46, WM=46)

[rtabmap-2] terminate called after throwing an instance of 'std::out_of_range'

[rtabmap-2] what(): map::at

[ERROR] [rtabmap-2]: process has died [pid 734493, exit code -6, cmd '/root/deps_ws/install/rtabmap_ros/lib/rtabmap_ros/rtabmap --ros-args -r __ns:=/ --params-file /tmp/launch_params_cebp69at --params-file /tmp/launch_params_lq75h926 -r rgb/image:=/gary_camera_nav/color/image_raw -r depth/image:=/gary_camera_nav/aligned_depth_to_color/image_raw -r rgb/camera_info:=/gary_camera_nav/color/camera_info -r rgbd_image:=/rgbd_image0 -r left/image_rect:=/stereo_camera/left/image_rect_color -r right/image_rect:=/stereo_camera/right/image_rect -r left/camera_info:=/stereo_camera/left/camera_info -r right/camera_info:=/stereo_camera/right/camera_info -r scan:=/gary/rplidar/scan -r scan_cloud:=/scan_cloud -r user_data:=/user_data -r user_data_async:=/user_data_async -r gps/fix:=/gps/fix -r tag_detections:=/tag_detections -r odom:=/odom -r imu:=/imu/data'].
```