introlab/rtabmap_ros #855: Error in compiling rtabmap_ros: "undefined reference to `rtabmap::util3d::[...]"

mjpc13, 2022-12-15:
Hi, I am trying to compile rtabmap_ros using `catkin_tools` but it is giving me the following error:

```
Errors     << rtabmap_ros:make /work/Documents/catkin_ws/logs/rtabmap_ros/build.make.004.log                                               
/usr/bin/ld: warning: libopencv_core.so.4.2, needed by /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so, may conflict with libopencv_core.so.406
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::concatenate(boost::shared_ptr<std::vector<int, std::allocator<int> > > const&, boost::shared_ptr<std::vector<int, std::allocator<int> > > const&)'
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::transformPointCloud(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, rtabmap::Transform const&)'
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::voxelize(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, float)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rtabmap_wifi_signal_pub.dir/build.make:549: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/rtabmap_ros/wifi_signal_pub] Error 1
make[1]: *** [CMakeFiles/Makefile2:1772: CMakeFiles/rtabmap_wifi_signal_pub.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: warning: libopencv_core.so.4.2, needed by /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so, may conflict with libopencv_core.so.406
/usr/bin/ld: warning: libopencv_imgproc.so.4.2, needed by /opt/ros/noetic/lib/libcv_bridge.so, may conflict with libopencv_imgproc.so.406
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::concatenate(boost::shared_ptr<std::vector<int, std::allocator<int> > > const&, boost::shared_ptr<std::vector<int, std::allocator<int> > > const&)'
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::transformPointCloud(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, rtabmap::Transform const&)'
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::voxelize(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, float)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rtabmap_stereo_camera.dir/build.make:549: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/rtabmap_ros/stereo_camera] Error 1
make[1]: *** [CMakeFiles/Makefile2:492: CMakeFiles/rtabmap_stereo_camera.dir/all] Error 2
/usr/bin/ld: warning: libopencv_core.so.4.2, needed by /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so, may conflict with libopencv_core.so.406
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::concatenate(boost::shared_ptr<std::vector<int, std::allocator<int> > > const&, boost::shared_ptr<std::vector<int, std::allocator<int> > > const&)'
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::transformPointCloud(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, rtabmap::Transform const&)'
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::voxelize(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, float)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rtabmap_odom_msg_to_tf.dir/build.make:549: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/rtabmap_ros/odom_msg_to_tf] Error 1
make[1]: *** [CMakeFiles/Makefile2:2921: CMakeFiles/rtabmap_odom_msg_to_tf.dir/all] Error 2
/usr/bin/ld: warning: libopencv_core.so.4.2, needed by /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so, may conflict with libopencv_core.so.406
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::concatenate(boost::shared_ptr<std::vector<int, std::allocator<int> > > const&, boost::shared_ptr<std::vector<int, std::allocator<int> > > const&)'
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::transformPointCloud(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, rtabmap::Transform const&)'
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::voxelize(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, float)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rtabmap_map_assembler.dir/build.make:549: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/rtabmap_ros/map_assembler] Error 1
make[1]: *** [CMakeFiles/Makefile2:605: CMakeFiles/rtabmap_map_assembler.dir/all] Error 2
/usr/bin/ld: warning: libopencv_core.so.4.2, needed by /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so, may conflict with libopencv_core.so.406
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::concatenate(boost::shared_ptr<std::vector<int, std::allocator<int> > > const&, boost::shared_ptr<std::vector<int, std::allocator<int> > > const&)'
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::transformPointCloud(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, rtabmap::Transform const&)'
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::voxelize(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, float)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rtabmap_data_player.dir/build.make:549: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/rtabmap_ros/data_player] Error 1
make[1]: *** [CMakeFiles/Makefile2:740: CMakeFiles/rtabmap_data_player.dir/all] Error 2
/usr/bin/ld: warning: libopencv_core.so.4.2, needed by /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so, may conflict with libopencv_core.so.406
/usr/bin/ld: CMakeFiles/rtabmap_wifi_signal_sub.dir/src/WifiSignalSubNode.cpp.o: in function `mapDataCallback(boost::shared_ptr<rtabmap_ros::MapData_<std::allocator<void> > const> const&)':
WifiSignalSubNode.cpp:(.text+0x15bb): undefined reference to `rtabmap::util3d::transformPointCloud(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, rtabmap::Transform const&)'
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::concatenate(boost::shared_ptr<std::vector<int, std::allocator<int> > > const&, boost::shared_ptr<std::vector<int, std::allocator<int> > > const&)'
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::voxelize(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, float)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rtabmap_wifi_signal_sub.dir/build.make:549: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/rtabmap_ros/wifi_signal_sub] Error 1
make[1]: *** [CMakeFiles/Makefile2:4498: CMakeFiles/rtabmap_wifi_signal_sub.dir/all] Error 2
/usr/bin/ld: warning: libopencv_core.so.4.2, needed by /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so, may conflict with libopencv_core.so.406
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::concatenate(boost::shared_ptr<std::vector<int, std::allocator<int> > > const&, boost::shared_ptr<std::vector<int, std::allocator<int> > > const&)'
/usr/bin/ld: /work/Documents/catkin_ws/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::transformPointCloud(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, rtabmap::Transform const&)'
[... cut ...]
With libpointmatcher:     true
With CCCoreLib:          false
With octomap:             true
With cpu-tsdf:           false
With open chisel:        false
With Alice Vision:       false
With LOAM:               false
With FLOAM:              false
With FOVIS:              false
With Viso2:              false
With DVO:                false
With ORB_SLAM:           false
With OKVIS:              false
With MSCKF_VIO:          false
With VINS-Fusion:        false
With OpenVINS:           false
```
Can you help me?

matlabbe, 2022-12-16:
On Focal/Noetic, if you don't want conflicts with opencv binaries (4.2.0) used by `cv_bridge` (thus `rtabmap_ros`), build/install OpenCV 4.2.0 from source (with xfeatures2d, nonfree modules...). If you really need OpenCV 4.6.0, uninstall opencv 4.2.0 binaries from your system. You will have to rebuild every packages depending on opencv 4.2.0... which can be painful.

For PCL, maybe it is the version. Based on [CI](https://github.com/introlab/rtabmap_ros/actions), rtabmap can build with PCL 1.12.1 ([Ubuntu 22.04](https://github.com/introlab/rtabmap_ros/actions/runs/3671149447/jobs/6206152989)), PCL 1.10.0 ([Ubuntu 20.04](https://github.com/introlab/rtabmap_ros/actions/runs/3701200323/jobs/6270303439)) and PCL 1.8.1 ([Ubuntu 18.04](https://github.com/introlab/rtabmap_ros/actions/runs/3701200323/jobs/6270303439)). For the in-between PCL releases, I cannot really test all of them, I generarly try to support the latest and the official binaries of each LTS ubuntu distro.

an99990, 2023-04-11:
currently encoutering this issue with a clean make

```
/usr/bin/ld: /home/huynh-anh/catkin_ws/devel/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::transformPointCloud(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, rtabmap::Transform const&)'
/usr/bin/ld: /home/huynh-anh/catkin_ws/devel/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::voxelize(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, float)'
collect2: error: ld returned 1 exit status
make[2]: *** [rtabmap_ros-0.20.23-noetic/CMakeFiles/rtabmap_map_optimizer.dir/build.make:413: /home/huynh-anh/catkin_ws/devel/lib/rtabmap_ros/map_optimizer] Error 1
make[1]: *** [CMakeFiles/Makefile2:8373: rtabmap_ros-0.20.23-noetic/CMakeFiles/rtabmap_map_optimizer.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 83%] Building CXX object rtabmap_ros-0.20.23-noetic/CMakeFiles/rtabmap_sync.dir/src/impl/CommonDataSubscriberStereo.cpp.o
[ 83%] Linking CXX executable /home/huynh-anh/catkin_ws/devel/lib/rtabmap_ros/map_assembler
/usr/bin/ld: /home/huynh-anh/catkin_ws/devel/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::transformPointCloud(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, rtabmap::Transform const&)'
/usr/bin/ld: /home/huynh-anh/catkin_ws/devel/lib/librtabmap_ros.so: undefined reference to `rtabmap::util3d::voxelize(boost::shared_ptr<pcl::PointCloud<pcl::PointXYZRGB> > const&, float)'
collect2: error: ld returned 1 exit status
make[2]: *** [rtabmap_ros-0.20.23-noetic/CMakeFiles/rtabmap_map_assembler.dir/build.make:413: /home/huynh-anh/catkin_ws/devel/lib/rtabmap_ros/map_assembler] Error 1
make[1]: *** [CMakeFiles/Makefile2:8402: rtabmap_ros-0.20.23-noetic/CMakeFiles/rtabmap_map_assembler.dir/all] Error 2

```

matlabbe, 2023-04-17:
Just tried with latest PCL code, and it should build:
```
$ cd rtabmap
$ docker build -t rtabmap_latest -f docker/latest_deps/Dockerfile .

[+] Building 1728.9s (15/15) FINISHED                                                                                          
 => [internal] load build definition from Dockerfile                                                                      0.0s
 => => transferring dockerfile: 2.04kB                                                                                    0.0s
 => [internal] load .dockerignore                                                                                         0.0s
 => => transferring context: 48B                                                                                          0.0s
 => [internal] load metadata for docker.io/osrf/ros:humble-desktop                                                        1.3s
 => [auth] osrf/ros:pull token for registry-1.docker.io                                                                   0.0s
 => [internal] load build context                                                                                         0.1s
 => => transferring context: 4.65MB                                                                                       0.1s
 => [1/9] FROM docker.io/osrf/ros:humble-desktop@sha256:a0783dfdd456d938049f9da0e7e1c394a6b2985add72ffb76e31992ae211d1  123.8s
 => => resolve docker.io/osrf/ros:humble-desktop@sha256:a0783dfdd456d938049f9da0e7e1c394a6b2985add72ffb76e31992ae211d123  0.0s
 => => sha256:53526f807b310160f31f38198712da92a6786e162fdf80506ab7bb864ad5dec3 6.13kB / 6.13kB                            0.0s
 => => sha256:74ac377868f863e123f24c409f79709f7563fa464557c36a09cf6f85c8b92b7f 30.43MB / 30.43MB                         15.6s
 => => sha256:41ad1f8f26857dba14fd27f7d8af8cd5f9ce4c25d5e4264090e4e33665838f4a 1.17MB / 1.17MB                            0.8s
 => => sha256:a0783dfdd456d938049f9da0e7e1c394a6b2985add72ffb76e31992ae211d123 2.84kB / 2.84kB                            0.0s
 => => sha256:4af12f0c8d10d787a739ce0beb1339ad34542095ac563892b59872c9b9229991 3.83MB / 3.83MB                            1.4s
 => => sha256:ab4f122dfff7592e93555f020a3ba95ce2e1d60b0b138d39e444d80b29e1fdce 230B / 230B                                1.0s
 => => sha256:55c195d49a74ea3cc9e44dd424ab6c5a62643eb9cc1fd60c335335d24eaee3fd 1.99kB / 1.99kB                            1.3s
 => => sha256:a62e2d9c70cd1fb5cfb879150c7d04b63ab2922e154e11b0fd47031a52c16a5a 106.34MB / 106.34MB                       32.7s
 => => sha256:273cc0c60a3141d6e964d07b9ac39f9126b3d3ca36a70493cc8496448375fd55 196B / 196B                                1.5s
 => => sha256:3a0800e2c533d39c10eda1c33c4cc5e7d8c164ae5d620b68a5674ec6ff18427d 97.89MB / 97.89MB                         15.2s
 => => sha256:130e7a70d397d234ae4137584d1d00a5ddf4d23b258ecf856deeefab1fe664f5 314.07kB / 314.07kB                       15.3s
 => => sha256:e4a79070b854f777682eb1cc3a62e2cf75029fdd5105fe7ed8e85a993ea5226c 2.40kB / 2.40kB                           15.5s
 => => sha256:84cfb85e0aafffb281729a9c1323877d16e594d6ccd10b8d798b53c0925247c4 23.07MB / 23.07MB                         20.5s
 => => sha256:53391740e69a4ddea057f37cb45328267ce1528da4571ce86314239f719db2bb 948.24MB / 948.24MB                      113.4s
 => => extracting sha256:74ac377868f863e123f24c409f79709f7563fa464557c36a09cf6f85c8b92b7f                                 0.3s
 => => extracting sha256:41ad1f8f26857dba14fd27f7d8af8cd5f9ce4c25d5e4264090e4e33665838f4a                                 0.1s
 => => extracting sha256:4af12f0c8d10d787a739ce0beb1339ad34542095ac563892b59872c9b9229991                                 0.0s
 => => extracting sha256:ab4f122dfff7592e93555f020a3ba95ce2e1d60b0b138d39e444d80b29e1fdce                                 0.0s
 => => extracting sha256:55c195d49a74ea3cc9e44dd424ab6c5a62643eb9cc1fd60c335335d24eaee3fd                                 0.0s
 => => extracting sha256:a62e2d9c70cd1fb5cfb879150c7d04b63ab2922e154e11b0fd47031a52c16a5a                                 1.5s
 => => extracting sha256:273cc0c60a3141d6e964d07b9ac39f9126b3d3ca36a70493cc8496448375fd55                                 0.0s
 => => extracting sha256:3a0800e2c533d39c10eda1c33c4cc5e7d8c164ae5d620b68a5674ec6ff18427d                                 0.9s
 => => extracting sha256:130e7a70d397d234ae4137584d1d00a5ddf4d23b258ecf856deeefab1fe664f5                                 0.0s
 => => extracting sha256:e4a79070b854f777682eb1cc3a62e2cf75029fdd5105fe7ed8e85a993ea5226c                                 0.0s
 => => extracting sha256:84cfb85e0aafffb281729a9c1323877d16e594d6ccd10b8d798b53c0925247c4                                 0.3s
 => => extracting sha256:53391740e69a4ddea057f37cb45328267ce1528da4571ce86314239f719db2bb                                 9.7s
 => [2/9] RUN apt-get update &&     apt-get install -y git software-properties-common ros-humble-rtabmap-ros libqt6* q  111.1s
 => [3/9] WORKDIR /root/                                                                                                  0.1s
 => [4/9] RUN rm /bin/sh && ln -s /bin/bash /bin/sh                                                                       0.4s
 => [5/9] RUN git clone https://github.com/Kitware/VTK.git &&     cd VTK &&     mkdir build &&     cd build &&     cma  430.2s
 => [6/9] RUN git clone https://github.com/PointCloudLibrary/pcl.git &&     cd pcl &&     mkdir build &&     cd build   583.7s
[... cut ...]
 => [9/9] RUN source /ros_entrypoint.sh &&     cd rtabmap/build &&     cmake .. &&     make -j$(nproc) &&     make ins  157.0s
 => exporting to image                                                                                                    5.9s
 => => exporting layers                                                                                                   5.9s
 => => writing image sha256:09338bca635fd8b17064dbde8c7f64f5b74bca4f2bd05645dcc67f0f7c4df0b5                              0.0s
 => => naming to docker.io/library/rtabmap_latest
```