introlab/rtabmap_ros #856: The remote computer rviz could not display cloudmap in rostopic rtabmap_ros/MapData. But in my locobot computer the rviz can display cloudmap.

sezer12138, 2022-12-16:


sezer12138, 2022-12-16:
When I rostopic echo rtabmap_ros/MapData, there is not data. But the other rtabmap topics have data. And the rostopic list is well, has rtabmap_ros topics. At the same time, the error in locobot terminal which is [ERROR] [1671167054.638756074]: Client [/rostopic_31925_1671167054167] wants topic /locobot/rtabmap/mapData to have datatype/md5sum [rtabmap_ros/MapData/e20572a812f1f74060a7f5d1a39f9106], but our version has [rtabmap_ros/MapData/8f9b7a6a9f675febbbe86bf0926d6fcd]. Dropping connection.

sezer12138, 2022-12-16:
[ INFO] [1671167023.543611041]: rtabmap (44): Rate=1.00s, Limit=0.000s, Conversion=0.0009s, RTAB-Map=0.0478s, Maps update=0.0084s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167024.572772328]: rtabmap (45): Rate=1.00s, Limit=0.000s, Conversion=0.0007s, RTAB-Map=0.0479s, Maps update=0.0057s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167025.725691488]: rtabmap (46): Rate=1.00s, Limit=0.000s, Conversion=0.0009s, RTAB-Map=0.0472s, Maps update=0.0070s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167026.624657717]: rtabmap (47): Rate=1.00s, Limit=0.000s, Conversion=0.0010s, RTAB-Map=0.0528s, Maps update=0.0044s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167027.656443482]: rtabmap (48): Rate=1.00s, Limit=0.000s, Conversion=0.0007s, RTAB-Map=0.0576s, Maps update=0.0078s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167028.692271550]: rtabmap (49): Rate=1.00s, Limit=0.000s, Conversion=0.0022s, RTAB-Map=0.0629s, Maps update=0.0045s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167029.704340612]: rtabmap (50): Rate=1.00s, Limit=0.000s, Conversion=0.0007s, RTAB-Map=0.0523s, Maps update=0.0049s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167030.730232293]: rtabmap (51): Rate=1.00s, Limit=0.000s, Conversion=0.0008s, RTAB-Map=0.0522s, Maps update=0.0069s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167031.764667448]: rtabmap (52): Rate=1.00s, Limit=0.000s, Conversion=0.0008s, RTAB-Map=0.0541s, Maps update=0.0073s pub=0.0000s (local map=1, WM=1)
[ERROR] [1671167032.312754145]: Client [/rostopic_31900_1671167031814] wants topic /locobot/rtabmap/mapData to have datatype/md5sum [rtabmap_ros/MapData/e20572a812f1f74060a7f5d1a39f9106], but our version has [rtabmap_ros/MapData/8f9b7a6a9f675febbbe86bf0926d6fcd]. Dropping connection.
[ INFO] [1671167032.782783370]: rtabmap (53): Rate=1.00s, Limit=0.000s, Conversion=0.0010s, RTAB-Map=0.0553s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167033.812089649]: rtabmap (54): Rate=1.00s, Limit=0.000s, Conversion=0.0008s, RTAB-Map=0.0446s, Maps update=0.0082s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167034.837379827]: rtabmap (55): Rate=1.00s, Limit=0.000s, Conversion=0.0007s, RTAB-Map=0.0482s, Maps update=0.0061s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167035.857987890]: rtabmap (56): Rate=1.00s, Limit=0.000s, Conversion=0.0007s, RTAB-Map=0.0465s, Maps update=0.0043s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167036.896347704]: rtabmap (57): Rate=1.00s, Limit=0.000s, Conversion=0.0008s, RTAB-Map=0.0505s, Maps update=0.0067s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167037.934212525]: rtabmap (58): Rate=1.00s, Limit=0.000s, Conversion=0.0009s, RTAB-Map=0.0659s, Maps update=0.0046s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167038.955928632]: rtabmap (59): Rate=1.00s, Limit=0.000s, Conversion=0.0011s, RTAB-Map=0.0557s, Maps update=0.0047s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167039.968839696]: rtabmap (60): Rate=1.00s, Limit=0.000s, Conversion=0.0007s, RTAB-Map=0.0452s, Maps update=0.0044s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167041.003258005]: rtabmap (61): Rate=1.00s, Limit=0.000s, Conversion=0.0014s, RTAB-Map=0.0467s, Maps update=0.0048s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167042.209155538]: rtabmap (62): Rate=1.00s, Limit=0.000s, Conversion=0.0026s, RTAB-Map=0.0955s, Maps update=0.0079s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167043.188746001]: rtabmap (63): Rate=1.00s, Limit=0.000s, Conversion=0.0008s, RTAB-Map=0.0480s, Maps update=0.0051s pub=0.0000s (local map=1, WM=1)
[ INFO] [1671167044.371055306]: rtabmap (64): Rate=1.00s, Limit=0.000s, Conversion=0.0029s

matlabbe, 2022-12-16:
For the error, it means your robot is not using the same rtabmap version than the remote computer. The msg has been updated.