introlab/rtabmap_ros #857: Using IMU data to improve the trajectory

takahiko-hasegawa-hdjp, 2022-12-18:
I am using RTABMAP-ros with our own rosbag data to estimate the trajectory using visual data. 
As the trajectory output in our environment isn't too great, i am trying to use our IMU data to to feed into rtabmap.

However, I'm a little confused how to feed the IMU input to the rtabmap system
What I tried so far is the followings:

- Change the imu topic name to our own imu topic name on launch file
- Add imu_filter_madgwick to the launch file 
- wait_imu_to_init to be true on launch file 

Above changes resulted in OdometryROS::callbackIMU function to be called everytime the imu topic is being fed to the system(rosbag data).

However, in the same callback function, it doesnt start prcoessing the data as the buffered data seems empty or not valid:
Wont call the processData function dince the bufferedData is not valid?
		if(bufferedData_.first.isValid() && stamp > bufferedData_.first.stamp())
		{
			SensorData data = bufferedData_.first;
			bufferedData_.first = SensorData();
			processData(data, bufferedData_.second);
		}

But as far as I understand, bufferedData_ is only accessed by processData function so there wouldnt be any data init when we first get the callbackIMU function called. 
I dont really undesrstand whats missing here?

Another thing I tried is to use the robot_localization package for noetic. (I read somewhere that I needed to use this package for sensor fusion??)

So, I added the following to the launch file: 

<node pkg="robot_localization" type="ekf_localization_node" name="ekf_localization" clear_params="true" output="screen">

      <param name="frequency" value="50"/>
      <param name="sensor_timeout" value="0.1"/>
      <param name="two_d_mode" value="false"/>

      <param name="odom_frame" value="odom"/>
      <param name="base_link_frame" value="$(arg frame_id)"/>
      <param name="world_frame" value="odom"/>

      <param name="transform_time_offset" value="0.0"/>

      <param name="odom0" value="/vo"/>
      <param name="imu0" value="$(arg imu_topic)"/> 

      
      <rosparam param="odom0_config">[true, true, true,
                                      false, false, false,
                                      true, true, true,
                                      false, false, false,
                                      false, false, false]</rosparam>

      <rosparam     if="$(arg imu_ignore_acc)" param="imu0_config">[
                                     false, false, false,
                                     true,  true,  true,
                                     false, false, false,
                                     true,  true,  true,
                                     false,  false,  false] </rosparam>
      <rosparam unless="$(arg imu_ignore_acc)" param="imu0_config">[
                                     false, false, false,
                                     true,  true,  true,
                                     false, false, false,
                                     true,  true,  true,
                                     true,  true,  true] </rosparam>  
      
      <param name="odom0_differential" value="false"/>
      <param name="imu0_differential" value="false"/>

      <param name="odom0_relative" value="true"/>
      <param name="imu0_relative" value="true"/>

      <param name="imu0_remove_gravitational_acceleration" value="$(arg imu_remove_gravitational_acceleration)"/>

      <param name="print_diagnostics" value="true"/>

      
      <param name="odom0_queue_size" value="5"/>
      <param name="imu0_queue_size" value="50"/> 

      
      <rosparam param="process_noise_covariance">[0.005, 0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                                                  0,    0.005, 0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                                                  0,    0,    0.006, 0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                                                  0,    0,    0,    0.003, 0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                                                  0,    0,    0,    0,    0.003, 0,    0,     0,     0,    0,    0,    0,    0,    0,    0,
                                                  0,    0,    0,    0,    0,    0.006, 0,     0,     0,    0,    0,    0,    0,    0,    0,
                                                  0,    0,    0,    0,    0,    0,    0.0025, 0,     0,    0,    0,    0,    0,    0,    0,
                                                  0,    0,    0,    0,    0,    0,    0,     0.0025, 0,    0,    0,    0,    0,    0,    0,
                                                  0,    0,    0,    0,    0,    0,    0,     0,     0.004, 0,    0,    0,    0,    0,    0,
                                                  0,    0,    0,    0,    0,    0,    0,     0,     0,    0.001, 0,    0,    0,    0,    0,
                                                  0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0.001, 0,    0,    0,    0,
                                                  0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0.002, 0,    0,    0,
                                                  0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0.001, 0,    0,
                                                  0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0.001, 0,
                                                  0,    0,    0,    0,    0,    0,    0,     0,     0,    0,    0,    0,    0,    0,    0.0015]</rosparam>

      
           <rosparam param="initial_estimate_covariance">[1e-9, 0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                                                          0,    1e-9, 0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                                                          0,    0,    1e-9, 0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                                                          0,    0,    0,    1e-9, 0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                                                          0,    0,    0,    0,    1e-9, 0,    0,    0,    0,    0,     0,     0,     0,    0,    0,
                                                          0,    0,    0,    0,    0,    1e-9, 0,    0,    0,    0,     0,     0,     0,    0,    0,
                                                          0,    0,    0,    0,    0,    0,    1e-9, 0,    0,    0,     0,     0,     0,    0,    0,
                                                          0,    0,    0,    0,    0,    0,    0,    1e-9, 0,    0,     0,     0,     0,    0,    0,
                                                          0,    0,    0,    0,    0,    0,    0,    0,    1e-9, 0,     0,     0,     0,    0,    0,
                                                          0,    0,    0,    0,    0,    0,    0,    0,    0,    1e-9,  0,     0,     0,    0,    0,
                                                          0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     1e-9,  0,     0,    0,    0,
                                                          0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     1e-9,  0,    0,    0,
                                                          0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     1e-9, 0,    0,
                                                          0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    1e-9, 0,
                                                          0,    0,    0,    0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    1e-9]</rosparam>

    </node>

But no luck. 

Could anyone help me on how to properly feed the IMU data. Also is there anyway that the IMU data is being used?? like a visualiser?

Thanks in advance.

Basic setup

Camera: Zed2i (rgb and depth image)
Zed2 sdk version: 3.7
Image size: 1280x720
Frequency: ~10Hz

matlabbe, 2022-12-21:
If you want to use IMU with rtabmap's odometry, for the best pose accuracy, I'll suggest to use `stereo_odometry`. See Zed related instructions on this page: http://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping

For the imu callback in OdometryROS, the `bufferedData_` will be set when the first image is received:
https://github.com/introlab/rtabmap_ros/blob/3040a5d8f5c5c80bd5cffddd26f6f189c56dee91/src/OdometryROS.cpp#L476-L477

then on next imuCallback, when imu_ buffer has enough data, processData will be done on the buffered image. In other world, The code is waiting valid imu data before processing the images. This is used to corectly initialize the orientation of the first odometry pose, accordingly to gravity, then imu is used in local BA to keep odometry aligned to gravity over time. IMU is also used to get better orientation guess for odometry's next visual frame.

takahiko-hasegawa-hdjp, 2022-12-21:
@matlabbe Thanks for your answer. 

OK, now i understand how bufferedData_ is set. 

In terms of the IMU integration, unfortunately I cannot use stereo_odometry due to the fact that the setup we have is with a camera with rgb-d info so no separate topic for left image and right image. 

Is there anything else that you would recommend in order to fuse the IMU information?

matlabbe, 2022-12-21:
rgbd_odometry can also use IMU the same way.

Beside rtabmap, if you have wheel odometry and IMU, you can use `robot_localization` or [robot_pose_ekf](http://wiki.ros.org/robot_pose_ekf) to fuse them. See their [documentation](http://docs.ros.org/en/noetic/api/robot_localization/html/index.html) to know how to do that.

takahiko-hasegawa-hdjp, 2022-12-21:
Do you recommend using [robot_localization](http://docs.ros.org/en/noetic/api/robot_localization/html/index.html) for imu fusion? If i understand this correctly, IMU data is only used for initial pose within rtabmap?

matlabbe, 2022-12-26:
On rtabmap I added:

> then imu is used in local BA to keep odometry aligned to gravity over time. IMU is also used to get better orientation guess for odometry's next visual frame.

For robot_localization, personally, I only used it for wheel odometry + IMU fusion (which worked well). For other configurations, you will have to ask support on their repo.

takahiko-hasegawa-hdjp, 2022-12-27:
Is there any launch file that i can take a look at for imu fusion for robot localization?

matlabbe, 2022-12-30:
May check https://github.com/cra-ros-pkg/robot_localization/blob/noetic-devel/params/ekf_template.yaml