introlab/rtabmap_ros #859: rtabmap_ros and rtabmap versions not compatible

GilMe, 2022-12-22:
Hi,
I downloaded and installed the latest versions of rtabmap and rtabmab_ros. When trying to install rtabmap_ros i get the following error:

CMake Error at /home/cs3/new_nano_ws/src/mapping/rtabmap_ros/CMakeLists.txt:35 (find_package):
  Could not find a configuration file for package "RTABMap" that is
  compatible with requested version "0.20.22".

  The following configuration files were considered but not accepted:

    /usr/local/lib/rtabmap-0.20/RTABMapConfig.cmake, version: 0.20.21

It seems like there is a problem with the versions of two libraries. When I changed the CMakelists.txt in rtabmap repository (line 23) to this:
SET(RTABMAP_PATCH_VERSION 22)

and reinstall rtabmap it works

matlabbe, 2022-12-23:
It should be already 22. 
https://github.com/introlab/rtabmap/blob/f12cc83fc2b40392726d683200809c0ecf4186e2/CMakeLists.txt#L23

umbnich, 2023-03-31:
Yes, it is, but in /usr/local/lib/, I only have rtabmap-0.20.

How to solve?

Thanks

matlabbe, 2023-04-02:
The version `0.20` includes `0.20.21` and `0.20.22`. Note that minor version has been updated not so long ago, if you reinstall latest version, it will be installed in `rtabmap-0.21`.