introlab/rtabmap_ros #862: Run RTABMAP on multiple rosbags simultaneosly

vocdex, 2022-12-29:
First of all, thanks a lot for sharing and maintaining the codebase.
I was able to run RTabMap and get really good 3d Maps. However, I have hundreds of rosbags with all the necessary topics for RTabMap usage. As far as I know, to create a map, we have to play the rosbags one by one and save their maps separately. Is there an option to have a single script or package that takes a rosbag as an input, runs RTabMap and saves the 3d map for each rosbag without playing the rosbags?
Thanks!

matlabbe, 2023-01-09:
Good question, I have never tried this myself. With [rosbag library](http://wiki.ros.org/rosbag/Code%20API), it could be possible to read the rosbag files offline and feed the data to rtabmap. However, for all topic synchronization stuff, I feel it would be complicated to provide correctly the topics to rtabmap node.

If I had to process offline hundred of bags, I would start by converting them to rtabmap's database format. When recording only data to a database, we can play the bag at faster rate without missing data. Take a look at the [data_recorder.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/data_recorder.launch) file. Topic synchronization will be done, and high rate data can be recorded.
```bash
roslaunch rtabmap_ros data_recorder.launch 
rosbag play --clock -r 10 my_recorded_data.bag
```
This will create `~/.ros/output.db` file. To create maps, we can reprocess this database offline like this (note that I added `--default` option in this [commit](https://github.com/introlab/rtabmap/commit/8d301afe6c1421042a71e94c8f8f90db167867d8), otherwise we have to manually override the parameters set by `data_recorder.launch`):
```bash
rtabmap-reprocess --default ~/.ros/output.db map.db
```

Here is a real example using the [demo_mapping.bag](https://docs.google.com/uc?id=0B46akLGdg-uadXhLeURiMTBQU28&export=download) used for the examples [here](http://wiki.ros.org/rtabmap_ros#Robot_mapping).

1. Converting rosbag to rtabmap's database (at 10 times faster than real-time)
```bash
$ rosbag info demo_mapping.bag 
path:        demo_mapping.bag
version:     2.0
duration:    4:05s (245s)
start:       May 16 2013 11:46:43.35 (1368730003.35)
end:         May 16 2013 11:50:49.28 (1368730249.28)
size:        295.0 MB
messages:    84745
compression: none [382/382 chunks]
types:       nav_msgs/Odometry           [cd5e73d190d741a2f92e81eda573aca7]
             sensor_msgs/CameraInfo      [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/CompressedImage [8f7a12909da2c9d3332d540a0977563f]
             sensor_msgs/LaserScan       [90c7ef2dc6895d81024acba2ac42f369]
             tf/tfMessage                [94810edda583a504dfda3829e70d7eec]
topics:      /az3/base_controller/odom                    11936 msgs    : nav_msgs/Odometry          
             /jn0/base_scan                                9843 msgs    : sensor_msgs/LaserScan      
             /tf                                          56909 msgs    : tf/tfMessage               
             data_throttled_camera_info                    2019 msgs    : sensor_msgs/CameraInfo     
             data_throttled_image/compressed               2019 msgs    : sensor_msgs/CompressedImage
             data_throttled_image_depth/compressedDepth    2019 msgs    : sensor_msgs/CompressedImage

$ roslaunch rtabmap_ros data_recorder.launch \
   odom_topic:=/az3/base_controller/odom \
   approx_sync:=true \
   frame_id:=base_footprint  \
   subscribe_odometry:=true \
   max_rate:=1 \
   rgb_topic:=/data_throttled_image \
   rgb_info_topic:=/data_throttled_camera_info \
   rgb_image_transport:=compressed depth_topic:=/data_throttled_image_depth \
   depth_image_transport:=compressedDepth \
   subscribe_scan:=true \
   scan_topic:=/jn0/base_scan

$ rosparam set use_sim_time true

$ rosbag play --clock -r 10 demo_mapping.bag
```

2. Creating the map offline:
```bash
$ rtabmap-reprocess \
   --default \
   --Reg/Strategy 1 \
   --Reg/Force3DoF true \
   --RGBD/CreateOccupancyGrid true \
   --Grid/Sensor 0 \
   --RGBD/ProximityPathMaxNeighbors 10 \
   --RGBD/NeighborLinkRefining true \
   ~/.ros/output.db \
   map.db
```

3. Visualizing the results:
```bash
$ rtabmap-databaseViewer map.db
```
[image]

4. Exporting the 3D point cloud:
```bash
$ rtabmap-export map.db
```
[image]

Summary: Steps 1-2-4 could be scripted.

vocdex, 2023-01-09:
Awesome! Thanks a lot for your response, it's really helpful!

vocdex, 2023-01-09:
Sorry, one last question.
I could run your example above and I have no problem, However, if I try follow the same process with my rosbag, after step 4, I get the following error:
``` bash 
 $ rtabmap-export map.db 
Loading database "map.db"...
Loading database "map.db"... done (0.015211s).
Optimizing the map...
Optimizing the map... done (0.008731s, poses=0).
The optimized graph is empty!? Aborting...
```
In short, the 3d map is not being saved and odometry data is also 0.
My rosbag contains Realsense d435 /rgb and /aligned_depth topics along with CameraInfo topics. Here is a sample rosbag in Google Drive [d435.bag](https://drive.google.com/file/d/1wHXSvOJWO70_w0w2PtGRNIjCq81Kcmkl/view?usp=sharing) ~1.5GB
Here are the order of commands I have typed:
- roslaunch realsense2_description view_d435_model.launch   
- roslaunch rtabmap_ros data_recorder.launch approx_sync:=false max_rate:=1 \ rgb_topic:=/camera/color/image_raw \ depth_topic:=/camera/aligned_depth_to_color/image_raw \ rgb_info_topic:=/camera/aligned_depth_to_color/camera_info 
- rosbag play -r 10 2022-09-22-11-41-22_21.bag 
- rtabmap-reprocess --default ~/.ros/output.db map.db 

Info about output.db:
```bash
Info:

Path:               output.db
Version:            0.20.22
Sessions:           1
Total odometry length:0.000000 m
Total time:         9.060867s
LTM:                10 nodes and 0 words (dim=0 type=8U)
WM:                 2 nodes and 0 words
Global graph:       0 poses and 0 links
Optimized graph:    0 poses (x=0->0, y=0->0, z=0->0)
Maps in graph:      0/1 []
Ground truth:       0 poses
GPS:                0 poses
Links:
  Neighbor:         0
  GlobalClosure:    0
  LocalSpaceClosure:0
  LocalTimeClosure: 0
  UserClosure:      0
  VirtualClosure:   0
  NeighborMerged:   0
  PosePrior:        0
  Landmark:         0
  Gravity:          0

Database size:      7 MB
Nodes size:         1 KB	(0.02%)
Links size:         6 KB	(0.08%)
RGB Images size:    4 MB	(63.76%)
Depth Images size:  2 MB	(34.33%)
Calibrations size:  3 KB	(0.05%)
Grids size:         120 Bytes	(0.00%)
Scans size:         0 Bytes	(0.00%)
User data size:     0 Bytes	(0.00%)
Dictionary size:    0 Bytes	(0.00%)
Features size:      0 Bytes	(0.00%)
Statistics size:    9 KB	(0.12%)
Other (indexing, unused):126 KB	(1.64%)
```

However, if I process the rosbag in usual way like the following, I can get 3D map:
```bash
- roslaunch realsense2_description view_d435_model.launch   
- roslaunch rtabmap_ros rtabmap.launch database_path:="my.path/to/database" /
rtabmap_args:="--delete_db_on_start" depth_topic:=/camera/aligned_depth_to_color/image_raw rgb_topic:=/camera/color/image_raw /
 camera_info_topic:=/camera/aligned_depth_to_color/camera_info approx_sync:=false
```
This approach works. So, I guess there is something wrong with my data_recorder launch params. 
Thanks!

matlabbe, 2023-01-15:
The problem is that you don't record any odometry. I updated `rtabmap-reprocess` tool in this [commit](https://github.com/introlab/rtabmap/commit/22fede93355bb59986e26cbb6b43d2a1c34413d7) to be able to process a database with only frames without odometry. The new option will make `rtabmap-reprocess` recomputing the odometry for us. 

Example with your bag containing only frames:
```
rosbag info 2022-09-22-11-41-22_21.bag
path:        2022-09-22-11-41-22_21.bag
version:     2.0
duration:    10.0s
start:       Sep 21 2022 19:41:22.17 (1663814482.17)
end:         Sep 21 2022 19:41:32.15 (1663814492.15)
size:        1.5 GB
messages:    1200
compression: none [900/900 chunks]
types:       sensor_msgs/CameraInfo [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/Image      [060021388200f6f0f447d0fcd9c64743]
topics:      /camera/aligned_depth_to_color/camera_info   300 msgs    : sensor_msgs/CameraInfo
             /camera/aligned_depth_to_color/image_raw     300 msgs    : sensor_msgs/Image     
             /camera/color/image_raw                      300 msgs    : sensor_msgs/Image     
             /camera/depth/image_rect_raw                 300 msgs    : sensor_msgs/Image
```
1. Bag to db conversion
```bash
# Launch a roscore with sim time:
roscore
rosparam set use_sim_time true

# To get D435 frames (note that for convenience, you could 
# record your bag with `/tf` and `/tf_static` so that we don't need 
# to publish the tf separately from the bag)
roslaunch realsense2_description view_d435_model.launch

# In this case we want all 300 frames in the database, so set max_rate:=0
roslaunch rtabmap_ros data_recorder.launch approx_sync:=false max_rate:=0 \
   rgb_topic:=/camera/color/image_raw \
   depth_topic:=/camera/aligned_depth_to_color/image_raw \
   rgb_info_topic:=/camera/aligned_depth_to_color/camera_info

rosbag play -r 10 --clock 2022-09-22-11-41-22_21.bag
```
2. Reprocess the map with odometry enabled:
```bash
rtabmap-reprocess --default --odom ~/.ros/output.db map.db
```

Hope this will help.

vocdex, 2023-01-16:
Thanks! It worked but I had to build rtabmap and rtabmap_ros separately. The latest commits ([--odom](https://github.com/introlab/rtabmap/commit/22fede93355bb59986e26cbb6b43d2a1c34413d7) and [--default](https://github.com/introlab/rtabmap/commit/8d301afe6c1421042a71e94c8f8f90db167867d8)) don't appear when I do:

- sudo apt install ros-$ROS_DISTRO-rtabmap-ros

At the moment, this will install rtabmap-reprocess without --odom and --default options.
Thanks again!

closed 2023-01-16 by vocdex