introlab/rtabmap_ros #863: deskew method is called despite parameter false deskewing_ setting which leads to error

talhaaliarslan, 2023-01-03:
Encountered this issue when using a simulated lidar via the default laser plugins from `gazebo_ros_pkgs` after upgrading `rtabmap_ros` to `0.20.22-noetic`. The laser scan is instantaneous and has the `time_increment` or `scan_time` fields in `LaserScan` message as `0`. Deskewing feature seems to have a parameter `deskewing_` to toggle it on or off. It is observed that `rtabmap_ros::deskew` method is called on the line below, despite this setting (`false`). A mitigation of changing those `LaserScan` fields to non-zero values is tested and it verified that indeed the current state does not work with instantaneous scans. 

https://github.com/introlab/rtabmap_ros/blob/d9996a03f83ffe09c17571a2dc3f861e3c95b4b9/src/nodelets/icp_odometry.cpp#L370

As a result, I get these errors repeatedly:

```
[ERROR][/icp_odometry][1672743139.506061137, 0.707500000]: First and last stamps in the scan are the same!
[ERROR][/icp_odometry][1672743139.506093092, 0.707500000]: Failed to deskew input cloud, aborting odometry update!
```

I would be happy to propose a change to fix this, but the intended sequence of steps here is not clear to me yet.

matlabbe, 2023-01-03:
This bug has been fixed in that commit https://github.com/introlab/rtabmap_ros/commit/d09faa9385649f33b7c417164e98ff445837a392

I'll try to make a new binary release next weekend.

talhaaliarslan, 2023-01-11:
Hi, I tested the commit above. The issue I was having is resolved.

closed 2023-01-16 by matlabbe