introlab/rtabmap_ros #865: How to use VIO and 2d lidar in rtabmap?

CoCube, 2023-01-05:
Hi all, I am a newbie and I would like to use my 2D radar with D435I to navigate my TurtleBOT3 for unknown maps.
I am based on https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/sensor_fusion.launch

Yes the RGBD of the D435 fuses the data from the IMU.
Now I wish to use my Rplidar A1M8 along with VIO.
According to the tutorial http://wiki.ros.org/rtabmap_ros/Tutorials/SetupOnYourRobot#Kinect_.2B-_2D_laser
But I didn't figure out how to add the LaserScan obtained from 2D Lidar to sensor_fusion.launch.
So how should I fuse the two?

maekjuplease, 2023-01-07:
up for this.

matlabbe, 2023-01-09:
You can feed 2d lidar to rtabmap node directly. With `Reg/Strategy=1` and `RGBD/NeighborLinkRefining=true`, rtabmap will correct the input VIO odometry based on the scan.

To do fusion of lidar in `robot_localization`, you would need lidar odometry. You may generate lidar odometry with `icp_odometry` or hector mapping node. Personally, the first approach above is enough in most case and use the less computation power.

CoCube, 2023-01-09:
<param name="RGBD/NeighborLinkRefining" type="string" value="true"/>
<param name="RGBD/ProximityBySpace"     type="string" value="true"/>
<param name="RGBD/AngularUpdate"        type="string" value="0.01"/>
<param name="RGBD/LinearUpdate"         type="string" value="0.01"/>
<param name="RGBD/OptimizeFromGraphEnd" type="string" value="false"/>
<param name="Grid/FromDepth"            type="string" value="false"/> 
<param name="Reg/Force3DoF"             type="string" value="true"/>
<param name="Reg/Strategy"              type="string" value="1"/> 

<param name="Icp/VoxelSize"                 type="string" value="0.05"/>
<param name="Icp/MaxCorrespondenceDistance" type="string" value="0.1"/>

matlabbe, 2023-01-15:
For odom correction with lidar, the minimum would be:
```
//"RGBD/NeighborLinkRefining" type="string" value="true"/
//"Reg/Force3DoF" type="string" value="true"/
//"Reg/Strategy" type="string" value="1"/
```

Optional/tuning:
```
//"RGBD/ProximityBySpace" type="string" value="true"/
//"Icp/VoxelSize" type="string" value="0.05"/
//"Icp/MaxCorrespondenceDistance" type="string" value="0.1"/
//"Grid/FromDepth" type="string" value="false"/
```

CoCube, 2023-01-16:
> 对于使用激光雷达的奥多姆校正，最小值为：
> 
> ```
> //"RGBD/NeighborLinkRefining" type="string" value="true"/
> //"Reg/Force3DoF" type="string" value="true"/
> //"Reg/Strategy" type="string" value="1"/
> ```
> 
> 可选/调整：
> 
> ```
> //"RGBD/ProximityBySpace" type="string" value="true"/
> //"Icp/VoxelSize" type="string" value="0.05"/
> //"Icp/MaxCorrespondenceDistance" type="string" value="0.1"/
> //"Grid/FromDepth" type="string" value="false"/
> ```

Thank you very much for helping me clear my doubts, I think I understand how to do it. Please accept my thanks again.

closed 2023-01-16 by CoCube