introlab/rtabmap_ros #866: RTABmap does not display any map

faheem-khaskheli, 2023-01-07:
i am using ros2 humble, rtabmap_ros install using sudo apt install ros-humble-rtabmap-ros.

i am trying to use rgbd_image, i tried both sync using rtabmap_ros/rgbd_sync and without rgbd_sync, i create a single function in python to publish all data at the same time and header.stamp is same time  for all 3.
rgb/image
rgb/camera_info
depth/image

i am using python code, to achieve this, i am getting data from webots simulation.
RGB image, displayed in rviz2 correctly. topic name "rgb/image"
Depth image, displayed in rviz2 correctly. topic name "depth/image"

rgb image is uint8, rgb8 type
depth image is uint8, mono type. where 0 mean close, and 255 mean far. it is captured using RangeSensor. it is in 2D, same width and height as rgb image.

odom work fine, and is displayed correctly in rviz2.
scan also work fine, i am able to create 2D scan of surrounding using slam_tollbox.

can you help me with the setup of rtabmap_ros, i tried all sort of parameters but none of them display any map results.

faheem-khaskheli, 2023-01-07:
My code
`
        self.camera_info_pubs = self.create_publisher(CameraInfo, "/rgb/camera_info", 10)
        self.img_pubs = self.create_publisher(Image, "/rgb/image", 10)
        self.depth_img_pubs = self.create_publisher(Image, "/depth/image", 10)
        self.create_timer(1.0, self.publish_rgbd_image)

        self.lidar_pubs = self.create_publisher(LaserScan, "scan", 10)
        self.create_timer(1.0, self.publish_lidar)

        self.odom_pub = self.create_publisher(Odometry, "odom", 10)
        self.create_timer(1.0, self.publish_odom)
----   
def publish_rgbd_image():   
        cam_info = CameraInfo()
        time_stamp = rclpy.time.Time(seconds=self._robot.getTime()).to_msg()
        cam_info.header.stamp = time_stamp
        cam_info.header.frame_id = "base_link"
        cam_info.height = self._webots_controller._camera.getHeight()
        cam_info.width = self._webots_controller._camera.getWidth()
        cam_info.distortion_model = "plumb_bob"
        cx = cam_info.width / 2.0
        cy = cam_info.height / 2.0

        fx = cam_info.width / 2.0 * tan(0.5 * self._webots_controller._camera.getFov())
        fy = cam_info.height / 2.0 * tan(0.5 * self._webots_controller._camera.getFov())

        cam_info.k = [fx, 0.0, cx, 0.0, fy, cy, 0.0, 0.0, 1.0]
        cam_info.d = [0.0, 0.0, 0.0, 0.0, 0.0]
        cam_info.r = [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
        cam_info.p = [fx, 0.0, cx, 0.0, 0.0, fy, cy, 0.0, 0.0, 0.0, 1.0, 0.0]

        self.camera_info_pubs.publish(cam_info)

        img_list, img_shape = self._webots_controller.get_camera_rgb_image()
        ros_img = Image(data=img_list, width=img_shape[0], height=img_shape[1], encoding='rgb8', step=img_shape[0]*3)
        ros_img.header.stamp = time_stamp

        img_list, img_shape = self._webots_controller.get_camera_depth_image()
        ros_depth_img = Image(data=img_list, width=img_shape[0], height=img_shape[1], encoding='mono8', step=img_shape[0]*1)
        ros_depth_img.header.stamp = time_stamp

        self.img_pubs.publish(ros_img)
        self.depth_img_pubs.publish(ros_depth_img)`

faheem-khaskheli, 2023-01-07:
recent error.

since i am using webots for simulation i have to manually configure camera info and image datatype, and range datatype i think there is some issue.,
[ERROR]  util3d.cpp:1231::cloudRGBFromSensorData() Stereo camera model 0 is invalid

faheem-khaskheli, 2023-01-09:
i will create a bag today and share,
but to let you know i am not using real camera, so i have to input camera info manually,
this is my code.
`
        cam_info = CameraInfo()
        time_stamp = rclpy.time.Time(seconds=self._robot.getTime()).to_msg()
        cam_info.header.stamp = time_stamp
        cam_info.header.frame_id = "base_link"
        cam_info.height = self._webots_controller._camera.getHeight()
        cam_info.width = self._webots_controller._camera.getWidth()
        cam_info.distortion_model = "plumb_bob"
        cx = cam_info.width / 2.0
        cy = cam_info.height / 2.0

        fx = cam_info.width / 2.0 * tan(0.5 * self._webots_controller._camera.getFov())
        fy = cam_info.height / 2.0 * tan(0.5 * self._webots_controller._camera.getFov())

        cam_info.k = [fx, 0.0, cx, 0.0, fy, cy, 0.0, 0.0, 1.0]
        cam_info.d = [0.0, 0.0, 0.0, 0.0, 0.0]
        cam_info.r = [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
        cam_info.p = [fx, 0.0, cx, 0.0, 0.0, fy, cy, 0.0, 0.0, 0.0, 1.0, 0.0]

        self.camera_info_pubs.publish(cam_info)

`

faheem-khaskheli, 2023-01-09:
can you look at code and tell me how can i input manual camera info for testing purpose.
i want to make sure code is correct so that i can buy real camera for usage.

faheem-khaskheli, 2023-01-09:
```
        img_list, img_shape = self._webots_controller.get_camera_rgb_image()
        ros_img = Image(data=img_list, width=img_shape[0], height=img_shape[1], encoding='rgb8', step=img_shape[0]*3)
        ros_img.header.stamp = time_stamp

        img_list, img_shape = self._webots_controller.get_camera_depth_image()
        ros_depth_img = Image(data=img_list, width=img_shape[0], height=img_shape[1], encoding='mono8', step=img_shape[0]*1)
        ros_depth_img.header.stamp = time_stamp

        self.img_pubs.publish(ros_img)
        self.depth_img_pubs.publish(ros_depth_img)

```

closed 2023-01-09 by faheem-khaskheli

matlabbe, 2023-01-16:
> thanks, its solved, issue was depth image, was in mono8 formate, but it required mono 16. and pixel values from webots are in meter but rtabmap need in millimeters. also need to convert inf with 0

rtabmap requires depth images to have format mono 16 in mm, or 32 float in meters.