introlab/rtabmap_ros #871: RTABMAP cannot activate VINS-Fusion

maekjuplease, 2023-01-12:
Hi Mathew,

I tried to build rtabmap with -DWITH_VINS-Fusion=ON but it was unsuccessful. Could you please give me some advice for this? I also attached the file below.

Thank you.

[image]

maekjuplease, 2023-01-12:
https://github.com/introlab/rtabmap_ros/issues/871#top

matlabbe, 2023-01-16:
Is VINS-fusion built in your `catkin_ws` ? If you are on ubuntu 20.04, vins fusion is not compatible with opencv4 out of the box, clone it and apply [pull request 136](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/pull/136), apply this [patch](https://gist.github.com/matlabbe/795ab37067367dca58bbadd8201d986c#file-vins-fusion_be55a93_pull136-patch), then catkin_make. Return to rtabmap library and redo cmake, vins-fusion should then be found (assuming your catkin workspace is sourced).