introlab/rtabmap_ros #874: rviz doesn't map much and is not the same as rtabmapviz

khuechuong, 2023-01-20:
Basically, I am using a zed mini and using their ROS package. I use ubuntu 20.04, ros noetic, ros1, with a gtx 1080 gpu. I follow the installation instruction: https://www.stereolabs.com/docs/ros/
And I used rtabmap ros. In the beginning, both rtabmapviz and rviz wasn't that good as seen here.

RVIZ with  _/zed/zed_node/point_cloud/cloud_registered_ topic:
[image]

RVIZ without  _/zed/zed_node/point_cloud/cloud_registered_ topic:
[image]

So I **changed** the param _subscribe_scan_cloud_ to true and add <remap from="scan_cloud"        to="/zed/zed_node/point_cloud/cloud_registered"/> to rtabmap_ros and rtabmapviz. That really helped rtabmapviz, but rviz still just doesn't have the good quality of mapping like I see in other youtube. The command I ran was _roslaunch zed_rtabmap_example zed_rtabmap.launch_

**zed_rtabmap.launch**

<?xml version="1.0"?>

<launch>
    <arg name="svo_file"             default="" /> 
    <arg name="stream"               default="" /> 

    <arg name="zed_node_name"        default="zed_node" />
    <arg name="camera_model"         default="zedm" />
    <arg name="publish_urdf"         default="true" />

    <arg name="camera_name"          default="zed" />

    <arg name="base_frame"           default="base_link" />

    <group ns="$(arg camera_name)">
        
        <include file="$(find zed_wrapper)/launch/include/zed_camera.launch.xml">
            <arg name="camera_name"         value="$(arg camera_name)" />
            <arg name="svo_file"            value="$(arg svo_file)" />
            <arg name="stream"              value="$(arg stream)" />
            <arg name="node_name"           value="$(arg zed_node_name)" />
            <arg name="camera_model"        value="$(arg camera_model)" />
            <arg name="base_frame"          value="$(arg base_frame)" />
            <arg name="publish_urdf"        value="$(arg publish_urdf)" />
        </include>    
  
        
        <arg name="custom_rtabmap_launch_file" default="$(find zed_rtabmap_example)/launch/include/sl_rtabmap.launch.xml"/>
        <include file="$(arg custom_rtabmap_launch_file)">
            <arg name="odom_topic"              default="$(arg zed_node_name)/odom" />
            <arg name="rgb_topic"               default="$(arg zed_node_name)/rgb/image_rect_color" />
            <arg name="depth_topic"             default="$(arg zed_node_name)/depth/depth_registered" />
            <arg name="camera_info_topic"       default="$(arg zed_node_name)/rgb/camera_info" />
            <arg name="depth_camera_info_topic" default="$(arg zed_node_name)/depth/camera_info" />
        </include>   

        
        <node name="rviz" pkg="rviz" type="rviz" args="-d $(find zed_rtabmap_example)/rviz/zed-rtabmap.rviz" output="screen" /> 
    </group>
</launch>

**sl_rtabmap.launch.xml**

<launch>
    
    <arg name="localization"            default="false"/>

    <arg     if="$(arg localization)" name="args" default=""/>
    <arg unless="$(arg localization)" name="args" default="--delete_db_on_start"/>

    
    <arg name="rgb_topic"               default="/zed/zed_node/rgb/image_rect_color" />
    <arg name="depth_topic"             default="/zed/zed_node/depth/depth_registered" />
    <arg name="camera_info_topic"       default="/zed/zed_node/rgb/camera_info" />
    <arg name="depth_camera_info_topic" default="/zed/zed_node/depth/camera_info" />

    <arg name="odom_topic"              default="/zed/zed_node/odom" />    
    

    
    

    
    <node name="rtabmapviz" pkg="rtabmap_ros" type="rtabmapviz" output="screen" args="" launch-prefix="">    
        <remap from="rgb/image"         to="$(arg rgb_topic)"/>
        <remap from="depth/image"       to="$(arg depth_topic)"/>
        <remap from="rgb/camera_info"   to="$(arg camera_info_topic)"/>

        <remap from="grid_map"          to="map" />
        <remap from="odom"              to="$(arg odom_topic)"/>
        
        
        <remap from="scan_cloud"        to="/zed/zed_node/point_cloud/cloud_registered"/>	
                
        
    </node> 

    
    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg args)" launch-prefix="">
        <rosparam command="load" file="$(find zed_rtabmap_example)/params/rtabmap.yaml" />
    
        <remap from="rgb/image"         to="$(arg rgb_topic)"/>
        <remap from="depth/image"       to="$(arg depth_topic)"/>
        <remap from="rgb/camera_info"   to="$(arg camera_info_topic)"/>

        <remap from="grid_map"          to="map" />
        <remap from="odom"              to="$(arg odom_topic)"/>	
        
        
        <remap from="scan_cloud"        to="/zed/zed_node/point_cloud/cloud_registered"/>	        
        	

	
        
        <param     if="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="false"/>
        <param unless="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="true"/>
        <param name="Mem/InitWMWithAllNodes" type="string" value="$(arg localization)"/>
    </node>  

</launch>

**rtabmap.yaml**
subscribe_depth:        true
subscribe_rgbd:         false
subscribe_stereo:       false
subscribe_stereo:       false
subscribe_scan:         false
subscribe_scan_cloud:   true
subscribe_user_data:    false
subscribe_odom_info:    false

database_path:      "~/.ros/rtabmap.db"
config_path:        "~/.ros/rtabmap.cfg"

frame_id:       "base_link"
map_frame_id:   "map"
odom_frame_id:  "odom"                   # odometry from odom msg to have covariance - Remapped by launch file
odom_tf_angular_variance:  0.001     # If TF is used to get odometry, this is the default angular variance
odom_tf_linear_variance:   0.001     # If TF is used to get odometry, this is the default linear variance
tf_delay:       0.02
publish_tf:     false                # Set to false if fusing different poses (map->odom) with UKF

odom_sensor_sync:               true
wait_for_transform_duration:    0.2
approx_sync:                    true

queue_size: 100

scan_normal_k:  0

Grid:
    3D:                     true
    FlatObstacleDetected:   true
    MapFrameProjection:     false
    GroundIsObstacle:       false
    PreVoxelFiltering:      true
    RayTracing:             true
    FromDepth:              true
    NormalsSegmentation:    false
    CellSize:               0.05
    ClusterRadius:          0.1
    MinClusterSize:         3
    DepthDecimation:        1
    DepthRoiRatios:         [0.0, 0.0, 0.0, 0.0]
    FootprintHeight:        2.0
    FootprintLength:        0.18
    FootprintWidth:         0.18
    MaxGroundAngle:         30.0
    MinGroundHeight:        -0.5
    MaxGroundHeight:        -0.4
    MaxObstacleHeight:      0.1
    NoiseFilteringMinNeighbors: 5
    NoiseFilteringRadius:   0.1
    NormalK:                20
    RangeMin:               0.7
    RangeMax:               3.0

GridGlobal:
    Eroded:                 false       # Erode obstacle cells
    FootprintRadius:        0.18        # Footprint radius (m) used to clear all obstacles under the graph
    FullUpdate:             true        # When the graph is changed, the whole map will be reconstructed instead of moving individually each cells of the map. Also, data added to cache won't be released after updating the map. This process is longer but more robust to drift that would erase some parts of the map when it should not
    MaxNodes:               0           # Maximum nodes assembled in the map starting from the last node (0=unlimited)
    MinSize:                1.0        # Minimum map size (m)
    OccupancyThr:           0.55        # Occupancy threshold (value between 0 and 1)
    ProbClampingMax:        0.971       # Probability clamping maximum (value between 0 and 1)
    ProbClampingMin:        0.1192      # Probability clamping minimum (value between 0 and 1)
    ProbHit:                0.7         # Probability of a hit (value between 0.5 and 1)
    ProbMiss:               0.4         # Probability of a miss (value between 0 and 0.5)
    UpdateError:            0.01        # Graph changed detection error (m). Update map only if poses in new optimized graph have moved more than this value

Here's a picture of what it looks like:

RVIZ with  _/zed/zed_node/point_cloud/cloud_registered_ topic:
[image]

RVIZ with  _/zed/zed_node/point_cloud/cloud_registered_ topic:
[image]

As you can see, in rtabmapviz, the quality improved drastically, but rviz mapping mapping points barely had anything. I don't really understand why. Any thought and help is appreciated. Let me know if you have any questions. Thank you!

matlabbe, 2023-01-21:
In rviz, use `rtabmap_ros/MapCloud` display instead of a PointCloud2 display. The `rtabmap/cloud_map` topic contains very downsampled point cloud for occupancy grid. In your second screen shot, you can see the point cloud in rtabmapviz has more points than in rviz. You should get something similar in rviz with MapCloud rviz display. You can then adjust the decimation value (default 4) to show more or less points.

khuechuong, 2023-01-23:
Is zed mini a TOF cameras?

matlabbe, 2023-01-23:
No, it is stereo

matlabbe, 2023-01-23:
Set cloud decimation to 1, and voxel to 0 to get all points.

khuechuong, 2023-01-23:
when I set the cloud vortex size to 0, the whole map stop working.

matlabbe, 2023-01-23:
The problem is that converting depth to point cloud without any decimation and voxel downsampling requires quite a lot of rendering power. You may increase decimation to 2 or 4 if you set voxel size to 0. In rtabmapviz, by default the depth image is created with decimation 4 without voxel filtering.

matlabbe, 2023-01-29:
Just tested with z ZED2 and default zed_rtabmap.launch, and it works as expected. Here comparison between rtabmapviz and rviz (using MapCloud plugin with default values):
[image]

khuechuong, 2023-01-29:
The accuracy is less than I thought. I thought it would be more solid. Also I have a zed mini, I don't know if there are performance difference between the 2

Myzhar, 2023-01-29:
@khuechuong what depth mode and resolution are you using?
Have you tried to set `depth_quality` to 4 (NEURAL)?

khuechuong, 2023-01-29:
@Myzhar I used 1 but I'll try 4 and see