introlab/rtabmap_ros #875: LeGO_LOAM integration with rtabmap

ijklmnopqr, 2023-01-26:
Hello,

Rtabmap is really an outstanding work. I really appreciated it.

Currently I am trying to integrate LeGO_LOAM with rtabmap using Velodyne VLP-16 and realsense D435 RGB-D camera. I want to compare the results generated from LeGO_LOAM, rtabmap and integrated result. To do so what change should I make in rtabmap package? Is there any working reference for LOAM and FLOAM using rtabmap. What files need to be changed to make it compatible with Velodyne VLP-16 and realsense D435, which launch file to be changed and what parameters to be given.

Your input will be of significant help.

Looking forward to your reply.

matlabbe, 2023-01-29:
you may look at this example: https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/test_velodyne_d435i_deskewing.launch

Note that in this example, the IMU from D435i is used to deskew the lidar. Otherwise the D435 is not used in odometry estimation, which is done using LiDAR only by rtabmap's icp odometry or FLOAM depending on the config used. If floam option is enabled, rtabmap library [should be built with FLOAM](https://github.com/introlab/rtabmap_ros/blob/46c2988ee9727344481fc0e5db79336ab32dddaf/launch/tests/test_velodyne_d435i_deskewing.launch#L42). FLOAM and LOAM odometry inside rtabmap can be found here: 
 * https://github.com/introlab/rtabmap/blob/master/corelib/src/odometry/OdometryLOAM.cpp
 * https://github.com/introlab/rtabmap/blob/master/corelib/src/odometry/OdometryFLOAM.cpp
 
 Note that another approach could be to use LeGO_LOAM ROS package directly, then feed its output odometry to rtabmap node subscribing to both lidar and D435.

ijklmnopqr, 2023-02-01:
Thank you for your response.

For the start I was trying to fuse D435 with Velodyne VLP 16. I have successfully checked the working of VLP16 with test_velodyne.launch. Which uses ICP as odometry am I right? 
I tried to integrate D435 with test_velodyne.launch but haven't had any luck. I have also referred to your [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf) STM section and tried to replicate it but I am doing something wrong. I can only see point clouds from Lidar and no image from camera on rtabmapviz. Also, I am not able to see/generate maps with my launch file. 
Below is test_velodyne.launch file in which I have tried to integrate D435 topics:
<?xml version="1.0"?>

<launch>

    

    <arg name="rtabmapviz"    default="true"/>
    <arg name="use_imu"       default="false"/> 
    <arg name="imu_topic"     default="/imu/data"/>
    <arg name="scan_20_hz"    default="false"/> 
    <arg name="deskewing"     default="true"/>
    <arg name="slerp"         default="false"/> 
    <arg name="organize_cloud" default="$(arg deskewing)"/> 
    <arg name="scan_topic"    default="/velodyne_points"/>
    <arg name="use_sim_time"  default="false"/>
    <param if="$(arg use_sim_time)" name="use_sim_time" value="true"/>

    <arg name="frame_id"        default="velodyne"/>
    <arg name="queue_size"      default="100"/> 
    <arg name="queue_size_odom" default="100"/>  
    <arg name="loop_ratio"      default="0.2"/>

    <arg name="resolution"         default="0.3"/> 
    <arg name="iterations"         default="10"/>

    
    <arg name="ground_is_obstacle" default="true"/>
    <arg name="grid_max_range"     default="20"/>

    
    <arg name="ground_normals_up" default="true"/> 
    <arg name="local_map_size"    default="15000"/>
    <arg name="key_frame_thr"     default="0.6"/>

    
    <arg name="floam"    default="false"/> 
    <arg name="floam_sensor" default="0"/> 

    
    <include unless="$(arg use_sim_time)" file="$(find velodyne_pointcloud)/launch/VLP16_points.launch">
       <arg     if="$(arg scan_20_hz)" name="rpm" value="1200"/>
       <arg unless="$(arg scan_20_hz)" name="rpm" value="600"/>
       <arg name="organize_cloud" value="$(arg organize_cloud)"/>
    </include>

    
    <include unless="$(arg use_sim_time)" file="$(find realsense2_camera)/launch/rs_camera.launch">
       <arg name="align_depth"     value="true"/>
    </include>

    
    <arg name="rgb_topic"               default="/camera/color/image_raw" />
    <arg name="depth_topic"             default="/camera/aligned_depth_to_color/image_raw" />
    <arg name="camera_info_topic"       default="/camera/color/camera_info" />
    <arg name="depth_camera_info_topic" default="$(arg camera_info_topic)" />

    
    <node unless="$(arg use_sim_time)" pkg="tf" type="static_transform_publisher" name="velodyne_to_camera_tf" args="0 0 0 0 0 0 velodyne camera_link 100"/>

    
    <node if="$(arg use_imu)" pkg="nodelet" type="nodelet" name="imu_nodelet_manager" args="manager">
      <remap from="imu/data_raw" to="$(arg imu_topic)"/>
      <remap from="imu/data" to="$(arg imu_topic)/filtered"/>
    </node>
    <node if="$(arg use_imu)" pkg="nodelet" type="nodelet" name="imu_filter" args="load imu_filter_madgwick/ImuFilterNodelet imu_nodelet_manager">
[... 123 lines cut ...]
        <remap unless="$(arg deskewing)" from="cloud" to="$(arg scan_topic)"/>
        <remap from="odom"            to="odom"/>
        <param     if="$(arg scan_20_hz)" name="max_clouds"      type="int"    value="20" />
        <param unless="$(arg scan_20_hz)" name="max_clouds"      type="int"    value="10" />
        <param name="fixed_frame_id"  type="string" value="" />
        <param name="queue_size"      type="int"    value="$(arg queue_size)" />
      </node>
  </group>

</launch>

After launching the file, I also get below warning:
`/rtabmap/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmap subscribed to (approx sync):
   /rtabmap/odom \
   /rtabmap/rgbd_image \
   /rtabmap/assembled_cloud
`

And this is rqt graph:

[image]

Can you please let me know what changes need to be made to launch file?
Looking forward to your help.

matlabbe, 2023-02-05:
Yes, with velodyne example, icp odometry is used.

For D435 integration, rtabmap seems subscribed to right topics, but one of them is not received. Check:
```
rostopic hz /rtabmap/odom  /rtabmap/rgbd_image  /rtabmap/assembled_cloud 
```

ijklmnopqr, 2023-02-09:
Hello, 
Thank you for your response.

I have checked the above topics they are received now:

`============================================================================
/rtabmap/odom              7.438    0.06866     0.1739      0.02662   39    
/rtabmap/rgbd_image        26.25    0.004028    0.2209      0.02421   39    
/rtabmap/assembled_cloud   0.6889   1.411       1.492       0.04028   129   

         topic              rate    min_delta   max_delta   std_dev   window
============================================================================
/rtabmap/odom              7.382    0.06866     0.1739      0.02519   46    
/rtabmap/rgbd_image        26.23    0.004028    0.2209      0.02397   46    
/rtabmap/assembled_cloud   0.7285   1.215       1.492       0.1163    156   

         topic              rate    min_delta   max_delta   std_dev   window
============================================================================
/rtabmap/odom              7.322    0.06866     0.1739      0.02397   53    
/rtabmap/rgbd_image        25.75    0.004028    0.2209      0.02449   53    
/rtabmap/assembled_cloud   0.7257   1.215       1.492       0.1012    179   
`

I am still not able to integrate D435 and VLP-16 LiDAR. Can you please verify the above launch file for its correctness?

Below is the rqt graph:

[image]

matlabbe, 2023-04-15:
Based on the frame rates, `queue_size` of 10 is too small. You should start rtabmap node with `queue_size` of at least 30