introlab/rtabmap_ros #876: ROS2 - Octomap can't be saved using Octomap Saver Node

AddyPete, 2023-01-27:
Hi I successfully created Octomap with Kinect 360 using Rtabmap RGBD Odometry.

[image]

I want to save this octomap using the [ros2 octomap saver node](https://github.com/OctoMap/octomap_mapping/tree/ros2) but it doesn't work. It always looping with a message "Waiting for Service...."

[image]

I used this command to try to save the map: **ros2 run octomap_server octomap_saver_node --ros-args -p octomap_path:=map.bt**

The Octomap topic shown in rviz is /rtabmap/octomap_binary

Thank you!

matlabbe, 2023-01-29:
It should work:
```
ros2 run octomap_server octomap_saver_node --ros-args -p octomap_path:=map.bt 
[INFO] [1674957009.554208229] [octomap_saver]: Map received (3516 nodes, 0.050000 m res), saving to map.bt
```

by default the service name is `/octomap_binary`.  If rtabmap is in `rtabmap` namespace, you have to remap the service name:
```
ros2 run octomap_server octomap_saver_node --ros-args -p octomap_path:=map.bt -r octomap_binary:=/rtabmap/octomap_binary
```

To list all services:
```
ros2 service list | grep octomap
```

AddyPete, 2023-01-30:
Hello I did the command

**ros2 run octomap_server octomap_saver_node --ros-args -p octomap_path:=map.bt -r octomap_binary:=/rtabmap/octomap_binary**

But it is still Waiting for Service.........

[image]

This is what it looks like in RTABMap-Viz

[image]

and lastly, this are the service list outputted using **ros2 service list | grep octomap**

[image]

I'm using ROS2 Foxy

Hopefully I gave you enough information. Thank you

matlabbe, 2023-04-15:
Other option is to start octomap_saver_node in ROS namespace.
```
ros2 run octomap_server octomap_saver_node --ros-args --remap  __ns:=/rtabmap -p octomap_path:=map.bt 
```

2805651606, 2024-10-21:
你好，我的基础版本是ros2 humble,使用的d435i相机，运用的算法是vins-fushion，请问也可以利用这个算法建造三维地图？

matlabbe, 2024-10-26:
> 你好，我的基础版本是ros2 humble,使用的d435i相机，运用的算法是vins-fushion，请问也可以利用这个算法建造三维地图？

Hopefully google translate works:

> Hello, my basic version is ros2 humble, I use a d435i camera, and the algorithm used is vins-fusion. Can I also use this algorithm to build a three-dimensional map?

It is possible to generate an Octomap with D435i camera form rtabmap node. You can feed vins-fusion output to rtabmap as odometry, then rtabmap will generate the 3D OctoMap.