introlab/rtabmap_ros #880: stereo_odometry error

maekjuplease, 2023-01-29:
Dear Matlabbe,

As I wanted to use internal VINS-Fusion in RTABMAP to generate the odom and then fuse with the 2d lidar data to do the localization for my robot, but when I ran the test_d435i.launch file it always appeared the problem from the /rtabmap_ros/stereo_odometry. Could you please tell me how to fix this problem. Thank you.
[image]

matlabbe, 2023-01-29:
The code is crashing. How did you build VINS-Fusion? Which version? I have a patch here https://gist.github.com/matlabbe/795ab37067367dca58bbadd8201d986c#file-vins-fusion_be55a93_pull136-patch to apply on VINS-fusion repo commit be55a93 + pull request 136.

Otherwise, to see better where it crashes, you would need to start the node in gdb, see also section A here: https://github.com/introlab/rtabmap_ros/issues/28

maekjuplease, 2023-02-01:
Thank you for your response I did exactly the same but it is still crashing in the code. I'll try to reinstall once again, thank you.