introlab/rtabmap_ros #882: node crashing right at the beginning

cy-2022, 2023-02-02:
Hello~~I'd like to ask about an issue we haven't managed to resolve yet. In our case, building and running rtabmap alone without ROS worked fine. With ZED2, mapping looks all normal. 
Building rtabmap_ros looked successful without an error, while there were warnings regarding possible conflicts of different versions of boost, etc.. 
[image]

However, when we tried to start rtabmap via roslaunch..., it crashed right at the beginning. So is this issue very likely to result from the existence of different versions of libraries? Thank you very much!

[image]

cy-2022, 2023-02-04:
Indeed it was due to the existence of different versions of boost, and now this specific issue has been solved. However, as I tested with one of the TUM datasets, the "malloc.c"-related error occurred. 
[image]

matlabbe, 2023-02-05:
Thread following here: https://github.com/introlab/rtabmap_ros/issues/881#issuecomment-1416726827

closed 2023-02-05 by matlabbe