introlab/rtabmap_ros #886: RGBD sync (as well as rtabmap) crashes on launch with no config at all.

peredwardsson, 2023-02-14:
ROS2 Humble
Jetson AGX 
Ubuntu 20
`rtabmap_ros` commit: 39e1a88b94b36ae59a774bb8f57b2c04a201a448
`rtabmap` commit: 549fc2a592b46f2e8d96cc2de06de56a39dc2c6d
All other nodes can run just fine
```
# echo $RMW_IMPLEMENTATION 
rmw_fastrtps_cpp
root@hugo-rally:/src/current_ws# ros2 run rtabmap_ros rgbd_sync 
[INFO] [1676363080.970082518] [rgbd_sync]: rgbd_sync: approx_sync = true
[INFO] [1676363080.971957501] [rgbd_sync]: rgbd_sync: approx_sync_max_interval = 0.000000
[INFO] [1676363080.972016511] [rgbd_sync]: rgbd_sync: queue_size  = 10
[INFO] [1676363080.972038879] [rgbd_sync]: rgbd_sync: qos         = 0
[INFO] [1676363080.972055199] [rgbd_sync]: rgbd_sync: qos_camera_info = 0
[INFO] [1676363080.972070048] [rgbd_sync]: rgbd_sync: depth_scale = 1.000000
[INFO] [1676363080.972084864] [rgbd_sync]: rgbd_sync: decimation = 1
[INFO] [1676363080.972099200] [rgbd_sync]: rgbd_sync: compressed_rate = 0.000000
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  could not create publisher: Type support not from this implementation. Got:
    Handle's typesupport identifier (rosidl_typesupport_cpp) is not supported by this library, at /opt/ros/humble/src/rosidl_typesupport/rosidl_typesupport_cpp/src/type_support_dispatch.hpp:111
    Could not load library librtabmap_ros__rosidl_typesupport_fastrtps_cpp.so: dlopen error: librtabmap_ros__rosidl_typesupport_fastrtps_cpp.so: cannot open shared object file: No such file or directory, at /opt/ros/humble/src/rcutils/src/shared_library.c:99, at /opt/ros/humble/src/rosidl_typesupport/rosidl_typesupport_cpp/src/type_support_dispatch.hpp:76
while fetching it, at /opt/ros/humble/src/rmw_fastrtps/rmw_fastrtps_cpp/src/publisher.cpp:119, at /opt/ros/humble/src/rcl/rcl/src/rcl/publisher.c:116
[ros2run]: Aborted
```

Didn't work with cyclonedds either. Rtabmap_ros from about last week worked fine.

matlabbe, 2023-02-14:
Thank you for reporting, I was updating last weekend to use cmake targets, and got problems with some targets with same name (rtabmap_ros) so I changed the messages to rtabmap_msgs prefix, which looks wrong. Reverting back to rtabmap_ros seems okay. Don't know why I had multiple targets with same name error previously... got some difficulties to correctly set cmake targets because the ROS2 documentation changed over time. The cmake now contains a mix of multiple ROS2 cmake examples, and seem working now!

matlabbe, 2023-02-14:
Confirmed that is fixed on Humble:
```
$ docker run -it --rm introlab3it/rtabmap_ros:humble-latest 

root@0840c5d8c193:~# ros2 launch rtabmap_ros rtabmap.launch.py rgbd_sync:=true rtabmapviz:=false
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-02-14-18-23-25-963228-0840c5d8c193-30
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rgbd_sync-1]: process started with pid [31]
[INFO] [rgbd_odometry-2]: process started with pid [33]
[INFO] [rtabmap-3]: process started with pid [35]
[rgbd_sync-1] [INFO] [1676399006.370081109] [rtabmap.rgbd_sync]: rgbd_sync: approx_sync = true
[rgbd_sync-1] [INFO] [1676399006.370179556] [rtabmap.rgbd_sync]: rgbd_sync: approx_sync_max_interval = 0.000000
[rgbd_sync-1] [INFO] [1676399006.370195204] [rtabmap.rgbd_sync]: rgbd_sync: queue_size  = 10
[rgbd_sync-1] [INFO] [1676399006.370200751] [rtabmap.rgbd_sync]: rgbd_sync: qos         = 1
[rgbd_sync-1] [INFO] [1676399006.370206215] [rtabmap.rgbd_sync]: rgbd_sync: qos_camera_info = 1
[rgbd_sync-1] [INFO] [1676399006.370212555] [rtabmap.rgbd_sync]: rgbd_sync: depth_scale = 1.000000
[rgbd_sync-1] [INFO] [1676399006.370218111] [rtabmap.rgbd_sync]: rgbd_sync: decimation = 1
[rgbd_sync-1] [INFO] [1676399006.370222986] [rtabmap.rgbd_sync]: rgbd_sync: compressed_rate = 0.000000
[rgbd_sync-1] [INFO] [1676399006.380673046] [rtabmap.rgbd_sync]: 
[rgbd_sync-1] rgbd_sync subscribed to (approx sync):
[rgbd_sync-1]    /camera/rgb/image_rect_color,
[rgbd_sync-1]    /camera/depth_registered/image_raw,
[rgbd_sync-1]    /camera/rgb/camera_info
[rgbd_odometry-2] [INFO] [1676399006.390883533] [rtabmap.rgbd_odometry]: Odometry: frame_id               = base_link
[rgbd_odometry-2] [INFO] [1676399006.391056249] [rtabmap.rgbd_odometry]: Odometry: odom_frame_id          = odom
[rgbd_odometry-2] [INFO] [1676399006.391065033] [rtabmap.rgbd_odometry]: Odometry: publish_tf             = true
[rgbd_odometry-2] [INFO] [1676399006.391068575] [rtabmap.rgbd_odometry]: Odometry: wait_for_transform     = 0.200000
[rgbd_odometry-2] [INFO] [1676399006.391089530] [rtabmap.rgbd_odometry]: Odometry: log_to_rosout_level    = 4
[rgbd_odometry-2] [INFO] [1676399006.391146069] [rtabmap.rgbd_odometry]: Odometry: initial_pose           = xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000
[rgbd_odometry-2] [INFO] [1676399006.391151335] [rtabmap.rgbd_odometry]: Odometry: ground_truth_frame_id  = 
[rgbd_odometry-2] [INFO] [1676399006.391154986] [rtabmap.rgbd_odometry]: Odometry: ground_truth_base_frame_id = 
[rgbd_odometry-2] [INFO] [1676399006.391158943] [rtabmap.rgbd_odometry]: Odometry: config_path            = 
[rgbd_odometry-2] [INFO] [1676399006.391162099] [rtabmap.rgbd_odometry]: Odometry: publish_null_when_lost = true
[rgbd_odometry-2] [INFO] [1676399006.391165830] [rtabmap.rgbd_odometry]: Odometry: guess_frame_id         = 
[rgbd_odometry-2] [INFO] [1676399006.391169648] [rtabmap.rgbd_odometry]: Odometry: guess_min_translation  = 0.000000
[rgbd_odometry-2] [INFO] [1676399006.391174747] [rtabmap.rgbd_odometry]: Odometry: guess_min_rotation     = 0.000000
[rgbd_odometry-2] [INFO] [1676399006.391180475] [rtabmap.rgbd_odometry]: Odometry: guess_min_time         = 0.000000
[rgbd_odometry-2] [INFO] [1676399006.391185072] [rtabmap.rgbd_odometry]: Odometry: expected_update_rate   = 0.000000 Hz
[rgbd_odometry-2] [INFO] [1676399006.391189351] [rtabmap.rgbd_odometry]: Odometry: max_update_rate        = 0.000000 Hz
[rgbd_odometry-2] [INFO] [1676399006.391193039] [rtabmap.rgbd_odometry]: Odometry: min_update_rate        = 0.000000 Hz
[rgbd_odometry-2] [INFO] [1676399006.391196491] [rtabmap.rgbd_odometry]: Odometry: wait_imu_to_init       = false
[rgbd_odometry-2] [INFO] [1676399006.391237003] [rtabmap.rgbd_odometry]: Odometry: stereoParams_=0 visParams_=1 icpParams_=0
[rgbd_odometry-2] [INFO] [1676399006.397799773] [rtabmap.rgbd_odometry]: RGBDOdometry: approx_sync    = false
[rgbd_odometry-2] [INFO] [1676399006.397824881] [rtabmap.rgbd_odometry]: RGBDOdometry: queue_size     = 10
[rgbd_odometry-2] [INFO] [1676399006.397830127] [rtabmap.rgbd_odometry]: RGBDOdometry: qos            = 1
[rgbd_odometry-2] [INFO] [1676399006.397834361] [rtabmap.rgbd_odometry]: RGBDOdometry: qos_camera_info = 1
[rgbd_odometry-2] [INFO] [1676399006.397837869] [rtabmap.rgbd_odometry]: RGBDOdometry: subscribe_rgbd = true
[rgbd_odometry-2] [INFO] [1676399006.397841670] [rtabmap.rgbd_odometry]: RGBDOdometry: rgbd_cameras   = 1
[rgbd_odometry-2] [INFO] [1676399006.397844903] [rtabmap.rgbd_odometry]: RGBDOdometry: keep_color     = false
[rgbd_odometry-2] [INFO] [1676399006.398023013] [rtabmap.rgbd_odometry]: 
[rgbd_odometry-2] rgbd_odometry subscribed to:
[rgbd_odometry-2]    /rtabmap/rgbd_image
[rtabmap-3] [INFO] [1676399006.404474392] [rtabmap.rtabmap]: rtabmap(maps): map_filter_radius          = 0.000000
[rtabmap-3] [INFO] [1676399006.404604861] [rtabmap.rtabmap]: rtabmap(maps): map_filter_angle           = 30.000000
[rtabmap-3] [INFO] [1676399006.404614583] [rtabmap.rtabmap]: rtabmap(maps): map_cleanup                = true
[rtabmap-3] [INFO] [1676399006.404619953] [rtabmap.rtabmap]: rtabmap(maps): map_always_update          = false
[rtabmap-3] [INFO] [1676399006.404625027] [rtabmap.rtabmap]: rtabmap(maps): map_empty_ray_tracing      = true
[rtabmap-3] [INFO] [1676399006.404631093] [rtabmap.rtabmap]: rtabmap(maps): cloud_output_voxelized     = true
[... cut ...]
[rgbd_sync-1] rgbd_sync subscribed to (approx sync):
[rgbd_sync-1]    /camera/rgb/image_rect_color,
[rgbd_sync-1]    /camera/depth_registered/image_raw,
[rgbd_sync-1]    /camera/rgb/camera_info
[rgbd_odometry-2] [WARN] [1676399011.398241397] [rtabmap.rgbd_odometry]: rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. 
[rgbd_odometry-2] rgbd_odometry subscribed to:
[rgbd_odometry-2]    /rtabmap/rgbd_image
```

closed 2023-02-14 by matlabbe

peredwardsson, 2023-02-14:
Great, will try at a later point. Speaking of which, it would be nice of you to provide your custom messages in a separate package so that I can build just the messages for visualisation purposes. Would be nice.

matlabbe, 2023-02-14:
Yes, I already thought about this, splitting `rtabmap_ros` into 
* `rtabmap_ros` (core nodes and library without UI), 
* `rtabmap_msgs`, 
* `rtabmap_rviz_plugins` (rviz displays), 
* `rtabmap_costmap_plugins` (costmap related plugins), 
* `rtabmap_viz` (UI stuff)... 

The problem right now is that my ros2 branch is still depending on the ros1 master branch to avoid diverging branches (so new patches/features added in ros1 can be ported to ros2 with a Merge). Adding submodules would require to change the github layout, and I am afraid that would be a painful merge to make. I created an issue about that: https://github.com/introlab/rtabmap_ros/issues/888