introlab/rtabmap_ros #887: rosbag zed mini for rtabmap doesn't show in rviz

khuechuong, 2023-02-14:
So I saved the zed mini data with rosbag. Here's the rosbag info:

types:       nav_msgs/Odometry       [cd5e73d190d741a2f92e81eda573aca7]
             sensor_msgs/CameraInfo  [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/Image       [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181]
             tf2_msgs/TFMessage      [94810edda583a504dfda3829e70d7eec]

topics:      /tf                                           76210 msgs    : tf2_msgs/TFMessage     
             /tf_static                                        3 msgs    : tf2_msgs/TFMessage      (3 connections)
             /zedm/zed_node/depth/camera_info               2824 msgs    : sensor_msgs/CameraInfo 
             /zedm/zed_node/depth/depth_registered          2824 msgs    : sensor_msgs/Image      
             /zedm/zed_node/odom                            2838 msgs    : nav_msgs/Odometry      
             /zedm/zed_node/point_cloud/cloud_registered    1892 msgs    : sensor_msgs/PointCloud2
             /zedm/zed_node/rgb/camera_info                 5650 msgs    : sensor_msgs/CameraInfo 
             /zedm/zed_node/rgb/image_rect_color            2825 msgs    : sensor_msgs/Image

Here's my rosbag and rtabmap launch file (I just modified the zed_rtabmap.launch that's already given a little bit):
<launch>
    <arg name="svo_file"             default="" /> 
    <arg name="stream"               default="" /> 

    <arg name="zed_node_name"        default="zed_node" />
    <arg name="camera_model"         default="zed2" />
    <arg name="publish_urdf"         default="true" />
    <arg name="camera_name"          default="zed" />

    <arg name="base_frame"           default="base_link" />
    <node pkg="rosbag" type="play" name="player" output="screen" args="--clock /home/ara/test_2023-02-09-17-21-48.bag"/>
    <group ns="$(arg camera_name)">
  
        
        <arg name="custom_rtabmap_launch_file" default="$(find zed_rtabmap_example)/launch/include/sl_rtabmap.launch.xml"/>
        <include file="$(arg custom_rtabmap_launch_file)">
            <arg name="odom_topic"              default="/zedm/zed_node/odom" />
            <arg name="rgb_topic"               default="/zedm/zed_node/rgb/image_rect_color" />
            <arg name="depth_topic"             default="/zedm/zed_node/rgb/image_rect_color" />
            <arg name="camera_info_topic"       default="/zedm/zed_node/rgb/camera_info" />
            <arg name="depth_camera_info_topic" default="/zedm/zed_node/depth/camera_info" />
            <arg name="cloud_topic"             default="/zedm/zed_node/point_cloud/cloud_registered" />
        </include>   

        
        <node name="rviz" pkg="rviz" type="rviz" args="-d $(find zed_rtabmap_example)/rviz/zed-rtabmap.rviz" output="screen" /> 
    </group>
</launch>

I've screenshotted what I got. Essentially, while the 3D mapping showed up in rtabmapviz, nothing shows up on rviz. The pointcloud in rviz was just me adding the pointcloud topic /zedm/zed_node/point_cloud/cloud_registered  from zedm. The 2 errors I saw is :

[ WARN] [1676406986.150804577]: We cannot estimated the baseline of the rectified images with tf! (zedm_left_camera_optical_frame->zedm_left_camera_optical_frame = xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000)

and 

[ERROR] (2023-02-14 12:36:26.400) util3d.cpp:1231::cloudRGBFromSensorData() Stereo camera model 0 is invalid

I don't understand why. 

[image]

adeldennaoui, 2023-04-10:
Hello. Did you succeed in solving this problem? I have the exact same error.

matlabbe, 2023-04-16:
In rviz, you could click on "From Scan" checkbox under MapCloud display.

This:
```xml
        <arg name="rgb_topic"               default="/zedm/zed_node/rgb/image_rect_color" />
        <arg name="depth_topic"             default="/zedm/zed_node/rgb/image_rect_color" />
```
looks pretty wrong. Set the right depth topic. It seems rtabmap is assuming that you are using stereo data as it received both RGB images. Set `depth_topic` to a depth topic so that rtabmap can assume it is RGB-D data.