introlab/rtabmap_ros #888: Split rtabmap_ros into sub-packages

matlabbe, 2023-02-14:
Taken from this discussion https://github.com/introlab/rtabmap_ros/issues/886, check if it would feasible to make the following changes in ros1 and ros2 branches (ideally do it in ros1 branch then merge back in ros2 branch).

Examples of sub-packages:
* `rtabmap_ros` (core nodes and library without UI), 
* `rtabmap_launch` (for main rtabmap.launch file),
* `rtabmap_msgs`, 
* `rtabmap_srvs`, 
* `rtabmap_rviz_plugins` (rviz displays), 
* `rtabmap_costmap_plugins` (costmap related plugins), 
* `rtabmap_viz` (UI stuff, mainly `rtabmapviz` node),
* `rtabmap_demo` (for demo launch files),
* `rtabmap_examples` (for tests/examples launch files)
* `rtabmap_legacy` (for all launch files not really used anymore or too specific, e.g., azimut3 related files or deprecated launch files)

peredwardsson, 2023-02-15:
If you are looking for input, keep in mind that the convention for ros2 is to stick all your interfaces into `rtabmap_interfaces` which I am not sure is the case for ros1.

matlabbe, 2023-02-15:
Thank you to point it out! Some references I've found about that, though they are more related to ROS1:
* https://ros.org/reps/rep-0144.html
* http://wiki.ros.org/ROS/Patterns/Conventions#Packages

From a quick search, I didn't find similar pages for ROS2, where is the `_interfaces` naming described? I don't see that many `_interfaces` packages on official repo: http://repo.ros2.org/status_page/foxy_default.html?q=interface. Looks like `_srvs` is not used either, probably everyone put their msgs and srvs in same `_msgs' suffix (which makes sense to void duplicating cmake code). I added a note in my original post to look in http://repo.ros2.org/status_page/foxy_default.html and http://repositories.ros.org/status_page/noetic_default.html to see which ones are used most often.

peredwardsson, 2023-02-16:
Yeah, I thought I found somewhere that `_interfaces` was correct for ROS2, but common use seems to dictate that `.msg` and `.srv` files (and possibly also `.action`) go in `_msgs`.

matlabbe, 2023-02-20:
Work in progress: https://github.com/introlab/rtabmap_ros/tree/rtabmap_ros_pkg_split

closed 2023-03-05 by matlabbe