introlab/rtabmap_ros #889: Error in compiling rtabmap_ros: undefined reference to `g2o::csparse::CSparse::solve(double*, double*) const'

wz8530, 2023-02-15:
Hi, the following problems occurred when compiling rtabmap_ros using `catkin_make`:
[ 74%] Built target rtabmap_ros_generate_messages
[ 74%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_camera.dir/src/CameraNode.cpp.o
[ 75%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/MapsManager.cpp.o
[ 75%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/MsgConversion.cpp.o
[ 75%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/OdometryROS.cpp.o
[ 75%] Building CXX object rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/PluginInterface.cpp.o
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::csparse::CSparse::solve(double*, double*) const'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::csparse::CSparse::CSparse()'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::csparse::CSparse::hasSymbolic() const'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::csparse::CSparse::sparseView()'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::cholmod::Cholmod::freeFactor()'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::cholmod::Cholmod::sparseView()'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::csparse::CSparse::amd(g2o::csparse::CSparse::SparseView const&, Eigen::Matrix<int, -1, 1, 0, -1, 1>&)'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::csparse::CSparse::analyze_p(int*)'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::cholmod::Cholmod::analyze_p(int*)'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::cholmod::Cholmod::choleskyNz() const'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::csparse::CSparse::writeSparse(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::csparse::CSparse::freeFactor()'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::cholmod::Cholmod::solve(double*, double*) const'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::csparse::CSparse::choleskyNz() const'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::csparse::CSparse::~CSparse()'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::SBACam::setKcam(double, double, double, double, double)'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::Factory::registerType(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<g2o::AbstractHyperGraphElementCreator> const&)'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::cholmod::Cholmod::Cholmod()'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::csparse::CSparse::factor()'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::EdgeSE3XYZPrior::EdgeSE3XYZPrior()'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::csparse::CSparse::freeSymbolic()'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::cholmod::Cholmod::factor()'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::cholmod::Cholmod::simplifyFactor()'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::csparse::CSparse::factorize()'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::cholmod::Cholmod::amd(g2o::cholmod::Cholmod::SparseView&, int*)'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::cholmod::Cholmod::hasFactor() const'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::SBACam::SBACam(Eigen::Quaternion<double, 0> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::cholmod::Cholmod::~Cholmod()'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::csparse::CSparse::analyze()'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::cholmod::Cholmod::analyze()'
/home/wz/project/ws_rtabmap-0.21.0/devel/lib/librtabmap_core.so.0.21.0: undefined reference to `g2o::cholmod::Cholmod::factorize()'
collect2: error: ld returned 1 exit status
rtabmap_ros/CMakeFiles/rtabmap_rgbdicp_odometry.dir/build.make:862: recipe for target '/home/wz/project/ws_rtabmap-0.21.0/devel/lib/rtabmap_ros/rgbdicp_odometry' failed
make[2]: *** [/home/wz/project/ws_rtabmap-0.21.0/devel/lib/rtabmap_ros/rgbdicp_odometry] Error 1
CMakeFiles/Makefile2:3070: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_rgbdicp_odometry.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_rgbdicp_odometry.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

system env:
1. Ubuntu 18.04
2. ROS melodic
3. OpenCV 3.2.0 (default)
4. g2o (https://github.com/RainerKuemmerle/g2o)
5. RTABMap v0.21.0

rtabmap cmake Configuring, and Compiled successfully:
$ cmake -DCMAKE_INSTALL_PREFIX=~/project/ws_rtabmap-0.21.0/devel -DWITH_CERES=ON -DWITH_CVSBA=0N ..
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- MOBILE_BUILD=OFF
-- Checking for module 'eigen3'
[... 194 lines cut ...]
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With ZED                  = NO (ZED sdk and/or cuda not found)
--   With ZEDOC                = NO (ZED Open Capture not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2 2.50.0    = YES (License: Apache-2)
--   With MyntEyeS             = NO (mynteye s sdk not found)
--   With DepthAI              = NO (WITH_DEPTHAI=OFF)
-- 
--  Odometry Approaches:
--   With loam_velodyne        = NO (WITH_LOAM=OFF)
--   With floam                = NO (WITH_FLOAM=OFF)
--   With libfovis             = NO (WITH_FOVIS=OFF)
--   With libviso2             = NO (WITH_VISO2=OFF)
--   With dvo_core             = NO (WITH_DVO=OFF)
--   With okvis                = NO (WITH_OKVIS=OFF)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With VINS-Fusion          = NO (WITH_VINS=OFF)
--   With OpenVINS             = NO (WITH_OPENVINS=OFF)
--   With ORB_SLAM             = NO (WITH_ORB_SLAM=OFF)
-- Show all options with: cmake -LA | grep WITH_
-- --------------------------------------------
-- Configuring done
-- Generating done

Can you give me some suggestions?

wz8530, 2023-02-15:
Through the `ldd xxx` command, it is found that the librtabmap_core.so.0.21.0 is linked to the ros-melodic-libg2o. 
Uninstalling the g2o library can solve this problem.
`$ sudo apt-get purge ros-melodic-libg2o`

closed 2023-02-15 by wz8530

dolikemedo, 2024-05-02:
Hi!

I had a similar issue with **gtsam**. 
I'm building on ros:noetic-ros-base for a deployable slam container.

```
[ 83%] Building CXX object rtabmap_ros/rtabmap_util/CMakeFiles/rtabmap_util_plugins.dir/src/nodelets/disparity_to_depth.cpp.o
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Point2::equals(gtsam::Point2 const&, double) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Pose3::range(gtsam::Point3 const&, gtsam::OptionalJacobian<1, 6>, gtsam::OptionalJacobian<1, 3>) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Point3::print(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Point2::print(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::fpEqual(double, double, double)'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Rot2::rotate(gtsam::Point2 const&, gtsam::OptionalJacobian<2, 1>, gtsam::OptionalJacobian<2, 2>) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Pose2::bearing(gtsam::Point2 const&, gtsam::OptionalJacobian<1, 3>, gtsam::OptionalJacobian<1, 2>) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Pose3::bearing(gtsam::Point3 const&, gtsam::OptionalJacobian<2, 6>, gtsam::OptionalJacobian<2, 3>) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Rot3::xyz() const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Pose2::range(gtsam::Point2 const&, gtsam::OptionalJacobian<1, 3>, gtsam::OptionalJacobian<1, 2>) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Rot3::RzRyRx(double, double, double)'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Point3::equals(gtsam::Point3 const&, double) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Rot3::operator*(gtsam::Point3 const&) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::NoiseModelFactor::print(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, boost::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (unsigned long)> const&) const'
collect2: error: ld returned 1 exit status
make[2]: *** [rtabmap_ros/rtabmap_util/CMakeFiles/rtabmap_map_optimizer.dir/build.make:639: /opt/husky/robo_ws/devel/lib/rtabmap_util/map_optimizer] Error 1
[ 84%] Building CXX object rtabmap_ros/rtabmap_util/CMakeFiles/rtabmap_util_plugins.dir/src/nodelets/pointcloud_to_depthimage.cpp.o
make[1]: *** [CMakeFiles/Makefile2:5257: rtabmap_ros/rtabmap_util/CMakeFiles/rtabmap_map_optimizer.dir/all] Error 2
[ 84%] Building CXX object rtabmap_ros/rtabmap_util/CMakeFiles/rtabmap_util_plugins.dir/src/nodelets/obstacles_detection.cpp.o
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Point2::equals(gtsam::Point2 const&, double) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Pose3::range(gtsam::Point3 const&, gtsam::OptionalJacobian<1, 6>, gtsam::OptionalJacobian<1, 3>) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Point3::print(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Point2::print(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::fpEqual(double, double, double)'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Rot2::rotate(gtsam::Point2 const&, gtsam::OptionalJacobian<2, 1>, gtsam::OptionalJacobian<2, 2>) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Pose2::bearing(gtsam::Point2 const&, gtsam::OptionalJacobian<1, 3>, gtsam::OptionalJacobian<1, 2>) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Pose3::bearing(gtsam::Point3 const&, gtsam::OptionalJacobian<2, 6>, gtsam::OptionalJacobian<2, 3>) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Rot3::xyz() const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Pose2::range(gtsam::Point2 const&, gtsam::OptionalJacobian<1, 3>, gtsam::OptionalJacobian<1, 2>) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Rot3::RzRyRx(double, double, double)'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Point3::equals(gtsam::Point3 const&, double) const'
/usr/bin/ld: /usr/local/lib/librtabmap_core.so.0.21.4: undefined reference to `gtsam::Rot3::operator*(gtsam::Point3 const&) const'
[... cut ...] el-grid
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  ros-noetic-gtsam
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 11.0 MB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 132235 files and directories currently installed.)
Removing ros-noetic-gtsam (4.2.0-2focal.20230620.181134) ...
Processing triggers for libc-bin (2.31-0ubuntu9.15) ...
```

Sorry if I misunderstand :-)
Thanks for any info.

matlabbe, 2024-05-12:
The latest ros package now depends on `ros-noetic-gtsam` and `ros-noetic-libg2o` so you don't have to install those packages individually. However, if you installed gtsam or g2o another way at the same time, there could be issue like above. You can have to uninstall one the version.