introlab/rtabmap_ros #890: Broken tf_tree in the " Orbbec Astra mini s + Odometry + 2D laser" mode

cy-2022, 2023-02-16:
Hi Mathieu @matlabbe, as mentioned here https://github.com/introlab/rtabmap_ros/issues/881#issuecomment-1425458430, the ros binary version of rtabmap now works on our devices. Therefore, as our ultimate goal is a visual-lidar-fused SLAM solution, we went ahead to try out the "Orbbec Astra mini s + Odometry + 2D laser" mode following the intructions here http://wiki.ros.org/rtabmap_ros/Tutorials/SetupOnYourRobot (thank you very much for it, very very helpful and informative). 

The mapping results actually looked great (especially considering that we used to try with ORB-SLAM2/3 + Orbbec Astra mini s only in the same circular-corridor scenario).
[image]

However, when we proceeded with rviz, we immediately noticed that there was something wrong with the TF. The state was warm. The tf_tree looks like this:
[image]

Here is the launch file we've been using:
<launch>
  <group ns="rtabmap">

    
    

    
    <node pkg="tf2_ros" type="static_transform_publisher" name="cam_base_link"
      args="0 0 0 -1.57 0 -1.57 /camera_link /camera_rgb_optical_frame" />

    <node pkg="nodelet" type="nodelet" name="rgbd_sync" args="standalone rtabmap_ros/rgbd_sync" output="screen">
      <remap from="rgb/image"       to="/camera/rgb/image_rect_color"/>
      <remap from="depth/image"     to="/camera/depth_registered/image_raw"/>
      <remap from="rgb/camera_info" to="/camera/rgb/camera_info"/>
      <remap from="rgbd_image"      to="rgbd_image"/> 
       
      
      
      <param name="approx_sync"       value="true"/> 
    </node>

    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="--delete_db_on_start">
          <param name="frame_id" type="string" value="base_link"/>

          <param name="subscribe_depth" type="bool" value="false"/>
          <param name="subscribe_rgbd" type="bool" value="true"/>
          <param name="subscribe_scan" type="bool" value="true"/>

          <remap from="odom" to="/odom"/>
          <remap from="scan" to="/scan"/>
          <remap from="rgbd_image" to="rgbd_image"/>

          <param name="queue_size" type="int" value="300"/>

          
          <param name="RGBD/NeighborLinkRefining" type="string" value="true"/>
          <param name="RGBD/ProximityBySpace"     type="string" value="true"/>
          <param name="RGBD/AngularUpdate"        type="string" value="0.01"/>
          <param name="RGBD/LinearUpdate"         type="string" value="0.01"/>
          <param name="RGBD/OptimizeFromGraphEnd" type="string" value="false"/>
          <param name="Grid/FromDepth"            type="string" value="false"/> 
          <param name="Reg/Force3DoF"             type="string" value="true"/>
          <param name="Reg/Strategy"              type="string" value="1"/> 
          
          
          <param name="Icp/VoxelSize"                 type="string" value="0.05"/>
          <param name="Icp/MaxCorrespondenceDistance" type="string" value="0.1"/>
    </node>

    
    <node pkg="rtabmap_ros" type="rtabmapviz" name="rtabmapviz" args="-d $(find rtabmap_ros)/launch/config/rgbd_gui.ini" output="screen">
      <param name="subscribe_depth"             type="bool" value="false"/>
      <param name="subscribe_rgbd"              type="bool" value="true"/>
      <param name="subscribe_scan"              type="bool" value="true"/>
      <param name="frame_id"                    type="string" value="base_link"/>

      <remap from="rgb/image"       to="/camera/rgb/image_rect_color"/>
      <remap from="depth/image"     to="/camera/depth_registered/image_raw"/>
      <remap from="rgb/camera_info" to="/camera/rgb/camera_info"/>
      <remap from="scan"            to="/scan"/>
      
    </node>
  </group>
  
  
</launch>

I've also tried to some "researching" regarding map - odom - base_link and have tried to launch a few other nodes on our device, like chassis_control and robot_pose_ekf in order to make the odom-base_link link available. But it just didn't work out. Previously, we were also using Cartographer for mapping. There I saw it broadcast map to base_link, while now in my case with rtabmap it was map to odom. So I got confused and haven't been able to sort it all out. Thought while I was thinking I may as well ask in the meantime. Thank you very much again, Mathieu, for your kind help. 

Cheers, 
Yao

cy-2022, 2023-02-17:
Here is a bit of follow-up. First, we've realized that we should remove the static transform added in the launch file
<node pkg="tf2_ros" type="static_transform_publisher" name="cam_base_link"
  args="0 0 0 -1.57 0 -1.57 /camera_link /camera_rgb_optical_frame" />
(well, originally, in our urdf, we had a link between base_link and a camera_link_02. Therefore, there was that error. Now that we've corrected the urdf. This static transform is no longer needed.)
[image]

But still, the tf_tree remains broken. And we've checked, by only launching the odom node, although its child_frame_id is base_link, there is nothing in the tf_tree. So I guess we should probably check our odom node and maybe the chassis_control node as well?

matlabbe, 2023-02-18:
It feels like the node publishing `/odom` topic is not publishing the corresponding TF. See http://wiki.ros.org/navigation/Tutorials/RobotSetup/Odom for example.

cy-2022, 2023-02-19:
Thanks a lot for your reply, Mathieu. I'll take a closer look at the node publishing the /odom topic. I thought as long as the child_frame_id was base_link, it should do the job already.

matlabbe, 2023-02-20:
The odom topic is fine, it is just to properly not break the TF tree, the corresponding TF is required. At worst, you can use this [node](https://github.com/introlab/rtabmap_ros/blob/master/src/OdomMsgToTFNode.cpp) to publish the TF:
```
rosrun rtabmap_ros odom_msg_to_tf odom:=/odom
```

cy-2022, 2023-02-20:
We have modified the node publishing the /odom topic accordingly such that the TF is now published. And indeed the TF tree is complete. Thank you very much, Mathieu. I'm closing this issue. 
[image]

[image]

I have a further question regarding lidar-camera calibration. And I'll open another issue for that.

closed 2023-02-20 by cy-2022