introlab/rtabmap_ros #893: OpenCV Error: Assertion failed with Realsense T265

novadr, 2023-02-20:
Hello,

I solved my last problem (the malloc error which was caused by my version of librealsense) by reinstalling a new environment https://github.com/introlab/rtabmap_ros/issues/881

I built RTABMAP with _ros melodic_, _librealsense_ _2.42_, _realsense-ros_ and _ORB-SLAM3_ on my **Ubuntu 18.04**.
I use the branch **0.20.22-melodic** of rtabmap and rtabmap_ros because I had some troubles with the 0.20.23.

Here are the dependencies according to `rtabmap --version`: 

```
RTAB-Map:              0.20.22
PCL:                     1.8.1
With VTK:                6.3.0
OpenCV:                  3.2.0
With OpenCV xfeatures2d: false
With OpenCV nonfree:     false
With ORB OcTree:          true
With SuperPoint Torch:   false
With Python3:            false
With FastCV:             false
With OpenGV:             false
With Madgwick:            true
With PDAL:               false
With TORO:                true
With g2o:                false
With GTSAM:              false
With Vertigo:             true
With CVSBA:              false
With Ceres:              false
With OpenNI2:             true
With Freenect:            true
With Freenect2:          false
With K4W2:               false
With K4A:                false
With DC1394:              true
With FlyCapture2:        false
With ZED:                false
With ZED Open Capture:   false
With RealSense:          false
With RealSense SLAM:     false
With RealSense2:          true
With MYNT EYE S:         false
With DepthAI:            false
With libpointmatcher:     true
With CCCoreLib:          false
With octomap:             true
With cpu-tsdf:           false
With open chisel:        false
With Alice Vision:       false
With LOAM:               false
With FLOAM:              false
With FOVIS:              false
With Viso2:              false
With DVO:                false
With ORB_SLAM3:           true
With OKVIS:              false
With MSCKF_VIO:          false
With VINS-Fusion:        false
With OpenVINS:           false
```

The compilation was successful and I can now run rtabmap with the ORBSLAM3 odometry on the euroc dataset.

It also works with my Realsense D435 in RGBD and Stereo mode.

My issue concern the Realsense T265. I calibrate the camera and follow all the instruction I founded here : 
http://official-rtab-map-forum.206.s1.nabble.com/Slam-using-Intel-RealSense-tracking-camera-T265-td6333.html#a6343

Unfortunately, I got an OpenCV error when the algorithm tries to undistort the fisheye images..
[image]

I can launch the algorithm with the two fisheye images without rectification but the point cloud will not be accurate and when I launch the algorithm with the rectification I got the error above.

The error comes when I use the command `python camera_info_pub.py `for the two cameras.

Has somebody encounter the error please or has somebody any idea on how to fix that please ?

matlabbe, 2023-02-21:
Did you removed `-march=native` when rebuilding ORB_SLAM3? See https://gist.github.com/matlabbe/f5cb281304a1305b2824a6ce19792e13

novadr, 2023-02-21:
Hi @matlabbe, yes I applied the patch for ORB-SLAM3. 
I follow the instruction of this post for the installation of ORB-SLAM3 and for building rtabmap:
http://official-rtab-map-forum.206.s1.nabble.com/How-to-use-RTAB-Map-with-ORB-SLAM3-in-ROS-td9093.html

It has been built successfully and I can run rtabmap with orbslam3 on the euroc dataset. I can also launch the RGB-D and Stereo HandHeldMapping with my Realsense D435.
https://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping
https://wiki.ros.org/rtabmap_ros/Tutorials/StereoHandHeldMapping

matlabbe, 2023-02-26:
I reproduced the error on my side. The problem is that realsense node is published the left and right images with exactly the same frame_id. That error from opencv happens when we pass (0,0,0) translation in cv::stereoRectify function. In commit above, I added an error message when both frame_id are the same (or there is no baseline in TF between the cameras).
```
[ERROR] [1677369584.928528120]: Parameter Rtabmap/ImagesAlreadyRectified is false but we cannot get a valid TF between 
the two cameras! Identity transform returned between left and right cameras. Verify that if TF between the cameras is valid: 
"rosrun tf tf_echo camera_fisheye1_optical_frame camera_fisheye1_optical_frame".
```

Either modify realsense node to set the right frame on fisheye2 topic, or use the hack below to publish the camera_info with right frame. So based on examples in this thread:
```
roscore
rosparam set /camera/tracking_module/enable_mapping false
roslaunch realsense2_camera rs_t265.launch enable_fisheye1:=true enable_fisheye2:=true

rosrun rtabmap_ros yaml_to_camera_info.py\
    _yaml_path:=/home/mathieu/left.yaml\
    image:=/camera/fisheye1/image_raw\
    camera_info:=/camera/fisheye1/camera_info_calib

rosrun rtabmap_ros yaml_to_camera_info.py\
    _yaml_path:=/home/mathieu/right.yaml\
    image:=/camera/fisheye2/image_raw\
    camera_info:=/camera/fisheye2/camera_info_calib\
    _frame_id:=camera_fisheye2_optical_frame

roslaunch rtabmap_ros rtabmap.launch\
    args:="-d --udebug --Rtabmap/ImagesAlreadyRectified false"\
    stereo:=true\
    left_image_topic:=/camera/fisheye1/image_raw\
    right_image_topic:=/camera/fisheye2/image_raw\
    left_camera_info_topic:=/camera/fisheye1/camera_info_calib\
    right_camera_info_topic:=/camera/fisheye2/camera_info_calib\
    visual_odometry:=false\
    odom_frame_id:=camera_odom_frame
```

closed 2023-02-26 by matlabbe

novadr, 2023-02-27:
Thank you a lot ! I just test it and it works !
About the commit you add, in the message 
```
[ERROR] [1677369584.928528120]: Parameter Rtabmap/ImagesAlreadyRectified is false but we cannot get a valid TF between 
the two cameras! Identity transform returned between left and right cameras. Verify that if TF between the cameras is valid: 
"rosrun tf tf_echo camera_fisheye1_optical_frame camera_fisheye1_optical_frame"
```
Shouldn't it be : 
`rosrun tf tf_echo camera_fisheye1_optical_frame camera_fisheye2_optical_frame` 
instead of : 
`rosrun tf tf_echo camera_fisheye1_optical_frame camera_fisheye1_optical_frame` 
?

matlabbe, 2023-03-06:
Actually, it means that the left and right camera topics use the same `frame_id`. The message is correct, but I could have another text message when its because the same `frame_id` is used in both messages.