introlab/rtabmap_ros #896: Possibility of using depth cameras on recent phones? Galaxy a53 for instance?

jz1212, 2023-02-22:
Hello, I was looking for info regarding depth camera applications on these newer generation of phones, I tried all sorts of keywords for web searches to look for more details regarding the sensors used in these but basically nothing comes up other than what is in the official spec sheet - "5MP depth camera", and almost no reviews show any real application for them apart from blurring background for photos/video.
Have you heard something more about these sensors, possibility of adapting them to stream data to RTAB like with the older Tango devices? Or perhaps the accuracy just isn't there and "blurring the background" is about all they can be used for?

matlabbe, 2023-02-26:
I didn't try latest android phones with depth cameras, not sure if it is still painful (e.g., camera time synchronization issues, different resolutions, cameras not logically on same camera device, poor depth quality...) to use them with ARCore. **IF** I had ressources and time to develop again on ARCore, I would go for phones listed with TOF camera on this page https://developers.google.com/ar/devices (labelled with "**Supports time-of-flight (ToF) hardware depth sensor**"). Note that back in the days, I've made RTAB-Map ArCore working with Note10+, but ARCore is not using the wide angle camera for tracking, so the visual odometry is drifting a lot sometime (just by passing in front of a white/textureless wall). 

My cheapest recommendation to get maximum value (even better than Google Tango) is to get an old iPad Pro 2020, then install [RTAB-Map iOS](https://www.youtube.com/watch?v=rVpIcrgD5c0) on it. :)

cheers,
Mathieu

jz1212, 2023-02-27:
Thank you for the reply, seems like newer phone TOF sensors are really low resolution, just 0.3MP, and the "5MP depth camera" they list in specs is really just a sensor with no IR filter, not capable of measuring distances accurately, probably something like a weak IR source, enough to illuminate something near the phone, and weak enough that more distant things are just dark, and then they can apply this as a mask to get the blur effect or swap the background.
I looked at iphones with lidar, but I really dislike apple as a company, and the situation is not as desperate to go and buy one of their products.
A used Velodyne VLP-16 is on its way to me, will see how that works out.

closed 2023-04-15 by matlabbe