introlab/rtabmap_ros #897: Compare mapping accuracy inside RTABMap

mrtnbm, 2023-02-23:
Hi Mathieu and everyone else,

i want to compare 3d + lidar + odom + Marker Slam with normal 3d + lidar + odom scan by evaluating the map accuracy only inside RTABMap.

What would be the best way to generate ground truth or which setup would be the best to use as ground truth? I don't have any ground truth data because I don't own  any simulated environment in gazebo. Furthermore, I also didn't obtain the architect/build plan of the building that is used for testing. So no real ground truth I can use.

I thought about either using 3d Lidar ICP+odometry guess or SLAM without markers as ground truth, and then comparing the 2d projected maps with SSMI against SLAM with Markers. What do you think? Is there a better option for comparing Mapping accuracy on real data using different setups?

I also noticed you have statistics output implemented to use metrics like ATE and RPE. But I'm not sure if these even have a meaning because I dont really know which of the trajectories are better in the first place. 

I also read about the evo comparison tool but it only supports APE and RPE and not the comparison of the map itself. I was successful in plotting APE & RPE tho.

Last but not least, I could maybe analyze the amount of visual features that get extracted comparing SLAM with and without markers and therefore conclude that the map is more accurate, but that's all not too precise I guess. The last thing I could do would be a visual comparison, but I don't think that is accurate enough to have any meaning at all.

And sorry for posting if this is offtopic. I thought it would fit better here provided that I only use RTAB-Map to compare between the different setups and no other SLAM approach.

matlabbe, 2023-02-26:
In SLAM, we often use ATE or RPE as metrics for evaluating SLAM/VO accuracy. To compute them, you need a ground truth trajectory, generaly computed from external tracking systems like VICON or optitrack.

In LiDAR world, instead of the trajectory, you can compare the resulting point cloud of the map with somekind of a ground truth. With CloudCompare, you can get the distance between two point clouds. To get the ground truth point cloud, one may use a Total Station LiDAR.

In your case, if you cannot generate a ground truth for the trajectory AND the point cloud map, you will have to setup the SLAM approach with highest settings (may require to process offline the map data) to get the most perfect map. You would inspect it visually, then when satisfied that map it will give you the trajectory ground truth and point cloud ground truth. You can then use different sensors/parameters on the SLAM approach and compare the trajectories with ATE or comparing the resulting point clouds with CloudCompare.

I don't know exactly your setup, but to generate a ground truth with RTAB-Map and a 3D lidar, unless you have an IMU and you can do lidar deskewing with it, I suggest to move slowly when recording the ground truth map (**avoid fast rotation!**). Examples of 3D lidar-based launch files: 
 * https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/test_ouster_gen2.launch
 * https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/test_velodyne.launch

mrtnbm, 2023-02-26:
Thank you for the input! I will try out cloud compare :)

I have an IMU (T265), 2x D435, 1x D415 and the VLP-16 Lidar. I tried the deskewing node, but it doesn't work correctly. It always gives the me error: 

`Could not estimate motion of velodyne accordingly to fixed frame t265_odom_frame between stamps 1677086050.733220 and 1677086050.933503! (Lookup would require extrapolation 0.201647126s into the future.  Requested time 1677086050.933502674 but the latest data is at time 1677086050.731855631, when looking up transform from frame [velodyne] to frame [t265_odom_frame]. canTransform returned after 0.0100646 timeout was 0.01.)`

Best regards
mb

matlabbe, 2023-04-15:
Are the clocks used for velodyne and T265 synchronized? If their time drift apart over time, you may get issue like this. If the delay is large, you may be able to see it by printing the header of odometry topic from T265 and the header of the velodyne topic side by side.

mrtnbm, 2023-05-25:
Hey matlabbe,

I'm currently not able to investigate this issue further. It will probably also take some time until I can work on it again.

Until then, I will close this issue to not waste your time and I will reopen it as soon as I review the whole project.

Thank you for all the help, best regards and see you soon!
mrtnbm

closed 2023-05-25 by mrtnbm