introlab/rtabmap_ros #899: Memory exception when calling set_mode_localization

peredwardsson, 2023-02-24:
Rtabmap is ticking along happily making a nice map of my office. I get happy with the map, decide to turn off mapping and just do localization, and this happens.
```
[rtabmap-6] [INFO] [1677231043.072151396] [rtabmap.rtabmap]: rtabmap: Set localization mode
[rtabmap-6] [ WARN] (2023-02-24 09:29:23.983) MsgConversion.cpp:1758::getTransform() (can transform odom -> base_link?) Lookup would require extrapolation into the future.  Requested time 1677230963.700329 but the latest data is at time 1677230945.963638, when looking up transform from frame [base_link] to frame [odom]. canTransform returned after 0.203109 timeout was 0.2. (wait_for_transform=0.200000)
[rtabmap-6] [ WARN] (2023-02-24 09:29:24.187) MsgConversion.cpp:1758::getTransform() (can transform odom -> base_link?) Lookup would require extrapolation into the future.  Requested time 1677230963.833892 but the latest data is at time 1677230945.963638, when looking up transform from frame [base_link] to frame [odom]. canTransform returned after 0.202991 timeout was 0.2. (wait_for_transform=0.200000)
[rtabmap-6] [ WARN] (2023-02-24 09:29:24.414) MsgConversion.cpp:1758::getTransform() (can transform odom -> base_link?) Lookup would require extrapolation into the future.  Requested time 1677230964.101129 but the latest data is at time 1677230945.963638, when looking up transform from frame [base_link] to frame [odom]. canTransform returned after 0.202894 timeout was 0.2. (wait_for_transform=0.200000)
[rtabmap-6] [ WARN] (2023-02-24 09:29:24.618) MsgConversion.cpp:1758::getTransform() (can transform odom -> base_link?) Lookup would require extrapolation into the future.  Requested time 1677230964.167735 but the latest data is at time 1677230945.963638, when looking up transform from frame [base_link] to frame [odom]. canTransform returned after 0.20293 timeout was 0.2. (wait_for_transform=0.200000)
[rtabmap-6] [ WARN] (2023-02-24 09:29:24.822) MsgConversion.cpp:1758::getTransform() (can transform odom -> base_link?) Lookup would require extrapolation into the future.  Requested time 1677230964.568496 but the latest data is at time 1677230945.963638, when looking up transform from frame [base_link] to frame [odom]. canTransform returned after 0.202422 timeout was 0.2. (wait_for_transform=0.200000)
[rtabmap-6] [ WARN] (2023-02-24 09:29:25.112) MsgConversion.cpp:1758::getTransform() (can transform odom -> base_link?) Lookup would require extrapolation into the future.  Requested time 1677230964.768844 but the latest data is at time 1677230945.963638, when looking up transform from frame [base_link] to frame [odom]. canTransform returned after 0.202324 timeout was 0.2. (wait_for_transform=0.200000)
[rtabmap-6] [ WARN] (2023-02-24 09:29:25.316) MsgConversion.cpp:1758::getTransform() (can transform odom -> base_link?) Lookup would require extrapolation into the future.  Requested time 1677230964.835502 but the latest data is at time 1677230945.963638, when looking up transform from frame [base_link] to frame [odom]. canTransform returned after 0.20324 timeout was 0.2. (wait_for_transform=0.200000)
[rtabmap-6] [ WARN] (2023-02-24 09:29:25.549) MsgConversion.cpp:1758::getTransform() (can transform odom -> base_link?) Lookup would require extrapolation into the future.  Requested time 1677230965.303008 but the latest data is at time 1677230945.963638, when looking up transform from frame [base_link] to frame [odom]. canTransform returned after 0.203106 timeout was 0.2. (wait_for_transform=0.200000)
[rtabmap-6] [ WARN] (2023-02-24 09:29:25.783) MsgConversion.cpp:1758::getTransform() (can transform odom -> base_link?) Lookup would require extrapolation into the future.  Requested time 1677230965.503245 but the latest data is at time 1677230945.963638, when looking up transform from frame [base_link] to frame [odom]. canTransform returned after 0.202658 timeout was 0.2. (wait_for_transform=0.200000)
[rtabmap-6] [ WARN] (2023-02-24 09:29:26.142) MsgConversion.cpp:1758::getTransform() (can transform odom -> base_link?) Lookup would require extrapolation into the past.  Requested time 1677230965.636889 but the earliest data is at time 1677230965.995936, when looking up transform from frame [base_link] to frame [odom]. canTransform returned after 0.202484 timeout was 0.2. (wait_for_transform=0.200000)
[rtabmap-6] [ WARN] (2023-02-24 09:29:26.450) MsgConversion.cpp:1758::getTransform() (can transform odom -> base_link?) Lookup would require extrapolation into the past.  Requested time 1677230965.970721 but the earliest data is at time 1677230965.995936, when looking up transform from frame [base_link] to frame [odom]. canTransform returned after 0.202953 timeout was 0.2. (wait_for_transform=0.200000)
[rtabmap-6] [INFO] [1677231045.149253566] [rtabmap.rtabmap]: rtabmap: Localization mode enabled!
[rtabmap-6] [FATAL] [1677231045.243857904] [rtabmap.rtabmap]: [FATAL] (2023-02-24 09:30:45.243) Rtabmap.cpp:4080::process() Condition (_memory->getSignature(id) != 0) not met! [id=678]
[rtabmap-6] terminate called after throwing an instance of 'UException'
[rtabmap-6]   what():  [FATAL] (2023-02-24 09:30:45.243) Rtabmap.cpp:4080::process() Condition (_memory->getSignature(id) != 0) not met! [id=678]
[ERROR] [rtabmap-6]: process has died [pid 91, exit code -6, cmd '/src/slam_ws/install/rtabmap_ros/lib/rtabmap_ros/rtabmap --ros-args -r __node:=rtabmap -r __ns:=/rtabmap --params-file {params_file} -r rgbd_image0:=/front_camera/rgbd -r rgbd_image1:=/back_camera/rgbd -r map:=/map'].
```

Running ROS2 Humble with rtabmap & rtabmap_ros versions 20.23-humble. Issue is intermittent, unfortuntately. Restarting rtabmap_ros could load the db fine, and i could then run the service. But this has happened a few times. Can I gather some logs from the occassion when it breaks for troubleshooting purposes?

matlabbe, 2023-02-25:
It seems to have a strange case where the last localization id is in the optimized local graph but not in memory, causing the assert. Reference: https://github.com/introlab/rtabmap/blob/95e6a9f03936697a60be2c26e119c519e47c11f5/corelib/src/Rtabmap.cpp#L4055-L4080

I tried switching slam/localization modes multiple times without reproducing the assert. You could launch rtabmap node with `--udebug` argument to get debugging logs.

Krzo99, 2024-07-05:
Hey, this is an old thread, but did you figure out any fix for this @peredwardsson?

closed 2025-03-05 by peredwardsson