introlab/rtabmap_ros #900: Is there a way to align gridmaps?

mrtnbm, 2023-02-26:
Hello,

I need to compare different grid maps from different setups. In order to do this I need perfectly aligned grid maps.

I've got it working by using Marker/Priors, which also transforms the grid but I need to do this for non marker setups as well and manually adding a transform for the map didn't work however.

Also, is there a way to limit the maximum distance in which cells are still occupied? Some lidar rays that are casted through the window (see picture) will measure the outside environment and with that the grid map gets much bigger. Should I just limit the lidar range or is there a Grid param for this? I saw that there is indeed Grid/RangeMax but that doesn't seem to limit the cell occupation or the range in general. 
[image]

Best regards,
mb

matlabbe, 2023-04-15:
`Grid/RangeMax` should limit the size of the local occupancy grid. You can use `rtabmap-databaseViewer`, open Core Parameters panel, select Grid group, change parameters and click on Edit->Regenerate all local grids. This is a quick way to see the effect of each parameter. You can open the 3D view to check individual node results.

mrtnbm, 2023-05-25:
Hey Matlabbe,

thank you for the help! Sorry, didn't see your message.

closed 2023-05-25 by mrtnbm