introlab/rtabmap_ros #902: Is there any flag/some step to enable initialpose?

rakshith95, 2023-03-02:
Hello,
I would like to provide an initial guess for rtabmap, in the localization mode, and I'm trying to publish to  the `/<rtabmap namespace>/initialpose` with the rviz 2D pose estimate button (I also tried `rostopic pub` ), but it doesn't seem to work.  

Is there some flag I should set for this, or are there  some specific things to keep in mind while trying to initialize this way? I'm unable to find documentation on this topic , so I'm not really sure what I'm missing.

rakshith95, 2023-03-02:
I was able to get it working with `rostopic pub` (I was sending invalid quaternions initially), but it still doesn't respond to the initialpose from rviz, which has to be a valid pose.

matlabbe, 2023-03-03:
Make sure your global frame in RVIZ is set to map. It should work with rviz 2D pose estimate button with right topic. If you ar on different computers, do a `rostopic echo /rtabmap/initialpose` on robot side to make sure it is receiving it.

rakshith95, 2023-06-06:
@matlabbe should the `initialpose` be in the `map` frame or the `<frame_id>` (base_link) ?