introlab/rtabmap_ros #904: Does rtabmap works on different image type and size?

julianraheema, 2023-03-05:
System setup:
Ubuntu 20.04 Noetic
Camera similar to Realsense D415/D435

I have a camera that is producing /image /depth/image /camera_info at rate 0.8203
**/spot/camera/back/image topic:**
header: 
  seq: 73
  stamp: 
    secs: 1677524361
    nsecs: 548337425
  frame_id: "back_fisheye"
height: 480
width: 640
encoding: "mono8"
is_bigendian: 1

**/spot/camera/back/camera_info topic:**
header: 
  seq: 113
  stamp: 
    secs: 1677524408
    nsecs: 767619322
  frame_id: "back_fisheye"
height: 480
width: 640
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [330.94464111328125, 0.0, 316.32916259765625, 0.0, 329.81549072265625, 249.16798400878906, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [330.94464111328125, 0.0, 316.32916259765625, 0.0, 0.0, 329.81549072265625, 249.16798400878906, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False

**/spot/depth/back/image topic:**
header: 
  seq: 85
  stamp: 
    secs: 1677524365
    nsecs: 215576725
  frame_id: "back"
height: 240
width: 424
encoding: "16UC1"
is_bigendian: 0
step: 848

**/spot/depth/back/camera_info topic**
header: 
  seq: 38
  stamp: 
    secs: 1677524317
    nsecs: 934621403
  frame_id: "back"
height: 240
width: 424
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [211.1915283203125, 0.0, 215.72769165039062, 0.0, 211.1915283203125, 118.88978576660156, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [211.1915283203125, 0.0, 215.72769165039062, 0.0, 0.0, 211.1915283203125, 118.88978576660156, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False

**/spot/back/laser_scan topic**
header: 
  seq: 982
  stamp: 
    secs: 1677524447
    nsecs: 326256348
  frame_id: "laser_back_frame"
angle_min: -0.7760326862335205
angle_max: 0.7960231304168701
angle_increment: 0.0037164438981562853
time_increment: 0.0
scan_time: 0.032999999821186066
range_min: 0.3499999940395355
range_max: 3.0

              topic                                           rate   min_delta   max_delta   std_dev   window
=====================================================================================
/spot/camera/back/image                       1.366   0.6358      0.8287      0.08678   5     
/spot/depth/back/image                          1.353   0.6461      0.8186      0.07324   5     
/spot/camera/back/camera_info             1.366   0.6358      0.8287      0.08678   6     
/spot/depth/back/camera_info                1.353   0.6461      0.8186      0.07324   6 
/spot/back/laser_scan                             1.131   0.8285      0.9896      0.0657    5

I am trying to get rtabmap to run as demo http://wiki.ros.org/rtabmap_ros 3.1 mapping but no luck. I would appreciate your help and support.

matlabbe, 2023-03-06:
What is the output of:
```
rostopic hz /spot/camera/back/image_rgb_crop \
  /spot/depth/back/image \
  /spot/camera/back/camera_info_crop
```
?
This error is related to publishing rate, topics are not published at a similar frame rate or one of this topic is not published.

Based on framerate of the base topics, 1.3 Hz is quite slow (too slow if the depth and color images are not exactly sync). Rtabmap could eventually synchronize them, ideally if they have exact time sync, you may launch with `approx_sync:=false` to make sure correct color+depth images are sync. 

For the resolution, rtabmap can support smaller depth image than color image, but they should have same aspect ratio and the depth image should be registered with color image. If they don't have same aspect ratio, this is a good indication that the depth image is not registered to color image.

rtabmap supports 1 channel gray-scale and 3-channels color images. For depth, it supports 16UC1 (mm) and 32FC1 (m) format.

julianraheema, 2023-03-06:
Thank you @matlabbe for your response. Here are the output for the topics that you asked for:

             topic                   rate    min_delta   max_delta   std_dev   window
======================================================================================
/spot/camera/back/image_rgb_crop     0.9823   0.5086      2.349       0.5644    17    
/spot/depth/back/image               0.9817   0.4985      2.351       0.5648    17    
/spot/camera/back/camera_info_crop   0.9823   0.5086      2.349       0.5644    17    

no new messages
              topic                   rate    min_delta   max_delta   std_dev   window
======================================================================================
/spot/camera/back/image_rgb_crop     0.9239   0.5086      2.349       0.605     18    
/spot/depth/back/image               0.9242   0.4985      2.351       0.6037    18    
/spot/camera/back/camera_info_crop   0.9244   0.5086      2.349       0.604     18    

              topic                   rate    min_delta   max_delta   std_dev   window
======================================================================================
/spot/camera/back/image_rgb_crop     0.9435   0.5086      2.349       0.5952    19    
/spot/depth/back/image               0.9425   0.4985      2.351       0.593     19    
/spot/camera/back/camera_info_crop   0.9435   0.5086      2.349       0.5939    19 

FYI: I am generating /spot/camera/back/rgb_image from /spot/camera/back/image to generate 3 channels image then I am using another node to reduce the image size from 640x480 to match the depth image 424x240. 

Out of the camera box, I am receiving these topics:
/spot/camera/back/image 640x480 mono8
/spot/depth/back/image 424x240 16UC1
/spot/camera/back/camera_info
/spot/depth/back/camera_info

When I plug these topics into rtabmap node, no map is produced. Here is my launch file:

<launch>

  
  
  

  
  <arg name="rviz" default="true" />
  <arg name="rtabmapviz" default="false" />
  <arg name="compressed" default="false"/>
  <arg name="align_depth" default="false"/>

  <param name="use_sim_time" type="bool" value="true"/>

  
  <arg name="localization"      default="false"/>
  <arg if="$(arg localization)" name="rtabmap_args"  default=""/>
  <arg unless="$(arg localization)" name="rtabmap_args"  default="--delete_db_on_start"/>

  <include file="$(find spot_description)/launch/description_only.launch" />
  <include file="$(find depthimage_to_laserscan)/launch/image_to_laser.launch" />
  <include file="$(find ros_imresize)/launch/imresize.launch" />
  
  <node name="odom_corrector" pkg="nav_utils" type="odom_twist_to_child_frame">
    <remap from="/odom" to="/spot/odometry"/>
    <remap from="/odom_out" to="/spot/odometry_corrected"/>
  </node>

  <node pkg="spot_driver" type="gray2rgb.py" name="gray2rgb">
    <remap from="gray_in" to="/spot/camera/back/image"/>
    <remap from="rgb_out" to="/spot/camera/back/image_rgb"/>
  </node>

<group ns="rtabmap">
    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="$(arg rtabmap_args)">
      <param name="frame_id"           type="string" value="base_link"/>
      <param name="wait_for_transform" type="bool" value="true"/>

      <param name="subscribe_depth" type="bool" value="true"/>
[... 49 lines cut ...]
      <remap from="depth/image"     to="/spot/depth/back/image"/>
      <remap from="rgb/camera_info" to="/spot/camera/back/camera_info_crop"/>
      <remap from="scan"            to="/spot/back/laser_scan"/>
      <remap from="odom"            to="/spot/odometry_corrected"/>
      <param name="compressed"        value="$(arg compressed)"/>
      <param name="queue_size" type="int" value="100"/>

    </node>
  </group>

  
  <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="-d $(find spot_viz)/rviz/spot_rtab.rviz" output="screen"/>

  <node pkg="nodelet" type="nodelet" name="points_xyzrgb" args="standalone rtabmap_ros/point_cloud_xyzrgb">
    <remap from="rgb/image"       to="/spot/camera/back/image_rgb_crop"/>
    <remap from="depth/image"     to="/spot/depth/back/image"/>
    <remap from="rgb/camera_info" to="/spot/camera/back/camera_info_crop"/>
    <remap from="cloud"           to="voxel_cloud" />
    <remap from="odom"            to="/spot/odometry_corrected"/>
    
    <param name="queue_size" type="int" value="100"/>
    <param name="voxel_size" type="double" value="0.01"/>
  </node>

</launch>

julianraheema, 2023-03-06:
When I plug these topics into rtabmap node, no map is produced. Here is my launch file:
/spot/camera/back/image 640x480 mono8
/spot/depth/back/image 424x240 16UC1
/spot/camera/back/camera_info
/spot/depth/back/camera_info

SUMMARY
========

PARAMETERS
 * /depthimage_to_laserscan/output_frame_id: laser_back_frame
 * /depthimage_to_laserscan/range_max: 3.0
 * /depthimage_to_laserscan/range_min: 0.35
 * /depthimage_to_laserscan/scan_height: 1
 * /depthimage_to_laserscan/scan_time: 0.033
 * /front_depthimage_to_laserscan/output_frame_id: laser_frontright_...
 * /front_depthimage_to_laserscan/range_max: 3.0
 * /front_depthimage_to_laserscan/range_min: 0.35
 * /front_depthimage_to_laserscan/scan_height: 1
 * /front_depthimage_to_laserscan/scan_time: 0.033
 * /points_xyzrgb/queue_size: 100
 * /points_xyzrgb/voxel_size: 0.01
 * /robot_description: <?xml version="1....
 * /ros_imresize/camera_info: /spot/camera/back...
 * /ros_imresize/resize_height: 240
 * /ros_imresize/resize_width: 424
 * /ros_imresize/topic_crop: /spot/camera/back...
 * /ros_imresize/undistord: True
 * /rosdistro: noetic
 * /rosversion: 1.15.15
 * /rtabmap/rtabmap/Grid/Sensor: false
 * /rtabmap/rtabmap/Icp/CorrespondenceRatio: 0.2
 * /rtabmap/rtabmap/Icp/MaxCorrespondenceDistance: 0.15
 * /rtabmap/rtabmap/Icp/PointToPlane: false
 * /rtabmap/rtabmap/Icp/Strategy: false
 * /rtabmap/rtabmap/Icp/VoxelSize: 0.05
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/Mem/STMSize: 30
 * /rtabmap/rtabmap/RGBD/LocalRadius: 5
 * /rtabmap/rtabmap/RGBD/NeighborLinkRefining: true
 * /rtabmap/rtabmap/RGBD/OptimizeFromGraphEnd: false
 * /rtabmap/rtabmap/RGBD/OptimizeMaxError: 4
 * /rtabmap/rtabmap/RGBD/ProximityBySpace: true
 * /rtabmap/rtabmap/RGBD/ProximityByTime: false
 * /rtabmap/rtabmap/RGBD/ProximityPathMaxNeighbors: 10
 * /rtabmap/rtabmap/Reg/Force3DoF: true
 * /rtabmap/rtabmap/Reg/Strategy: 1
 * /rtabmap/rtabmap/Vis/MinInliers: 12
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/approx_sync_max_interval: 0.9
 * /rtabmap/rtabmap/frame_id: base_link
 * /rtabmap/rtabmap/odom_frame_id: odom
 * /rtabmap/rtabmap/odom_tf_angular_variance: 0.001
 * /rtabmap/rtabmap/odom_tf_linear_variance: 0.001
 * /rtabmap/rtabmap/queue_size: 100
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_scan: True
 * /rtabmap/rtabmap/wait_for_transform: True
 * /use_sim_time: True

NODES
  /
    depthimage_to_laserscan (depthimage_to_laserscan/depthimage_to_laserscan)
    front_depthimage_to_laserscan (depthimage_to_laserscan/depthimage_to_laserscan)
    gray2rgb (spot_driver/gray2rgb.py)
    odom_corrector (nav_utils/odom_twist_to_child_frame)
    points_xyzrgb (nodelet/nodelet)
    ros_imresize (ros_imresize/ros_imresize)
    rviz (rviz/rviz)
[... 205 lines cut ...]
  P[7]: 0
  P[8]: 0
  P[9]: 0
  P[10]: 1
  P[11]: 0
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: 0

[ INFO] [1678066341.164819048, 1677524288.734709635]: RECEIVED ROS camera calibration!

[ WARN] [1678066341.255826782, 1677524288.825684014]: Could not get transform from odom to base_link after 0.200000 seconds (for stamp=1677524287.130922)! Error="Lookup would require extrapolation 0.188154501s into the past.  Requested time 1677524287.130922079 but the earliest data is at time 1677524287.319076538, when looking up transform from frame [base_link] to frame [odom]. canTransform returned after 0.202224 timeout was 0.2.".
[ INFO] [1678066341.310007118, 1677524288.876194287]: Running

[FATAL] (2023-03-05 17:32:21.455) MsgConversion.cpp:1862::convertRGBDMsgs() Condition (imageWidth/depthWidth == imageHeight/depthHeight) not met! [rgb=640x480 depth=424x240]
[FATAL] [1678066341.455382541, 1677524289.018328854]: [FATAL] (2023-03-05 17:32:21.455) MsgConversion.cpp:1862::convertRGBDMsgs() Condition (imageWidth/depthWidth == imageHeight/depthHeight) not met! [rgb=640x480 depth=424x240]
terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2023-03-05 17:32:21.455) MsgConversion.cpp:1862::convertRGBDMsgs() Condition (imageWidth/depthWidth == imageHeight/depthHeight) not met! [rgb=640x480 depth=424x240]
[rtabmap/rtabmap-7] process has died [pid 43774, exit code -6, cmd /home/jraheema/src/autonomous_spot/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start scan:=/spot/back/laser_scan rgb/image:=/spot/camera/back/image depth/image:=/spot/depth/back/image rgb/camera_info:=/spot/camera/back/camera_info odom:=/spot/odometry_corrected __name:=rtabmap __log:=/home/jraheema/.ros/log/b89e6a02-bbbe-11ed-8ce0-a3bc6edd2b63/rtabmap-rtabmap-7.log].
log file: /home/jraheema/.ros/log/b89e6a02-bbbe-11ed-8ce0-a3bc6edd2b63/rtabmap-rtabmap-7*.log

matlabbe, 2023-03-06:
This is the result of my comment:

> For the resolution, rtabmap can support smaller depth image than color image, but they should have same aspect ratio and the depth image should be registered with color image. If they don't have same aspect ratio, this is a good indication that the depth image is not registered to color image.

If resolutions are not compatible, it will assert like this.

julianraheema, 2023-03-06:
Thank you @matlabbe. Any advice on how to register depth with color image?

matlabbe, 2023-03-06:
You could use [pointcloud_to_depthimage](http://wiki.ros.org/rtabmap_ros#rtabmap_ros.2Fpointcloud_to_depthimage) node. I would set `decimation` parameter to 2 in your case (added `point_cloud_xyz` node to convert depth to point cloud if you don't have point cloud topic already from the depth camera):
```
rosrun rtabmap_ros point_cloud_xyz \
   depth/image:=/spot/depth/back/image \
   depth/camera_info:=/spot/depth/back/camera_info \
   cloud:=/spot/depth/back/cloud

rosrun rtabmap_ros pointcloud_to_depthimage \
   camera_info:=/spot/camera/back/camera_info_crop \
   cloud:=/spot/depth/back/cloud \
   _decimation:=2 \
   _fixed_frame_id:=base_link \
   image_raw:=/spot/depth/back/registered_to_color
```
Note that `fixed_frame_id=base_link` only makes sense if the images from color and depth cameras were taken at the same time, otherwise, if you have external odometry, you would have to set to `fixed_frame_id=odom`. Feed `/spot/depth/back/registered_to_color` instead of `/spot/depth/back/image` to rtabmap node.

Note that if you are using latest code from source, change `rtabmap_ros` by `rtabmap_util` above.

julianraheema, 2023-03-07:
Thank you so much @matlabbe that works. You have been great helpful.

closed 2023-03-07 by julianraheema