introlab/rtabmap_ros #907: weird map

an99990, 2023-03-06:
HI,
I am using rtabmap ros along with the oak-d camera. The camera is put on top of a small vehicule. The generated map is a bit weird. Is it because the floor is shinning ? what could cause the map to be like that

```
roslaunch depthai_examples stereo_inertial_node.launch depth_aligned:=false
```

`roslaunch rtabmap_ros rtabmap.launch     args:="--delete_db_on_start"     stereo:=true     left_image_topic:=/stereo_inertial_publisher/left/image_rect     right_image_topic:=/stereo_inertial_publisher/right/image_rect     left_camera_info_topic:=/stereo_inertial_publisher/left/camera_info     right_camera_info_topic:=/stereo_inertial_publisher/right/camera_info     imu_topic:=/stereo_inertial_publisher/imu     frame_id:=oak-d_frame     approx_sync:=true     wait_imu_to_init:=true`

[image]

matlabbe, 2023-03-06:
It is expected, the floor acts like a mirror. To fill the empty space, try with `Grid/RayTracing=true` and `Grid/3D=false`.

cheers,
Mathieu

an99990, 2023-03-06:
are those arguments in the rtabmap.launch file ?

matlabbe, 2023-03-07:
With rtabmap.launch, you can do this:
```bash
roslaunch rtabmap_ros rtabmap.launch \
    args:="--delete_db_on_start --Grid/RayTracing true --Grid/3D false" \
    stereo:=true \
    left_image_topic:=/stereo_inertial_publisher/left/image_rect \
    right_image_topic:=/stereo_inertial_publisher/right/image_rect \
    left_camera_info_topic:=/stereo_inertial_publisher/left/camera_info \
    right_camera_info_topic:=/stereo_inertial_publisher/right/camera_info \
    imu_topic:=/stereo_inertial_publisher/imu \
    frame_id:=oak-d_frame \
    approx_sync:=true \
   wait_imu_to_init:=true
```

an99990, 2023-03-07:
wow it made such an improvement, thank you !

closed 2023-03-07 by an99990