introlab/rtabmap_ros #908: CUDA error when trying to install rtabmap for ROS2 Foxy

Aloha-Churchill, 2023-03-07:
Hello,

I would like to use RGBD-SLAM with the OAK-D Luxonis. Rtabmap in ROS1 Noetic worked great for this purpose, however I am having trouble getting Rtabmap installed in ROS2. Is there a way to work around this issue? Thanks for your help in advance! 

`aloha@gibbon:~/dev_ws$ colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> rtabmap 
Starting >>> rtabmap_msgs
Starting >>> rtabmap_python                                              
Starting >>> diff-drive-robot
Finished <<< diff-drive-robot [2.82s]                                                        
--- stderr: rtabmap                                                                           
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find CUDA: Found unsuitable version "10.1", but required is at
  least "11" (found /usr)
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:391 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.16/Modules/FindCUDA.cmake:1104 (find_package_handle_standard_args)
  /usr/local/zed/zed-config.cmake:72 (find_package)
  CMakeLists.txt:570 (find_package)

---
Failed   <<< rtabmap [5.24s, exited with code 1]
Aborted  <<< rtabmap_python [5.18s]
Aborted  <<< rtabmap_msgs [5.68s]                              

Summary: 1 package finished [7.59s]
  1 package failed: rtabmap
  2 packages aborted: rtabmap_msgs rtabmap_python
  1 package had stderr output: rtabmap
  11 packages not processed
`

matlabbe, 2023-03-07:
It is failing because of the ZED cmake config:
```
/usr/local/zed/zed-config.cmake
```
Try adding `-DWITH_ZED=OFF` to `--cmake-args`

Aloha-Churchill, 2023-03-09:
This worked, thank you so much for your quick response!

closed 2023-03-09 by Aloha-Churchill