introlab/rtabmap_ros #909: cloudFromDepthRGB() Cloud with only NaN values created!

JayParikh20, 2023-03-08:
Point cloud does fails to generate for most of the datasets
getting this error,
      
      [ INFO] [1678312002.799662227]: rtabmap 0.20.23 started...
      [ WARN] [1678312003.812500837]: Callbacks not syncronized for 5.0 secs!
      [ INFO] [1678312005.244696531]: Odom: quality=0, std dev=99.995000m|99.995000rad, update time=0.065001s
      [ WARN] (2023-03-08 16:46:45.477) util3d.cpp:605::cloudFromDepthRGB() Cloud with only NaN values created!

I tried on these datasets,
- TUM -> fr1/desk2 [Link](https://cvg.cit.tum.de/data/datasets/rgbd-dataset/download)
- rgbd-scenes-v2 [Link](http://rgbd-dataset.cs.washington.edu/dataset/rgbd-scenes-v2/)

It works perfectly for,
Redwood dataset -> living room 1 [Link](http://redwood-data.org/indoor/dataset.html)

JayParikh20, 2023-03-08:
... logging to /home/valorrobotics/.ros/log/b4184c62-bdfa-11ed-a065-48b02d833aa8/roslaunch-ubuntu-129589.log
      Checking log directory for disk usage. This may take a while.
      Press Ctrl-C to interrupt
      Done checking log file disk usage. Usage is <1GB.
      
      started roslaunch server http://ubuntu:36157/
      
      SUMMARY
      ========
      
      PARAMETERS
       * /rosdistro: noetic
       * /rosversion: 1.15.15
       * /vr/rgbd_odometry/frame_id: vr_camera
       * /vr/rgbd_odometry/subscribe_rgbd: True
       * /vr/rgbd_sync/approx_sync: False
       * /vr/rtabmap/RGBD/AngularUpdate: 0.01
       * /vr/rtabmap/RGBD/LinearUpdate: 0.01
       * /vr/rtabmap/RGBD/OptimizeFromGraphEnd: false
       * /vr/rtabmap/Rtabmap/DetectionRate: 30.0
       * /vr/rtabmap/approx_sync: False
       * /vr/rtabmap/frame_id: vr_camera
       * /vr/rtabmap/queue_size: 10
       * /vr/rtabmap/subscribe_depth: False
       * /vr/rtabmap/subscribe_rgb: False
       * /vr/rtabmap/subscribe_rgbd: True
      
      NODES
        /
          display_frames (stereo_camera/display_frames)
          pointcloud_visualizer (stereo_camera/pointcloud_visualizer)
          read_rgbd (stereo_camera/read_rgbd.py)
        /vr/
          rgbd_odometry (rtabmap_ros/rgbd_odometry)
          rgbd_sync (nodelet/nodelet)
          rtabmap (rtabmap_ros/rtabmap)
      
      auto-starting new master
      process[master]: started with pid [129608]
      ROS_MASTER_URI=http://localhost:11311
      
      setting /run_id to b4184c62-bdfa-11ed-a065-48b02d833aa8
      process[rosout-1]: started with pid [129629]
      started core service [/rosout]
      process[read_rgbd-2]: started with pid [129636]
      process[pointcloud_visualizer-3]: started with pid [129637]
      process[display_frames-4]: started with pid [129638]
      process[vr/rgbd_sync-5]: started with pid [129639]
      process[vr/rgbd_odometry-6]: started with pid [129640]
      type is rtabmap_ros/rgbd_sync
      process[vr/rtabmap-7]: started with pid [129641]
      [ INFO] [1678311999.248404989]: Initializing nodelet with 12 worker threads.
      [ INFO] [1678311999.403056922]: Odometry: frame_id               = vr_camera
      [ INFO] [1678311999.403121915]: Odometry: odom_frame_id          = odom
      [ INFO] [1678311999.403144539]: Odometry: publish_tf             = true
      [ INFO] [1678311999.403162939]: Odometry: wait_for_transform     = true
      [ INFO] [1678311999.403187899]: Odometry: wait_for_transform_duration  = 0.100000
      [ INFO] [1678311999.403206203]: Odometry: log_to_rosout_level    = 4
      [ INFO] [1678311999.403270236]: Odometry: initial_pose           = xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000
      [ INFO] [1678311999.403294044]: Odometry: ground_truth_frame_id  = 
      [ INFO] [1678311999.403311677]: Odometry: ground_truth_base_frame_id = vr_camera
      [ INFO] [1678311999.403330045]: Odometry: config_path            = 
      [ INFO] [1678311999.403347485]: Odometry: publish_null_when_lost = true
      [ INFO] [1678311999.403363421]: Odometry: guess_frame_id         = 
      [ INFO] [1678311999.403382013]: Odometry: guess_min_translation  = 0.000000
      [ INFO] [1678311999.403399358]: Odometry: guess_min_rotation     = 0.000000
      [ INFO] [1678311999.403417342]: Odometry: guess_min_time         = 0.000000
      [ INFO] [1678311999.403435742]: Odometry: expected_update_rate   = 0.000000 Hz
      [ INFO] [1678311999.403453342]: Odometry: max_update_rate        = 0.000000 Hz
      [ INFO] [1678311999.403471006]: Odometry: min_update_rate        = 0.000000 Hz
[... 95 lines cut ...]
      [ INFO] [1678312007.246098115]: Odom: quality=323, std dev=0.041772m|0.018867rad, update time=0.242835s
      [ WARN] (2023-03-08 16:46:47.401) util3d.cpp:605::cloudFromDepthRGB() Cloud with only NaN values created!
      [ INFO] [1678312007.439167988]: Odom: quality=382, std dev=0.040896m|0.017944rad, update time=0.187887s
      [ INFO] [1678312007.443841481]: Assembled 0 obstacle and 0 ground clouds (0 points, 0.000010s)
      [ INFO] [1678312007.444371087]: rtabmap (7): Rate=0.03s, Limit=0.000s, Conversion=0.0003s, RTAB-Map=0.2480s, Maps update=0.0242s pub=0.0006s (local map=7, WM=7)
      [ INFO] [1678312007.644015371]: Odom: quality=364, std dev=0.047601m|0.021486rad, update time=0.201133s
      [ WARN] (2023-03-08 16:46:47.708) util3d.cpp:605::cloudFromDepthRGB() Cloud with only NaN values created!
      [ INFO] [1678312007.722803594]: Assembled 0 obstacle and 0 ground clouds (0 points, 0.000011s)
      [ INFO] [1678312007.723373329]: rtabmap (8): Rate=0.03s, Limit=0.000s, Conversion=0.0004s, RTAB-Map=0.2700s, Maps update=0.0063s pub=0.0006s (local map=8, WM=8)
      ^C[vr/rtabmap-7] killing on exit
      [ INFO] [1678312007.838382059]: Odom: quality=341, std dev=0.043961m|0.019672rad, update time=0.190245s
      [vr/rgbd_odometry-6] killing on exit
      [vr/rgbd_sync-5] killing on exit
      [display_frames-4] killing on exit
      [pointcloud_visualizer-3] killing on exit
      [read_rgbd-2] killing on exit
      [ WARN] (2023-03-08 16:46:47.884) util3d.cpp:605::cloudFromDepthRGB() Cloud with only NaN values created!
      [ WARN] (2023-03-08 16:46:48.016) util3d.cpp:605::cloudFromDepthRGB() Cloud with only NaN values created!
      rtabmap: Saving database/long-term memory... (located at /home/valorrobotics/.ros/rtabmap.db)
      rtabmap: Saving database/long-term memory...done! (located at /home/valorrobotics/.ros/rtabmap.db, 3 MB)
      [rosout-1] killing on exit
      [master] killing on exit
      shutting down processing monitor...
      ... shutting down processing monitor complete
      done

JayParikh20, 2023-03-09:
Most of the datasets that are not working have a black border around the edges, which might be causing problem.

JayParikh20, 2023-03-09:
Most of the datasets that are not working have a black border(zero depth) around the edges, which might be causing the issue. Especially, the ones with black border on top and bottom of the image.

matlabbe, 2023-04-16:
Can you provide the launch file for at least rtabmap part?

For TUM RGB-D, there is a tool called `rtabmap-rgbd_dataset` (see --help to associate images) for evaluation. Anyway, I tried with standalone, using camera calibration from here:
https://github.com/introlab/rtabmap/blob/0f221ba3cd9913b68ec508af0d914e3f49860430/tools/RgbdDataset/main.cpp#L190

[image]

It is also possible to use [rgbdslam_datasets.launch](https://github.com/introlab/rtabmap_ros/blob/master/rtabmap_examples/launch/rgbdslam_datasets.launch) file for rtabmap_ros.
 
For the other dataset, I tried also with standalone, but I didn't find where the camera calibration can be downloaded on their [website](http://rgbd-dataset.cs.washington.edu/dataset/rgbd-scenes-v2/), so there is with the TUM fr1 calibration as above:
[image]

closed 2023-04-18 by JayParikh20