introlab/rtabmap_ros #911: how to use the vins as an Odom in rtabmap?

kenan9875645, 2023-03-16:
hello introlab , I'm just a newbee and a small student,  want to learn how to use your great work rtabmap to build my own map .
I just learned using vins-fusion with d435i, it can provide a VIO Odom. And I want to use this odom to guide my rtabmap building a great map. I learned that the rtabmap can build with vins , but I tried a lot, it failed many times, I just gave up.......... I build the rtabmap and rtabmap_ros with apt-get install, and it works well in the euroc dataset. But the rtabmap build with apt is not able to  use the vins together....
in another hand, I find the rtabmap  can recive "Odom" as the input odom to guide the mapping.
So I change the rtabmap-ros launch file as this:
`<launch>

    <arg name="clip_distance"             default="-2"/>
    <arg name="use_rviz"                  default="true"/>
    <arg name="use_rtabmapviz"            default="false"/>
    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
            <arg name="rtabmap_args"       value="--delete_db_on_start"/>
            <arg name="depth_topic"        value="/camera/aligned_depth_to_color/image_raw"/>
            <arg name="frame_id"           value="camera_link"/>
            <arg name="visual_odometry"    value="false"/>
            <arg name="odom_topic"         value="/vins_estimator/odometry"/>
            <arg name="rgb_topic"          value="/camera/color/image_raw"/>
            <arg name="camera_info_topic"  value="/camera/color/camera_info"/>
            <arg name="queue_size"         value="200"/>
            <arg name="rviz"               value="$(arg use_rviz)"/>
            <arg name="rtabmapviz"         value="$(arg use_rtabmapviz)"/>
    </include>

</launch>`
which the vins odom topic is "/vins_estimator/odometry", and just use d435i as my camera.
And I tried run vins and rtabmap-ros together 
I can get  VIO Odom well, but the rtabmap-ros output like this
[image]
and the map is like this:
[image]
 it seems that the tf tree is not set well and  the map can not  fusion together. And in rtabmapviz, it seems that the camera does not move, it holds on in the start point(although rviz the camera is move in the world frame).  
how can I get the correct tf between world and map?   or how to use the vins in the rtabmap? what  the odom frame means about?

matlabbe, 2023-04-02:
Hi,

The `frame_id` of `rtabmap` should be set to `child_frame_id` set in vins odometry topic (`/vins_estimator/odometry`).

Which frames VINS is published? Can you show the tf tree? `rosrun tf2_tools view_frames.py`

kenan9875645, 2023-04-03:
> 

Hello! I just learned that it may be the tf_tree not connect correctly.
After running **rosrun tf2_ tools view_ Frames.py**, the following results appear:
[image]
I think it should be connected inside, but I don't know why this picture is separated inside.
And this time, I run with the d455 (the effect should be the same as the d435i), and then after changing the value of **frame_id** to **body**,  with the pkg static_transformer node like this:
`<launch>
    <node pkg="tf" type="static_transform_publisher" name="camera2camera_link_frame1" args="0 0 0 0 0 0 camera camera_link 1000"/>
    <arg name="clip_distance"             default="-2"/>
    <arg name="use_rviz"                  default="true"/>
    <arg name="use_rtabmapviz"            default="false"/>
    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
            <arg name="rtabmap_args"       value="--delete_db_on_start"/>
            <arg name="depth_topic"        value="/camera/aligned_depth_to_color/image_raw"/>
            <arg name="frame_id"           value="body"/>
            <arg name="visual_odometry"    value="false"/>
            <arg name="odom_topic"         value="/vins_estimator/odometry"/>
            <arg name="rgb_topic"          value="/camera/color/image_raw"/>
            <arg name="camera_info_topic"  value="/camera/color/camera_info"/>
            <arg name="queue_size"         value="200"/>
            <arg name="rviz"               value="$(arg use_rviz)"/>
            <arg name="rtabmapviz"         value="$(arg use_rtabmapviz)"/>
    </include>
  
</launch>
`
  ,the result of running rtabmap_vins is shown in the left figure:
[image]
There is no yellow warning.
The content of the launch file here is as follows:
[image]
However, it seems that the desired effect has not been achieved, which means that the established diagrams are still not well integrated. is there something wrong?

kenan9875645, 2023-04-03:
> Hi,
> 
> The `frame_id` of `rtabmap` should be set to `child_frame_id` set in vins odometry topic (`/vins_estimator/odometry`).
> 
> Which frames VINS is published? Can you show the tf tree? `rosrun tf2_tools view_frames.py`

Hello! I just learned that it may be the tf_tree not connect correctly.
After running **rosrun tf2_ tools view_ Frames.py**, the following results appear:
[image]
I think it should be connected inside, but I don't know why this picture is separated inside.
And this time, I run with the d455 (the effect should be the same as the d435i), and then after changing the value of **frame_id** to **body**,  with the pkg static_transformer node like this(The content of the launch file here is as follows:):
`
<launch>
    <node pkg="tf" type="static_transform_publisher" name="camera2camera_link_frame1" args="0 0 0 0 0 0 camera camera_link 1000"/>
    <arg name="clip_distance"             default="-2"/>
    <arg name="use_rviz"                  default="true"/>
    <arg name="use_rtabmapviz"            default="false"/>
    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
            <arg name="rtabmap_args"       value="--delete_db_on_start"/>
            <arg name="depth_topic"        value="/camera/aligned_depth_to_color/image_raw"/>
            <arg name="frame_id"           value="body"/>
            <arg name="visual_odometry"    value="false"/>
            <arg name="odom_topic"         value="/vins_estimator/odometry"/>
            <arg name="rgb_topic"          value="/camera/color/image_raw"/>
            <arg name="camera_info_topic"  value="/camera/color/camera_info"/>
            <arg name="queue_size"         value="200"/>
            <arg name="rviz"               value="$(arg use_rviz)"/>
            <arg name="rtabmapviz"         value="$(arg use_rtabmapviz)"/>
    </include>
  
</launch>
`
  ,the result of running rtabmap_vins is shown in the left figure:
[image]
There is no yellow warning.

[image]
However, it seems that the desired effect has not been achieved, which means that the established diagrams are still not well integrated. is there something wrong?

kenan9875645, 2023-04-03:
> Hi,
> 
> The `frame_id` of `rtabmap` should be set to `child_frame_id` set in vins odometry topic (`/vins_estimator/odometry`).
> 
> Which frames VINS is published? Can you show the tf tree? `rosrun tf2_tools view_frames.py`

Hello! I just learned that it may be the tf_tree not connect correctly.
After running **rosrun tf2_ tools view_ Frames.py**, the following results appear:
/home/chuzhaoxue/图片/2023-04-03 20-25-17 的屏幕截图.png
I think it should be connected inside, but I don't know why this picture is separated inside.
And this time, I run with the d455 (the effect should be the same as the d435i), and then after changing the value of **frame_id** to **body**, with the pkg static_transformer node like this(The content of the launch file here is as follows:):

 ,the result of running rtabmap_vins is shown in the left figure:
/home/chuzhaoxue/图片/2023-04-03 20-27-06 的屏幕截图.png
There is no yellow warning.

/home/chuzhaoxue/图片/2023-04-03 20-20-58 的屏幕截图.png
However, it seems that the desired effect has not been achieved, which means that the established diagrams are still not well integrated. is there something wrong?

kenan9875645, 2023-04-03:
> Hi,
> 
> The `frame_id` of `rtabmap` should be set to `child_frame_id` set in vins odometry topic (`/vins_estimator/odometry`).
> 
> Which frames VINS is published? Can you show the tf tree? `rosrun tf2_tools view_frames.py`

Hello! I just learned that it may be the tf_tree not connect correctly.
After running rosrun tf2_ tools view_ Frames.py, the following results appear:
[image]

I think it should be connected inside, but I don't know why this picture is separated inside.
And this time, I run with the d455 (the effect should be the same as the d435i), and then after changing the value of frame_id to body, with the pkg static_transformer node like this(The content of the launch file here is as follows:):
**<launch>
    <node pkg="tf" type="static_transform_publisher" name="camera2camera_link_frame1" args="0 0 0 0 0 0 camera camera_link 1000"/>
    <arg name="clip_distance"             default="-2"/>
    <arg name="use_rviz"                  default="true"/>
    <arg name="use_rtabmapviz"            default="false"/>
    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">
            <arg name="rtabmap_args"       value="--delete_db_on_start"/>
            <arg name="depth_topic"        value="/camera/aligned_depth_to_color/image_raw"/>
            <arg name="frame_id"           value="body"/>
            <arg name="visual_odometry"    value="false"/>
            <arg name="odom_topic"         value="/vins_estimator/odometry"/>
            <arg name="rgb_topic"          value="/camera/color/image_raw"/>
            <arg name="camera_info_topic"  value="/camera/color/camera_info"/>
            <arg name="queue_size"         value="200"/>
            <arg name="rviz"               value="$(arg use_rviz)"/>
            <arg name="rtabmapviz"         value="$(arg use_rtabmapviz)"/>
    </include>
  
</launch>**

,the result of running rtabmap_vins is shown in the left figure:
[image]

There is no yellow warning.

[image]

However, it seems that the desired effect has not been achieved, which means that the established diagrams are still not well integrated. is there something wrong?

matlabbe, 2023-04-16:
One issue sometimes with other packages is that they provide odometry in camera optical frame (x->left, y->down, z->forward) instead of ros standard (x->forward, y->left, z->up). Even before trying rtabmap, you could launch VINS alone, open RVIZ, add a DepthCloud display, set an history of some seconds and move the camera to make sure the point clouds overlap correctly.

tulingcheng86, 2024-03-12:
你好，能分享下怎么将vins作为odom在rtabmap中使用吗，我也用的是d435i相机，谢谢(ಥ_ಥ)

Wyp20, 2025-09-08:
When using the IMU, the performance of the integrated VINS is worse than the original VINS. Therefore, I also used VINS as the odom (odometry) and encountered the same issue. Is there a solution now?

Wyp20, 2025-09-10:
I am using VINS-Fusion, and my Ubuntu version is 20.04. I have already resolved the issues I mentioned above. The following four key steps need to be completed:

1. ​Modify the frame_id and child_frame_id of /vins_estimator/odometry to match those in /rtabmap/odom.​​
2. ​Set multiple_thread to 0 in the euroc_stereo_imu_config.yaml file.​​
3. ​Set approx_sync to true in your custom RTAB-Map launch file.​
4. ​Set the parameter imu_base_link in your custom RTAB-Map launch file to "0 0 0 0 0 0 base_link imu4 100".​