introlab/rtabmap_ros #916: colcon build with "colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release"

takahiko-hasegawa-hdjp, 2023-03-23:
I have two packages: rtabmap & rtabmap_ros (ros2 foxy) on my local machine and trying to build them using colcon build.
However, it always takes about 30 mins & my pc freezes (unable to move cursor and use my pc at all). How can I resolve this problem?

I am also using --parallel-workers option too but no luck/

matlabbe, 2023-03-23:
You are maybe filling your RAM and your SWAP memory, then the computer freezes. Build one file at a time:
```
export MAKEFLAGS="-j1" 
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
```

takahiko-hasegawa-hdjp, 2023-03-23:
Unfortunately still the same issue occurs after making MKAEFLAGS to 1

matlabbe, 2023-03-23:
How much RAM do you have? Some files may require 2GB of ram to compile.

matlabbe, 2023-03-24:
Should be fine, can you see the RAM filling up when starting the build?

takahiko-hasegawa-hdjp, 2023-03-26:
screen will freeze so I can't even check my RAM usage. Even if i do --packages-select, the same issue occurs.

matlabbe, 2023-03-27:
If the cmake phase is done, you may go under `ros_ws/build/rtabmap`, then do just `make`, this will launch only 1 compilation thread for only this package.

hodnajit, 2023-04-11:
You can try to use sequential executor:

`colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release --executor sequential`