introlab/rtabmap_ros #917: tried gdb; [rgbd_odometry-1]: process has died [pid 10089, exit code -11...

gxywang, 2023-03-23:
Hi,

I've been trying to get rtabmap running for a while, and am running into a "process has died" issue. Tried following #28, but was unable to figure out the issue.

Details:
- Camera: Intel Realsense D455
- Computer: Asus TUF Dash F15
- OS: Ubuntu 20 (ROS 2 Galactic)

ROS 2 commands run:
```
# Realsense:
ros2 launch realsense2_camera rs_launch.py \
   enable_accel:=true \
   enable_gyro:=true \
   unite_imu_method:=2

# IMU filter
ros2 run imu_filter_madgwick imu_filter_madgwick_node --ros-args -p use_mag:=false -r /imu/data_raw:=/camera/imu

# Rtabmap
ros2 launch rtabmap_ros rtabmap.launch.py \
    rtabmap_args:="--delete_db_on_start" \
    frame_id:=camera_link \
    rgb_topic:=/camera/color/image_raw \
    depth_topic:=/camera/depth/image_rect_raw \
    camera_info_topic:=/camera/color/camera_info \
    approx_sync:=true \
    wait_imu_to_init:=true \
    imu_topic:=/imu/data \
    rviz:=true
```

Current TF tree:
[image]

Topics are being published (ros2 topic hz):
[image]

Published camera topics as seen in Rviz2:
[image]

Errors (running normally):
```
iris@iris-laptop:~$ ros2 launch rtabmap_ros rtabmap.launch.py     rtabmap_args:="--delete_db_on_start"     frame_id:=camera_link     rgb_topic:=/camera/color/image_raw     depth_topic:=/camera/depth/image_rect_raw     camera_info_topic:=/camera/color/camera_info     approx_sync:=true     wait_imu_to_init:=true     imu_topic:=/imu/data     rviz:=true
[INFO] [launch]: All log files can be found below /home/iris/.ros/log/2023-03-23-18-06-27-432138-iris-laptop-10087
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rgbd_odometry-1]: process started with pid [10089]
[INFO] [rtabmap-2]: process started with pid [10091]
[INFO] [rtabmapviz-3]: process started with pid [10093]
[INFO] [rviz2-4]: process started with pid [10095]
[INFO] [point_cloud_xyzrgb-5]: process started with pid [10097]
[point_cloud_xyzrgb-5] [INFO] [1679612788.035928137] [point_cloud_xyzrgb]: Approximate time sync = true
[rgbd_odometry-1] [INFO] [1679612788.044947630] [rtabmap.rgbd_odometry]: Odometry: frame_id               = camera_link
[rgbd_odometry-1] [INFO] [1679612788.045041032] [rtabmap.rgbd_odometry]: Odometry: odom_frame_id          = odom
[rgbd_odometry-1] [INFO] [1679612788.045052074] [rtabmap.rgbd_odometry]: Odometry: publish_tf             = true
[rgbd_odometry-1] [INFO] [1679612788.045058182] [rtabmap.rgbd_odometry]: Odometry: wait_for_transform     = 0.310000
[rgbd_odometry-1] [INFO] [1679612788.045071983] [rtabmap.rgbd_odometry]: Odometry: log_to_rosout_level    = 4
[rgbd_odometry-1] [INFO] [1679612788.045094251] [rtabmap.rgbd_odometry]: Odometry: initial_pose           = xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000
[rgbd_odometry-1] [INFO] [1679612788.045100777] [rtabmap.rgbd_odometry]: Odometry: ground_truth_frame_id  = 
[rgbd_odometry-1] [INFO] [1679612788.045106597] [rtabmap.rgbd_odometry]: Odometry: ground_truth_base_frame_id = 
[rgbd_odometry-1] [INFO] [1679612788.045112719] [rtabmap.rgbd_odometry]: Odometry: config_path            = 
[rgbd_odometry-1] [INFO] [1679612788.045118082] [rtabmap.rgbd_odometry]: Odometry: publish_null_when_lost = true
[rgbd_odometry-1] [INFO] [1679612788.045123871] [rtabmap.rgbd_odometry]: Odometry: guess_frame_id         = 
[rgbd_odometry-1] [INFO] [1679612788.045129131] [rtabmap.rgbd_odometry]: Odometry: guess_min_translation  = 0.000000
[rgbd_odometry-1] [INFO] [1679612788.045135621] [rtabmap.rgbd_odometry]: Odometry: guess_min_rotation     = 0.000000
[rgbd_odometry-1] [INFO] [1679612788.045143708] [rtabmap.rgbd_odometry]: Odometry: guess_min_time         = 0.000000
[rgbd_odometry-1] [INFO] [1679612788.045150286] [rtabmap.rgbd_odometry]: Odometry: expected_update_rate   = 0.000000 Hz
[rgbd_odometry-1] [INFO] [1679612788.045156269] [rtabmap.rgbd_odometry]: Odometry: max_update_rate        = 0.000000 Hz
[rgbd_odometry-1] [INFO] [1679612788.045161734] [rtabmap.rgbd_odometry]: Odometry: min_update_rate        = 0.000000 Hz
[... 130 lines cut ...]
Errors (running in debug mode):
```
iris@iris-laptop:~$ ros2 launch rtabmap_ros rtabmap.launch.py     rtabmap_args:="--delete_db_on_start"     frame_id:=camera_link     rgb_topic:=/camera/color/image_raw     depth_topic:=/camera/depth/image_rect_raw     camera_info_topic:=/camera/color/camera_info     approx_sync:=true     wait_imu_to_init:=true     imu_topic:=/imu/data     rviz:=true     rtabmap_args:="-d --udebug"     launch_prefix:="xterm -e gdb -ex run --args"
[INFO] [launch]: All log files can be found below /home/iris/.ros/log/2023-03-23-18-10-56-136730-iris-laptop-10428
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rgbd_odometry-1]: process started with pid [10430]
[INFO] [rtabmap-2]: process started with pid [10432]
[INFO] [rtabmapviz-3]: process started with pid [10434]
[INFO] [rviz2-4]: process started with pid [10436]
[INFO] [point_cloud_xyzrgb-5]: process started with pid [10438]
[point_cloud_xyzrgb-5] [INFO] [1679613056.720825151] [point_cloud_xyzrgb]: Approximate time sync = true
[rviz2-4] [INFO] [1679613056.769993988] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [INFO] [1679613056.770056953] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-4] [INFO] [1679613056.785040401] [rviz2]: Stereo is NOT SUPPORTED
[point_cloud_xyzrgb-5] terminate called after throwing an instance of 'cv::Exception'
[point_cloud_xyzrgb-5]   what():  OpenCV(4.7.0) /home/iris/orb_slam3/opencv-4.7.0/modules/core/src/alloc.cpp:73: error: (-4:Insufficient memory) Failed to allocate 104297171071440 bytes in function 'OutOfMemoryError'
[point_cloud_xyzrgb-5] 
[ERROR] [point_cloud_xyzrgb-5]: process has died [pid 10438, exit code -6, cmd '/home/iris/colcon_ws/install/rtabmap_ros/lib/rtabmap_ros/point_cloud_xyzrgb --ros-args --params-file /tmp/launch_params_6vytua3e --params-file /tmp/launch_params__0l9toxq -r left/image:=/stereo_camera/left/image_rect_color -r right/image:=/stereo_camera/right/image_rect -r left/camera_info:=/stereo_camera/left/camera_info -r right/camera_info:=/stereo_camera/right/camera_info -r rgb/image:=/camera/color/image_raw -r depth/image:=/camera/depth/image_rect_raw -r rgb/camera_info:=/camera/color/camera_info -r rgbd_image:=rgbd_image_relay -r cloud:=voxel_cloud'].
```
GDB for rgbd_odometry:
[image]
[image]
[image]

Thanks!

matlabbe, 2023-03-24:
That error is impressive:
```
Failed to allocate 151885209101040 bytes
```
151 TeraBytes!

I think you are mixing multiple OpenCV versions. `cv_bridge` is using opencv binaries (on focal it is 4.2), and you also built and installed OpenCV 4.7. In the log, we see it crashing in `cv_bridge`, while some opencv 4.7 libraries have been loaded on runtime.

The issue is explained [here](https://github.com/introlab/rtabmap_ros#build-from-source), which also applies to any ros versions: 

> On Melodic/Noetic, build from source with xfeatures2d module (and nonfree module if needed) the same OpenCV version already installed on the system. You will then avoid breaking cv_bridge with rtabmap_ros. If you want to install a more recent OpenCV version, I recommend to uninstall libopencv* libraries (with all ros packages depending on it) and rebuild all those ros packages in your catkin workspace (to make sure cv_bridge is linked on the OpenCV version you just compiled).

cheers,
Mathieu

gxywang, 2023-03-25:
Just to clarify, if I need OpenCV 4.7, I should run ```sudo apt remove libopencv-dev*```, and ```sudo apt autoremove libopencv-dev*```? And then rebuild my colcon workspace that contains OpenCV 4.7. 

When running the first with the ```--simulate``` flag, it removes all the following:
```
libcv-bridge-dev 
libopencv-apps-dev 
libopencv-dev 
ros-galactic-cv-bridge
ros-galactic-depthimage-to-laserscan 
ros-galactic-image-geometry 
ros-galactic-image-tools
ros-galactic-intra-process-demo
```
I just wanted to make sure that is expected behavior.

matlabbe, 2023-03-25:
Yes, this what that text said:

> I recommend to uninstall libopencv* libraries (with all ros packages depending on it) and rebuild all those ros packages in your catkin workspace (to make sure cv_bridge is linked on the OpenCV version you just compiled).

I agree it is unfortunate, but as you can see that issue was also there on ros1.

gxywang, 2023-03-26:
Fixed! Ended up deleting local version of OpenCV 4.7 and reinstalling & rebuilding the whole system. Thank you!

closed 2023-03-26 by gxywang