introlab/rtabmap_ros #919: rtabmap_ros tutorial example on L515 and Nvidia AGX not working

araujokth, 2023-03-24:
Hi, I am trying to run the rtabmap_ros on an Nvidia AGX and L515. I am using the binaries.

I successfully run the tutorial example http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping (Section 1) to test the camera and it works.

When I run the example from Section 2 on mapping mode for realsense D435i/L515 I get the error I paste below. Can you see any hint of where the problem could be? I also tested to check if the topics from the camera are being published via rostopic hz /camera/color/image_raw /camera/aligned_depth_to_color/image_raw /camera/color/camera_info
and they are not.

Thank you!!

roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start --Optimizer/GravitySigma 0.3" depth_topic:=/camera/aligned_depth_to_color/image_raw rgb_topic:=/camera/color/image_raw camera_info_topic:=/camera/color/camera_info approx_sync:=false wait_imu_to_init:=true imu_topic:=/rtabmap/imu rviz:=true rtabmapviz:=false
... logging to /home/ejoaraj/.ros/log/f8c3fd0e-ca55-11ed-bb61-48b02d2b8048/roslaunch-lab-agx-3-24595.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lab-agx-3:33213/

SUMMARY
CLEAR PARAMETERS

/points_xyzrgb/
/rtabmap/rgbd_odometry/
/rtabmap/rtabmap/
PARAMETERS

/points_xyzrgb/approx_sync: False
/points_xyzrgb/approx_sync_max_interval: 0.0
/points_xyzrgb/decimation: 4.0
/points_xyzrgb/voxel_size: 0.0
/rosdistro: melodic
/rosversion: 1.14.12
/rtabmap/rgbd_odometry/approx_sync: False
/rtabmap/rgbd_odometry/approx_sync_max_interval: 0.0
/rtabmap/rgbd_odometry/config_path:
/rtabmap/rgbd_odometry/expected_update_rate: 0.0
/rtabmap/rgbd_odometry/frame_id: camera_link
/rtabmap/rgbd_odometry/ground_truth_base_frame_id:
/rtabmap/rgbd_odometry/ground_truth_frame_id:
/rtabmap/rgbd_odometry/guess_frame_id:
/rtabmap/rgbd_odometry/guess_min_rotation: 0.0
/rtabmap/rgbd_odometry/guess_min_translation: 0.0
/rtabmap/rgbd_odometry/keep_color: False
/rtabmap/rgbd_odometry/max_update_rate: 0.0
/rtabmap/rgbd_odometry/odom_frame_id: odom
/rtabmap/rgbd_odometry/publish_tf: True
/rtabmap/rgbd_odometry/queue_size: 10
/rtabmap/rgbd_odometry/subscribe_rgbd: False
/rtabmap/rgbd_odometry/wait_for_transform_duration: 0.2
/rtabmap/rgbd_odometry/wait_imu_to_init: True
/rtabmap/rtabmap/Mem/IncrementalMemory: true
/rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
/rtabmap/rtabmap/approx_sync: False
/rtabmap/rtabmap/config_path:
/rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
/rtabmap/rtabmap/frame_id: camera_link
/rtabmap/rtabmap/gen_depth: False
/rtabmap/rtabmap/gen_depth_decimation: 1
/rtabmap/rtabmap/gen_depth_fill_holes_error: 0.1
/rtabmap/rtabmap/gen_depth_fill_holes_size: 0
/rtabmap/rtabmap/gen_depth_fill_iterations: 1
/rtabmap/rtabmap/gen_scan: False
/rtabmap/rtabmap/ground_truth_base_frame_id:
/rtabmap/rtabmap/ground_truth_frame_id:
/rtabmap/rtabmap/initial_pose:
/rtabmap/rtabmap/landmark_angular_variance: 9999.0
/rtabmap/rtabmap/landmark_linear_variance: 0.0001
/rtabmap/rtabmap/map_frame_id: map
/rtabmap/rtabmap/odom_frame_id:
/rtabmap/rtabmap/odom_frame_id_init:
/rtabmap/rtabmap/odom_sensor_sync: False
[... 98 lines cut ...]
[ INFO] [1679670620.695521288]: /rtabmap/rtabmap: subscribe_depth = true
[ INFO] [1679670620.696437714]: /rtabmap/rtabmap: subscribe_rgb = true
[ INFO] [1679670620.696669149]: /rtabmap/rtabmap: subscribe_stereo = false
[ INFO] [1679670620.696802627]: /rtabmap/rtabmap: subscribe_rgbd = false (rgbd_cameras=1)
[ INFO] [1679670620.696845829]: /rtabmap/rtabmap: subscribe_odom_info = true
[ INFO] [1679670620.696906984]: /rtabmap/rtabmap: subscribe_user_data = false
[ INFO] [1679670620.696967531]: /rtabmap/rtabmap: subscribe_scan = false
[ INFO] [1679670620.697147155]: /rtabmap/rtabmap: subscribe_scan_cloud = false
[ INFO] [1679670620.697447105]: /rtabmap/rtabmap: subscribe_scan_descriptor = false
[ INFO] [1679670620.697514948]: /rtabmap/rtabmap: queue_size = 10
[ INFO] [1679670620.697557702]: /rtabmap/rtabmap: approx_sync = false
[ INFO] [1679670620.697695148]: Setup depth callback
[ INFO] [1679670620.772383987]:
/rtabmap/rtabmap subscribed to (exact sync):
/rtabmap/odom
/camera/color/image_raw
/camera/aligned_depth_to_color/image_raw
/camera/color/camera_info
/rtabmap/odom_info
[ INFO] [1679670621.611931354]: rtabmap 0.20.23 started...
[ WARN] [1679670624.113684806]: /rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rgbd_odometry subscribed to (exact sync):
/camera/color/image_raw
/camera/aligned_depth_to_color/image_raw
/camera/color/camera_info

matlabbe, 2023-04-02:
That's odd, you may try setting `approx_sync` to true, however if it works, that means the topics don't share the same stamp, which would be a bug on realsense side.

araujokth, 2023-05-04:
Sorry, this was my fault. I didnt start the camera correctly. Solved now

closed 2023-05-04 by araujokth