introlab/rtabmap_ros #923: Could we use optical flow instead of feature based for RGBD input

takahiko-hasegawa-hdjp, 2023-03-28:
I have RGBD input and would like to use optical flow but changing vis/CorType does not seem to be doing its job.
Could we use optical flow for RGBD and if so how?

matlabbe, 2023-04-02:
You have to change `Odometry/Strategy` to 1 too.

takahiko-hasegawa-hdjp, 2023-04-04:
Ok, thats great thanks. 
It doesnt seem to affect the rtabmapviz tho? do I need to change any parameters? 

I can see that in RegistrationVis.cpp if(_correspondencesApproach == 0 ) checks the corType and seems to be going inside the if statetment but somehow reaches "Optical Flow correspondences requires image in data!"

matlabbe, 2023-04-16:
Set these parameters only for rgbd_odometry node, not rtabmap node. And yes, you won't see the changes in rtabmapviz, as rtabmapviz shows parameters from rtabmap node, not rgbd_odometry.

"Optical Flow correspondences requires image in data!", would have to see more logs. You can launch rgbd_odometry with --udebug argument, or call `rosservice call rgbd_odometry/set_log_debug` (adjust namespace if necessary).

takahiko-hasegawa-hdjp, 2023-06-26:
I suppose this video https://www.youtube.com/watch?v=xIGKaE_rZ_Q shows that result of optical flow can be visualized in  the rtabmap viz?

matlabbe, 2023-06-26:
Yes, set `subscribe_odom_info` to true for rtabmap_viz to have the optical flow result from odometry node.