introlab/rtabmap_ros #925: build error on humble

H-tr, 2023-03-29:
OS: ubuntu22.04
ROS version: ROS2 humble

Hi, I just follow the install instructions to build the rtabmap_ros package in ros2_ws. 

Then there is a CMake Error
```
CMake Error at /path_to_ws/ros2_ws/install/rtabmap_conversions/share/rtabmap_conversions/cmake/rtabmap_conversionsExport.cmake:61 (set_target_properties):
  The link interface of target "rtabmap_conversions::rtabmap_conversions"
  contains:

    rtabmap::utilite

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /path_to_ws/ros2_ws/install/rtabmap_conversions/share/rtabmap_conversions/cmake/ament_cmake_export_targets-extras.cmake:9 (include)
  /path_to_ws/ros2_ws/install/rtabmap_conversions/share/rtabmap_conversions/cmake/rtabmap_conversionsConfig.cmake:41 (include)
  CMakeLists.txt:10 (find_package)
```

matlabbe, 2023-04-02:
How `rtabmap` has been built? It seems you are maybe using an old `rtabmap` branch with new `rtabmap_ros` branch.

H-tr, 2023-04-04:
Hi, thanks for your reply.
You are right, I need to set the branch to humble. That is my fault.

H-tr, 2023-04-11:
Hi, sorry that I met the same issue. I followed the instructions exactly, using the latest version of rtabmap and ros2 branch of rtabmap-ros.
```
cd ~/ros2_ws
git clone https://github.com/introlab/rtabmap.git src/rtabmap
git clone --branch ros2 https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros
rosdep update && rosdep install --from-paths src --ignore-src -r -y
export MAKEFLAGS="-j6" # Can be ignored if you have a lot of RAM (>16GB)
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
```
I found the rtabmap updated yesterday and 3 weeks before separatly. I'm wondering is it because the rtabmap is too new?

matlabbe, 2023-04-16:
What is the error? maybe rtabmap_ros is finding an older rtabmap versions installed somewhere else... humble CI works: https://github.com/introlab/rtabmap_ros/actions/runs/4652852998/jobs/8233294645

closed 2023-04-18 by H-tr