introlab/rtabmap_ros #926: Loop closure is not detected but map->odom is not an identity matrix

Hypothesis-Z, 2023-03-29:
I have set Vis/MinInliers to a large number so that no loop closure is accepted. The transform between map and odom should be an identity matrix because RTAB-Map won't do any global optimization.  But the fact is not what thought to be. Why?

[image]

Hypothesis-Z, 2023-03-30:
It was because I fed imu data to RTAB-Map node. Close the issue.

closed 2023-03-30 by Hypothesis-Z

matlabbe, 2023-04-02:
To disable visual loop closure detection, to most efficient way is to set `Kp/MaxFeatures=-1` to just not extract any features.

Hypothesis-Z, 2023-04-10:
@matlabbe Thank you for advice. But if no feature is extracted, how to generate visual odometry? Don't we need features to compute PnP and BA?

matlabbe, 2023-04-16:
`Kp/MaxFeatures=-1` disable loop closure detection, not odometry.