introlab/rtabmap_ros #928: External Stereo Image Proc support

miscott, 2023-03-30:
We are interested in using an external instance of a stereo image-based point cloud generation node to create the depth maps utilized by rtambmap. Our motivation is that 1) we have a GPU accelerated instance of stereo point cloud generation that we'd like to use over the call currently utilized in rtabmap_ros and 2) we are running on an edge device and would like to reduce the computational overhead, and this depth data is already being generated. 

It doesn't look like rtabmap_ros is currently built to allow for external stereo-based point cloud/depth map subscription? We recognize there is an RGB-D-based subscription, but our setup utilizes stereo cameras which will not produce depth maps of the same quality as an RGB-D sensor, and we therefore don't believe this is the appropriate callback (although we may be mistaken here). 

If this feature does not currently exist, we believe we have a way to add it by creating a new sync callback that subscribes to a point cloud/depth map topic and overriding the disparity created by rtabmap_ros internally (i.e., replacing [this](https://github.com/introlab/rtabmap_ros/blob/master/rtabmap_slam/src/CoreWrapper.cpp#L1324) call with our own disparity map). Are there any issues with overriding that function that we're missing?

matlabbe, 2023-04-02:
I would use RGB-D mode instead, with depth image as the output of your stereo disparity approach. The code your are pointing is actually the same than if we use `stereo_img_proc` node to produce the depth image (which uses OpencV StereoBM) and use RGB-D mode for rtabmap node instead of stereo. Just make sure to set `approx_sync` to false so that the correct depth image is sync with the image.

miscott, 2023-04-03:
Thanks @matlabbe, we'll try that solution!

closed 2023-04-03 by miscott