introlab/rtabmap_ros #930: double free or corruption (out) - Ubuntu 20.04, ROS Noetic

umbnich, 2023-04-01:
Hi, 
I tried different times to build from source. 
Launching: 
`roslaunch rtabmap_ros demo_stereo_outdoor.launch`
What I get is:
`[rtabmap/stereo_odometry-5] process has died [pid 3096, exit code -6, cmd /home/umbnich/catkin_ws/devel/lib/rtabmap_ros/stereo_odometry left/image_rect:=/stereo_camera/left/image_rect right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info_throttle right/camera_info:=/stereo_camera/right/camera_info_throttle rgbd_image:=/stereo_camera/rgbd_image odom:=/stereo_odometry __name:=stereo_odometry __log:=/home/umbnich/.ros/log/959783e6-d0a3-11ed-a761-00155dca3f64/rtabmap-stereo_odometry-5.log].
log file: /home/umbnich/.ros/log/959783e6-d0a3-11ed-a761-00155dca3f64/rtabmap-stereo_odometry-5*.log
[rtabmap/rtabmap-6] process has died [pid 3098, exit code -6, cmd /home/umbnich/catkin_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info_throttle right/camera_info:=/stereo_camera/right/camera_info_throttle rgbd_image:=/stereo_camera/rgbd_image odom:=/stereo_odometry __name:=rtabmap __log:=/home/umbnich/.ros/log/959783e6-d0a3-11ed-a761-00155dca3f64/rtabmap-rtabmap-6.log].
log file: /home/umbnich/.ros/log/959783e6-d0a3-11ed-a761-00155dca3f64/rtabmap-rtabmap-6*.log
`
Same launching the example of rtabmap from zed_ros_wrapper:
`double free or corruption (out)
[zed/rtabmap-5] process has died [pid 2535, exit code -6, cmd /home/umbnich/catkin_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start rgb/image:=zed_node/rgb/image_rect_color depth/image:=zed_node/depth/depth_registered rgb/camera_info:=zed_node/rgb/camera_info grid_map:=map odom:=zed_node/odom __name:=rtabmap __log:=/home/umbnich/.ros/log/8255a73c-d0a2-11ed-a1e9-00155dca3f64/zed-rtabmap-5.log].
log file: /home/umbnich/.ros/log/8255a73c-d0a2-11ed-a1e9-00155dca3f64/zed-rtabmap-5*.log
double free or corruption (out)
[zed/rtabmapviz-4] process has died [pid 2529, exit code -6, cmd /home/umbnich/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz rgb/image:=zed_node/rgb/image_rect_color depth/image:=zed_node/depth/depth_registered rgb/camera_info:=zed_node/rgb/camera_info grid_map:=map odom:=zed_node/odom __name:=rtabmapviz __log:=/home/umbnich/.ros/log/8255a73c-d0a2-11ed-a1e9-00155dca3f64/zed-rtabmapviz-4.log].
log file: /home/umbnich/.ros/log/8255a73c-d0a2-11ed-a1e9-00155dca3f64/zed-rtabmapviz-4*.log
`

Tried installing both following the guide and using the noetic-devel branch.

matlabbe, 2023-04-02:
See section A of https://github.com/introlab/rtabmap_ros/issues/28#issue-99654759

umbnich, 2023-04-04:
I tried running:
`roslaunch rtabmap_ros rtabmap.launch launch_prefix:="xterm -e gdb -ex run --args"`
but gdb doesn't start.

matlabbe, 2023-04-16:
Do you have more than one OpenCV installed on the computer?

umbnich, 2023-04-17:
No, just OpenCV 4.2
(checked with dpkgs -l  | grep libopencv).

Anyway I managed to fix everything but had to install a previous version of my WSL due to a bug with the USB cam. 
Now what I get after tried to reinstall rtabmap and run zed-rtabmap example is:

`[zed/rtabmap-5] process has died [pid 1622, exit code -6, cmd /home/umbnich/catkin_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start rgb/image:=zed_node/rgb/image_rect_color depth/image:=zed_node/depth/depth_registered rgb/camera_info:=zed_node/rgb/camera_info grid_map:=map odom:=zed_node/odom __name:=rtabmap __log:=/home/umbnich/.ros/log/33047f24-dcf9-11ed-b95e-00155d49293b/zed-rtabmap-5.log].`

and 

`[zed/rtabmapviz-4] process has died [pid 1621, exit code 127, cmd /home/umbnich/catkin_ws/devel/lib/rtabmap_ros/rtabmapviz rgb/image:=zed_node/rgb/image_rect_color depth/image:=zed_node/depth/depth_registered rgb/camera_info:=zed_node/rgb/camera_info grid_map:=map odom:=zed_node/odom __name:=rtabmapviz __log:=/home/umbnich/.ros/log/33047f24-dcf9-11ed-b95e-00155d49293b/zed-rtabmapviz-4.log].
log file: /home/umbnich/.ros/log/33047f24-dcf9-11ed-b95e-00155d49293b/zed-rtabmapviz-4*.log`

Can you help me?

matlabbe, 2023-04-20:
Can you try first `ros-noetic-rtabmap-ros` after cleaning up your `catkin_ws/devel` folder?