introlab/rtabmap_ros #932: An error occurred when I was running the EuRoc dataset with orbslam3 as the odometer

albdlackzrc, 2023-04-02:
Hi @matlabbe !
An error occurred when I used orbslam3 as the odometry to run EuRoc data set, I tried V1_01_easy.bag, MH_01_easy.bag, MH_02_easy and the same error occurred. How can I solve this problem?
[image]

matlabbe, 2023-04-02:
You may launch `stereo_odometry` in gdb, and do `bt` (backtrace) when it crashes. This would give more information where it fails. Note that I am not actively maintaining ORB-SLAM3 support, the last working version I tried is the one referred in this patch: https://gist.github.com/matlabbe/f5cb281304a1305b2824a6ce19792e13

albdlackzrc, 2023-04-06:
Sorry I'm not familiar with GDB and ROS, but this is what rtabmap_ws/devel/lib/stereo_odometry looked like when I opened it with GDB.
[image]
So how do I get this up and running, thank you for your answer

matlabbe, 2023-04-16:
If you are using a launch file to launch the nodes, you can enable gdb with `launch-prefix` attribute: http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB

zhangpeng66, 2023-06-13:
[image]
Me too，I use gdb when it crash

matlabbe, 2023-06-17:
We cannot see anything in that screenshot, copy/paste the text instead after you do `bt` (backtrace).