introlab/rtabmap_ros #933: Unknown odometry

SiinaIng, 2023-04-02:
Hello,

What odometry is this launch configuration trying to use? I know it's ICP, but is it from Lidar or depth scans?

`roslaunch rtabmap_ros rtabmap.launch stereo:=false \ rgb_topic:=/zed2/zed_node/rgb/image_rect_color \ camera_info_topic:=/zed2/zed_node/rgb/camera_info use_sim_time:=true \ depth_topic:=/zed2/zed_node/depth/depth_registered \ depth_camera_info_topic:=/zed2/zed_node/depth/camera_info \ frame_id:=base_link \ subscribe_scan_cloud:=true \ rgbd_sync:=true \ scan_cloud_topic:=/os_cloud_node/points \ scan_cloud_max_points:=131072  \ approx_sync:=true \ icp_odometry:=true \ args:="-d  \ --Odom/ResetCountdown 0 \ --delete_db_on_start \ --RGBD/CreateOccupancyGrid true \ --Rtabmap/DetectionRate 0.5 \ --Odom/ScanKeyFrameThr 0.8 \ --OdomF2M/ScanMaxSize 10000  \ --OdomF2M/ScanSubtractRadius 0.5  \ --Icp/PM true \ --Icp/VoxelSize 0.05   \ --Icp/MaxTranslation 0   \ --Icp/MaxCorrespondenceDistance 0.5 \ --Icp/PMOutlierRatio 0.85 \ --Icp/Iterations 10 \ --Icp/PointToPlane false \ --Icp/PMMatcherKnn 3 \ --Icp/PMMatcherEpsilon 1 \ --Icp/Epsilon 0.0001 \ --Icp/PointToPlaneK 0 \ --Icp/PointToPlaneRadius 0 \ --Icp/CorrespondenceRatio 0.01"`

matlabbe, 2023-04-02:
This is using ouster lidar `/os_cloud_node/points`. You can also just launch this, then do `rosnode info ...` on the node you want to check which input it is using. Also `rqt_graph` can be useful.

SiinaIng, 2023-04-02:
Okay thanks, I don't know ROS, so this was a bit difficult to interpret! I was just confused since I opened the database made with this configuration (presumably) and the reg/strategy parameter was 0 (visual).

matlabbe, 2023-04-02:
The database would not contain odometry parameters. In this case I would add `--Reg/Strategy 1` to arguments, ti use the lidar for loop closure detection.

SiinaIng, 2023-04-02:
The database did contain all the modified ICP parameters though? It seems I've misunderstood the meaning of that parameter. It thought it meant which sensor data would be used for localization.  So when I test the odometry with different reg/strategy values, what does it mean?

matlabbe, 2023-04-02:
`Reg/Strategy` only means something for rtabmap node. For odometry it is implicilty set to appropriate `Reg/Strategy` depending if you are using icp_odometry (`Reg/Strategy=1`) or stereo_odometry (`Reg/Strategy=0`) or rgbd_odometry (`Reg/Strategy=0`).

ICP parameters are shared between odomerty and rtabmap nodes.

SiinaIng, 2023-04-03:
Okay, I'm getting the gist, maybe. So if you only have a database at your disposal, how can you determine what odometry has been used?

matlabbe, 2023-04-03:
You cannot just by looking at the database, unless it was recorded on standalone version.

On ROS, we would need the launch files to know it.

SiinaIng, 2023-04-03:
Okay. So if I open the database in the standalone app, then change the reg/strategy value and test the odometry, it doesn't mean I switch between visual and LiDAR odometry?

SiinaIng, 2023-04-03:
@matlabbe

matlabbe, 2023-04-16:
> Okay. So if I open the database in the standalone app, then change the reg/strategy value and test the odometry, it doesn't mean I switch between visual and LiDAR odometry?

No. On ROS, you can only change odometry parameters from the launch file you used.