introlab/rtabmap_ros #936: /tag_detection topics

an99990, 2023-04-05:
Hi,
I am publishing poses of my object in the \tag_detections topic but it doesnt seem to be doing anything. How do I know if the tags are processed by rtabmap ? Am I supposed to see them in the UI ? Also are they supposed to be used during mapping or localization ?

[image]

To publish to the \tag_detection i create AprilTagDetection msg. Here is a snippet of the part 

```
tags_msg = AprilTagDetectionArray()
        tags_msg.header.frame_id = "odom"
        #tags_msg.header.stamp = self.get_clock().now().to_msg()

        # loop in the updated centroids
        cpmt = 0
        for pose in self.obj_pose_detect.values():
            # gets pose for each label
            rospy.loginfo(f"pose : {pose}")
            tag = AprilTagDetection()
            tag_pose = PoseWithCovarianceStamped()

            # we have to inverse to camera frame so x->y y->z z->x
            tag_pose.pose.pose.position.x = float(pose[0])
            tag_pose.pose.pose.position.y = float(pose[1])
            tag_pose.pose.pose.position.z = -float(pose[2])
```
            
I am in ros noetic

Thank you

an99990, 2023-04-05:
So with the RGBD/MarkerDetection true flag on I am able to see that it goes through the tag_detections part. However I am getting this error and by debugging i notice that the tags are not the one that I am publishing. The position is zero.And theres an ID so what am I doing wrong.

[ERROR] [1680732638.441682266]: Invalid landmark received! IDs should be > 0 (it is 0). Ignoring this landmark.

matlabbe, 2023-04-16:
Tags callback is here:
https://github.com/introlab/rtabmap_ros/blob/3e140a04e6d4827cb913825c19a06dc8626f790c/rtabmap_slam/src/CoreWrapper.cpp#L2319-L2365

It seems you don't set the timestamp. The `frame_id` of the pose should be relative to the robot or sensor estimating the pose (e.g., camera frame).

Setting `RGBD/MarkerDetection` enables marker detection from inside rtabmap. This doesn't make your tags working, they should work already with it.
```
rtabmap --params | grep Marker
```
If the wrong tag dictionnary is selected, a wrong id may be detected. Also rtabmap doesn't support "0" as id of a marker.

an99990, 2023-04-16:
i was able to debug a bit and i notice that m headeris somehow not good and also the pose receveid is really weird
[image]

```

        tags_msg = AprilTagDetectionArray()
        tags_msg.header.frame_id = "oak-d_frame"
        #tags_msg.header.stamp = self.get_clock().now().to_msg()

        # loop in the updated centroids
        cpmt = 0
        for pose in self.obj_pose_detect.values():
            # gets pose for each label
            rospy.loginfo(f"pose : {pose}")
            tag = AprilTagDetection()
            tag_pose = PoseWithCovarianceStamped()
            tag_pose.header.stamp = rospy.Time.now()
            tag_pose.header.frame_id = "oak-d_frame"

            # we have to inverse to camera frame so x->y y->z z->x
            tag_pose.pose.pose.position.x = float(pose[0])
            tag_pose.pose.pose.position.y = float(pose[1])
            tag_pose.pose.pose.position.z = -float(pose[2])
```

an99990, 2023-04-16:
is there a way to check if the tag are being detected during localization, i have fied my timestamp and it seems like the tag have been added to the graph but my rtabmap_guid doesnt work so i cant see if the tags are there but i added logger in the callaback to confirm that the tags are added.

an99990, 2023-04-16:
also the pose received by rtabmap_ros are weird 
[image]

[image]

matlabbe, 2023-05-06:
There is a topic called `landmarks` published by rtabmap node. You may look at this https://github.com/introlab/rtabmap_ros/blob/master/rtabmap_slam/src/CoreWrapper.cpp#L281