introlab/rtabmap_ros #937: Euroc_dataset.launch

rachelzh9, 2023-04-06:
I'm trying to run the V2_03 and V2_02 Euroc bag files with `euroc_datasets.launch`, but I'm getting this warning:
```
Could not get transform from base_link to cam0 after 0.100000 seconds (for stamp=1413394997.905761)! Error="Lookup would require extrapolation -0.216061209s into the future.  Requested time 1413394997.905760527 but the latest data is at time 1413394997.689699173, when looking up transform from frame [cam0] to frame [base_link]. canTransform returned after 0.101487 timeout was 0.1.".
```
I'm confused because V1_01 works perfectly fine, and when checking tf_monitor, the tf is being published between the frames at 20Hz.

matlabbe, 2023-04-16:
I checked again the V2 bags, and there is an issue with the `static_transform_publisher`s not publishing fast enough stamps of topics.

I [updated](https://github.com/introlab/rtabmap_ros/commit/e0cae222be08786f57e88b7a0b1512fa91aec193) the launch file and now it works with those bags.

cheers,
Mathieu

rachelzh9, 2023-04-16:
Great thanks! that worked for me.

closed 2023-04-16 by rachelzh9

JajamesVid, 2023-05-29:
Hello, following the argumental line of this issue I wanted to ask if the /tf topic must have a specific frequency of publication, Im executing stereo_odometry node and it's working fine for a limited time but then the /odom topic starts to low it's frequency value and this warns appears: 

[ WARN] [1685374109.176753085]: Could not get transform from base_link to camera_link after 0.200000 seconds (for stamp=1685374109.235505)! Error="Lookup would require extrapolation 0.480000000s into the future.  Requested time 1685374109.235505342 but the latest data is at time 1685374108.755505323, when looking up transform from frame [camera_link] to frame [base_link]. canTransform returned after 0.20014 timeout was 0.2.".

Thanks in advance

matlabbe, 2023-05-30:
@JajamesVid Is it with Euroc dataset launch file example?

If not, what is your setup? The `/odom` topic you are referring, is it the output of `stereo_odometry`? How is `base_link` to `camera_link` published? It looks published at least 480 ms in the future.