introlab/rtabmap_ros #943: RTABMap Ros Noetic Installation on NVIDIA Jetson

smartslab, 2023-04-12:
I am running into problems trying to install the rtabmap ros version on my Jetson. The Jetson is running Ubuntu 20.04, Jetpack 5.1. I have ros noetic working fine, and have the standalone rtabmap version working properly, but whenever I try to install the ros version (from source or from the binaries, I get the same issue either way) I end up with the following error when I try to run the rtabmap from ros:

```
ERROR: cannot launch node of type [rtabmap_ros/stereo_odometry]: Cannot locate node of type [stereo_odometry] in package [rtabmap_ros]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [rtabmap_ros/rtabmap]: Cannot locate node of type [rtabmap] in package [rtabmap_ros]. Make sure file exists in package path and permission is set to executable (chmod +x)
```

I have been following the from source instructions for installation [here](https://github.com/introlab/rtabmap_ros#installation) since there is no instructions for Jetsons running 20.04. When building the ros version I also end up with several warnings talking about conflicting opencv versions (4.2 and 4.5), which might be part of the issue causing things to not build the files referenced in the error I get, in all of the different rebuilds I have tried, I have fully removed opencv in order to start clean. The current build I have now also gives a bunch of warnings (many of which are contradictory, as the pair below) which I think might be part of the problem, except that I have no idea how multiple versions got on the system, since I start with a system that does not have any opencv on it before I install ros (the command `sudo find / -name "*opencv*";` gives no results). I think this is probably what is causing the files to not exist, but I don't know how to get rtabmap to not install a second version of opencv, and use the one that comes with ros instead (or if rtabmap does need that specific version, how to remove or update the opencv version that comes with ros without removing ros.) I have tried installing opencv from the source as well, that just resulted in a third version conflicting with the other two. 

```
/usr/bin/ld: warning: libopencv_core.so.4.2, needed by /home/navsea-jetson/sim_ws/devel/lib/librtabmap_conversions.so, may conflict with libopencv_core.so.4.5
/usr/bin/ld: warning: libopencv_stitching.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_stitching.so.4.2
```

Tomorrow I plan to do a complete reflashing of the system to see if maybe something in opencv is sneaking its way back in, but I don't expect that to do anything different, is there any other things I could try that you can think of? I've done searches on both errors, and tried everything I can find to the best of my ability, I just can't seem to make any of them work.

matlabbe, 2023-04-17:
With the ros binaries, did you `source /opt/ros/noetic/setup.bash` before launching?

On jetson, because they often install a different OpenCV version than the one ros is using, either use rtabmap ros binaries or if you need latest from source, I would use docker rtabmap_ros container: https://hub.docker.com/r/introlab3it/rtabmap_ros

Some docker usage examples [here](https://github.com/introlab/rtabmap_ros/tree/master/docker).

cheers,
Mathieu

matlabbe, 2023-04-24:
Based on https://github.com/introlab/rtabmap_ros/tree/master/docker

Launch rtabmap node:
```
mkdir ~/.ros # make sure ~/.ros exists
docker run -it --rm \
   --user $UID \
   -e ROS_HOME=/tmp/.ros \
   --network host \
   -v ~/.ros:/tmp/.ros \
    introlab3it/rtabmap_ros:noetic-latest \
    roslaunch rtabmap_launch rtabmap.launch rtabmap_viz:=false database_path:=/tmp/.ros/rtabmap.db rtabmap_args:="--delete_db_on_start"
```

For visualization, use rviz, or `rtabmap_viz` can be launched like this:
```
XAUTH=/tmp/.docker.xauth
touch $XAUTH
xauth nlist $DISPLAY | sed -e 's/^..../ffff/' | xauth -f $XAUTH nmerge -

docker run -it --rm \
   --privileged \
   -e DISPLAY=$DISPLAY \
   -e QT_X11_NO_MITSHM=1 \
   -e NVIDIA_VISIBLE_DEVICES=all \
   -e NVIDIA_DRIVER_CAPABILITIES=all \
   -e XAUTHORITY=$XAUTH \
   --runtime=nvidia \
   --network host \
   -v $XAUTH:$XAUTH \
   -v /tmp/.X11-unix:/tmp/.X11-unix \
   -e LD_LIBRARY_PATH=/opt/ros/noetic/lib:/opt/ros/noetic/lib/aarch64-linux-gnu:/usr/lib/aarch64-linux-gnu/tegra \
    introlab3it/rtabmap_ros:noetic-latest \
    /bin/bash -c "export ROS_NAMESPACE=rtabmap && rosrun rtabmap_viz rtabmap_viz"
```

Avi241, 2023-06-23:
how can i do if I want to use realsense D435i here ?

Thank you

matlabbe, 2023-06-24:
You may install realsense ros package.

branden1992, 2023-11-24:
Can someone build me a location and data recovery