introlab/rtabmap_ros #945: Jittery position when stationary

peredwardsson, 2023-04-13:
The `map -> odom` transform looks a little nervous when stationary. The environment is a little pale so I understand the positioning not being exact in this case, but here, you can clearly see from the odometry that we are not moving. Is it possible to factor this into the `map -> odom` transform corrections? It jumps around 40 cm in one direction only, when the input (in my case, two rgbd cameras) look at the same targets. I think the depth camera might be shifting, which could cause the corrections to trigger, but it is absolutely stationary, just like the odometry is suggesting.
[image]

matlabbe, 2023-04-17:
What is the covariance of your odometry? If odometry covariance is small, it won't jump as high than this.

peredwardsson, 2023-04-18:
The odometry is diverging, it seems. 
[image]

This is plotting all the covariances of the output of the robot_localization `ekf_node`. You can see one or some diverging, this is the variance of x and y. I understand that this is a different package, but if you have any suggestion on how to proceed i would appreciate it a lot.

peredwardsson, 2023-04-19:
I added a pose input to the EKF (from rgbd odom) which reduces the jittering significantly. Thanks!

matlabbe, 2023-04-20:
Variance of 80-100 is huge! This means 100x100 meters error. Normally it should be way lower than 1 (like 0.0005 and below).

Glad you found a solution,
Mathieu

peredwardsson, 2023-05-04:
The covariances are now below 0.1 for most of the time, but the `map -> odom` transform can still be jittery.  Our use case is to follow a very strict path, and so small jumps in the localization produces very erratic behavior. How can we troubleshoot this?

matlabbe, 2023-05-15:
If you know that your odometry is reliable for some time, you may transform your path in odom frame, then follow the path using odom instead. Maybe update the path in odom frame using map->odom just after some time. map->odom will always jitters, see [REP 105](https://www.ros.org/reps/rep-0105.html).

closed 2025-03-05 by peredwardsson