introlab/rtabmap_ros #949: 0.21.1 Noetic update

wakoko79, 2023-04-19:
Just noticed that noetic branch just updated to 0.21.1, problem is all of my scripts now stopped working, and the documentation in [wiki.ros.org/rtabmap_ros](http://wiki.ros.org/rtabmap_ros) is incomplete for noetic. 

New structure tutorial will help lots.
Is there any new documentation out there that can help? Thanks!

matlabbe, 2023-04-19:
Working on it! Just added this migration guide: http://wiki.ros.org/rtabmap_ros/noetic_and_newer#Migration_Guide_New_Interface_Noetic.2FROS2

I'll migrate the documentation

antoniskara, 2023-04-21:
I faced the same issue. Using rtabmap noetic, after a new build ,all my scripts stopped working. Seems like there are missing files from /ros/opt/noetic/share/rtabmap_ros. More specifically the files missing are : rviz_plugins.xml , nodelet_plugins.xml costmap_plugins.xml msg srv

matlabbe, 2023-04-24:
See the [link](http://wiki.ros.org/rtabmap_ros#rtabmap_ros.2Fnoetic_and_newer.Migration_Guide_New_Interface_Noetic.2FROS2) of my previous post. These files have been moved in new sub packages, changing `rtabmap_ros` prefix to new package name of the nodelet/costmap plugin.

closed 2023-04-24 by wakoko79