introlab/rtabmap_ros #951: Weird errors when running; (Couldn't find executable named rtabmap below /opt/ros/noetic/share/rtabmap_ros)

CJdev99, 2023-04-19:
When I run rtabmap_ros i get really weird errors, first of all, it's shown as a ros package:
```
rospack find rtabmap_ros
/opt/ros/noetic/share/rtabmap_ros
```
and when i try to run directly:
```
rosrun rtabmap_ros rtabmap
[rosrun] Couldn't find executable named rtabmap below /opt/ros/noetic/share/rtabmap_ros
```

When I run the nodelets from a launch file, I get the following errors (there's a lot). Any ideas what these mean and how to fix? I tried to uninstall and install rtabmap_ros from the ros distro.
```

[ERROR] [1681936079.384683493]: Failed to load nodelet [/rtabmap/rgbd_sync] of type [rtabmap_ros/rgbd_sync] even after refreshing the cache: According to the loaded plugin descriptions the class rtabmap_ros/rgbd_sync with base class type nodelet::Nodelet does not exist. Declared types are  RobotLocalization/EkfNodelet RobotLocalization/NavSatTransformNodelet RobotLocalization/UkfNodelet SlamGMappingNodelet apriltag_ros/ContinuousDetector depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/point_cloud_xyzrgb_radial depth_image_proc/register depthimage_to_laserscan/DepthImageToLaserScanNodelet image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image imu_filter_madgwick/ImuFilterNodelet laser_filters/scan_to_cloud_filter_chain nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid pointcloud_to_laserscan/laserscan_to_pointcloud_nodelet pointcloud_to_laserscan/pointcloud_to_laserscan_nodelet realsense2_camera/RealSenseNodeFactory rtabmap_legacy/data_odom_sync rtabmap_legacy/data_throttle rtabmap_legacy/obstacles_detection_old rtabmap_legacy/stereo_throttle rtabmap_legacy/undistort_depth rtabmap_odom/icp_odometry rtabmap_odom/rgbd_odometry rtabmap_odom/rgbdicp_odometry rtabmap_odom/stereo_odometry rtabmap_slam/rtabmap rtabmap_sync/rgb_sync rtabmap_sync/rgbd_sync rtabmap_sync/rgbdx_sync rtabmap_sync/rtabmap rtabmap_sync/stereo_sync rtabmap_util/disparity_to_depth rtabmap_util/imu_to_tf rtabmap_util/lidar_deskewing rtabmap_util/obstacles_detection rtabmap_util/point_cloud_aggregator rtabmap_util/point_cloud_assembler rtabmap_util/point_cloud_xyz rtabmap_util/point_cloud_xyzrgb rtabmap_util/pointcloud_to_depthimage rtabmap_util/rgb_sync rtabmap_util/rgbd_relay rtabmap_util/rgbd_split rtabmap_util/stereo_sync stereo_image_proc/disparity stereo_image_proc/point_cloud2
[ERROR] [1681936079.384751839]: The error before refreshing the cache was: According to the loaded plugin descriptions the class rtabmap_ros/rgbd_sync with base class type nodelet::Nodelet does not exist. Declared types are  RobotLocalization/EkfNodelet RobotLocalization/NavSatTransformNodelet RobotLocalization/UkfNodelet SlamGMappingNodelet apriltag_ros/ContinuousDetector depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/point_cloud_xyzrgb_radial depth_image_proc/register depthimage_to_laserscan/DepthImageToLaserScanNodelet image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image imu_filter_madgwick/ImuFilterNodelet laser_filters/scan_to_cloud_filter_chain nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid pointcloud_to_laserscan/laserscan_to_pointcloud_nodelet pointcloud_to_laserscan/pointcloud_to_laserscan_nodelet realsense2_camera/RealSenseNodeFactory rtabmap_legacy/data_odom_sync rtabmap_legacy/data_throttle rtabmap_legacy/obstacles_detection_old rtabmap_legacy/stereo_throttle rtabmap_legacy/undistort_depth rtabmap_odom/icp_odometry rtabmap_odom/rgbd_odometry rtabmap_odom/rgbdicp_odometry rtabmap_odom/stereo_odometry rtabmap_slam/rtabmap rtabmap_sync/rgb_sync rtabmap_sync/rgbd_sync rtabmap_sync/rgbdx_sync rtabmap_sync/rtabmap rtabmap_sync/stereo_sync rtabmap_util/disparity_to_depth rtabmap_util/imu_to_tf rtabmap_util/lidar_deskewing rtabmap_util/obstacles_detection rtabmap_util/point_cloud_aggregator rtabmap_util/point_cloud_assembler rtabmap_util/point_cloud_xyz rtabmap_util/point_cloud_xyzrgb rtabmap_util/pointcloud_to_depthimage rtabmap_util/rgb_sync rtabmap_util/rgbd_relay rtabmap_util/rgbd_split rtabmap_util/stereo_sync stereo_image_proc/disparity stereo_image_proc/point_cloud2
[FATAL] [1681936079.384904865]: Failed to load nodelet '/rtabmap/rgbd_sync` of type `rtabmap_ros/rgbd_sync` to manager `/camera/realsense2_camera_manager'
```

CJdev99, 2023-04-19:
Nevermind, I just saw that the package was restructured, so when I upgraded, the nodelet names weren't recognized

closed 2023-04-19 by CJdev99

matlabbe, 2023-04-20:
For others having similar issues, see [migration guide here](http://wiki.ros.org/rtabmap_ros#rtabmap_ros.2Fnoetic_and_newer.Migration_Guide_New_Interface_Noetic.2FROS2).