introlab/rtabmap_ros #957: Nonfree opencv algorithms available in rtabmap but not in rtabmap_ros

emilymes, 2023-04-24:
Hi, I am trying to use the nonfree opencv algorithms in combination with rtabmap_ros, to compare the performance of the various visual features.
I have built opencv from source with nonfree algorithms enabled and was able to build rtabmap from source as well, which works as expected. 

`rtabmap --version`
gives
`RTAB-Map:               0.21.1
PCL:                    1.10.0
With VTK:                7.1.1
OpenCV:              4.7.0-dev
With OpenCV xfeatures2d:  true
With OpenCV nonfree:      true
With ORB OcTree:          true
With SuperPoint Torch:    true
With Python3:             true
With FastCV:             false
With OpenGV:             false
With Madgwick:            true
With PDAL:               false
With TORO:                true
With g2o:                 true
With GTSAM:               true
With Vertigo:             true
With CVSBA:              false
With Ceres:              false
With OpenNI2:             true
With Freenect:            true
With Freenect2:          false
With K4W2:               false
With K4A:                false
With DC1394:              true
With FlyCapture2:        false
With ZED:                 true
With ZED Open Capture:   false
With RealSense:          false
With RealSense SLAM:     false
With RealSense2:         false
With MYNT EYE S:         false
With DepthAI:            false
With libpointmatcher:     true
With CCCoreLib:          false
With octomap:             true
With cpu-tsdf:           false
With open chisel:        false
With Alice Vision:       false
With LOAM:               false
With FLOAM:              false
With FOVIS:              false
With Viso2:              false
With DVO:                false
With ORB_SLAM:           false
With OKVIS:              false
With MSCKF_VIO:          false
With VINS-Fusion:        false
With OpenVINS:           false`

and when I use the rtabmap standalone I can see in Window > Preferences > Visual Registration > Visual Feature that all the feature detectors are available.

However, when I `roslaunch rtabmap_ros rtabmap.launch args:="--Vis/FeatureType SURF"` for example, I get the following warning: `[ WARN] (2023-04-24 16:32:08.975) Features2d.cpp:529::create() SURF and SIFT features cannot be used because OpenCV was not built with xfeatures2d module. GFTT/ORB is used instead.`
Also in Window > Preferences > Visual Registration the nonfree feature detectors are not available.

Would you happen to know what goes wrong? Thank you in advance!

matlabbe, 2023-05-06:
Hi, I think you meant `--Vis/FeatureType 0`, but you should build rtabmap_ros with your rtabmap built from source. To make sure the right one is detected on catkin make, remove the ros binaries:
```
sudo apt remove ros-noetic-rtabmap*
```

cheers,
Mathieu

emilymes, 2023-05-23:
Yes, thank you! The binaries were not installed, but in the end a complete rebuild worked. Now I am able to roslaunch rtabmap_ros using all the feature types! Except for superpoint..

`roslaunch rtabmap_ros rtabmap.launch args:="--delete_db_on_start --Vis/FeatureType 11 --SuperPoint/ModelPath ~/catkin_ws/superpoint.pt"` gives the following errors:
- [rtabmap/rgbd_odometry-1] process has died [pid 1303898, exit code -11, cmd /home/emily/catkin_ws/devel/lib/rtabmap_odom/rgbd_odometry
- [rtabmap/rtabmap-2] process has died [pid 1303899, exit code -11, cmd /home/emily/catkin_ws/devel/lib/rtabmap_slam/rtabmap

Following [https://github.com/introlab/rtabmap_ros/issues/28](the debugging instructions) this is the gdb backtrace for both died processes:
[image]

I suspect it maybe goes wrong in the rtabmap_ros/rtabmap_odom/src/RGBDOdometryNode.cpp and rtabmap_ros/rtabmap_slam/src/CoreNode.cpp, and then specifically when loading the nodelet `nodelet.load(nodelet_name, "rtabmap_slam/rtabmap", remap, nargv);`
Would you have an idea how it can be fixed to get SuperPoint working? Thank you in advance.

emilymes, 2023-05-24:
Hi Mathieu,
I have spent some time debugging and printing statements to see where it goes wrong, and this is what I found:
- **process rtabmap/rtabmap** (in rtabmap_slam/src/CoreNode.cpp) goes into the `nodelet.load()` line, but does not exit it. Once in `nodelet.load()` it goes into the `OnInit` in src/CoreWrapper.cpp, the `LoadParameters`, the `SaveParameters`, and then it goes in the `~CoreWrapper` function, which is also exits without errors. By then I can not determine which function it enters. I thought that once the `~CoreWrapper` function exits, it will go back to the `ros::spin()` line in CoreNode.cpp but this is not the case.
- **process rtabmap/rgbd_odometry** (in rtabmap_odom/src/RGBDOdometryNode.cpp). It seemed to be the same problem because it also enters but does not exit the line `nodelet.load()`, but this time it goes into the `OnInit` of the OdometryROS.cpp, updates the parameters, and is then stuck on `odometry_ = Odometry::create(parameters_);` But I don't know where exactly in that line it goes wrong.

Do you have any idea what the problem could be? Thanks in advance.

emilymes, 2023-05-26:
Hi Mathieu,
I have continued with my debugging and printing, and have now narrowed it down to one line, namely:
`torch::load(model_, uReplaceChar(path, '~', UDirectory::homeDir()));` in the `SPDetector::SPDetector()` function in the `SuperPoint.cc` file. The path should be correct and I have tried both `superpoint.pt` and `superpoint_v1.pth` files.
I don't get an error message on that line, but my print statement after that line does not print, and instead the destructor `Odometry::~OdometryROS()` function is executed. This suggests that an error occurred during the loading process.

Would you have any idea what causes this issue? Have you seen this problem?

I am on Ubuntu 20.04 and use `libtorch-cxx11-abi-shared-with-deps-2.0.1+cpu` (but get the same issue when using with libtorch2.0.1+cuda11.7 on another PC with a GPU)

`rtabmap --version` gives:
```
RTAB-Map:               0.21.1
PCL:                    1.10.0
With VTK:                7.1.1
OpenCV:                  4.2.0
With OpenCV xfeatures2d:  true
With OpenCV nonfree:      true
With ORB OcTree:          true
With SuperPoint Torch:    true
With Python3:             true
```

and `cmake ..` in `~/rtabmap/build` gives:
```
-- MOBILE_BUILD=OFF
-- Eigen found (include: /usr/include/eigen3, version: 3.3.7)
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- The imported target "vtkParseOGLExt" references the file
   "/usr/bin/vtkParseOGLExt-7.1"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- The imported target "pvtk" references the file
   "/usr/bin/pvtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-7.1/VTKTargets.cmake"
but not all the files it references.

-- QHULL found (include: /usr/include, lib: optimized;/usr/lib/x86_64-linux-gnu/libqhull.so;debug;/usr/lib/x86_64-linux-gnu/libqhull.so)
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so
-- Found OpenMP: /usr/lib/gcc/x86_64-linux-gnu/10/libgomp.so;/usr/lib/x86_64-linux-gnu/libpthread.so
-- Add PCL_OMP to definitions
-- Found OpenCV: /usr/local/include/opencv4
[... 79 lines cut ...]
--   With ZEDOC                = NO (ZED Open Capture not found)
--   With RealSense            = NO (librealsense not found)
--   With RealSense2           = NO (librealsense2 not found)
--   With MyntEyeS             = NO (mynteye s sdk not found)
--   With DepthAI              = NO (WITH_DEPTHAI=OFF)
-- 
--  Odometry Approaches:
--   With loam_velodyne        = NO (WITH_LOAM=OFF)
--   With floam                = NO (WITH_FLOAM=OFF)
--   With libfovis             = NO (WITH_FOVIS=OFF)
--   With libviso2             = NO (WITH_VISO2=OFF)
--   With dvo_core             = NO (WITH_DVO=OFF)
--   With okvis                = NO (WITH_OKVIS=OFF)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With VINS-Fusion          = NO (WITH_VINS=OFF)
--   With OpenVINS             = NO (WITH_OPENVINS=OFF)
--   With ORB_SLAM             = NO (WITH_ORB_SLAM=OFF)
-- Show all options with: cmake -LA | grep WITH_
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/emily/rtabmap/build
```
 
Thank you in advance!

matlabbe, 2023-05-30:
Where did you download `superpoint.pt`? I had to update the file in that [release](https://github.com/introlab/rtabmap/releases/tag/0.21.0) (see Assets) to make it work with recent libtorch.

To regenerate the model with your libtorch version, you may follow last bullet point of step 1 [here](https://github.com/introlab/rtabmap/tree/master/archive/2022-IlluminationInvariant#how-reproduce-results-shown-in-the-paper).