introlab/rtabmap_ros #961: Colcon Build - Foxy - rosidl_typesupport

UnoMeDD, 2023-05-02:
Hello everyone, 

I am currently running into a problem when trying to build rtabmap. 
Has someone encountered the same issue or can help me solving it? 

[image]

UnoMeDD, 2023-05-02:
I have also tried running the build in sequential mode. 
But I still got some undefined references. 

[image]

matlabbe, 2023-05-12:
Humm, there are no errors on CI https://github.com/introlab/rtabmap_ros/actions/runs/4954721903/jobs/8863479017

I think you may have ZED sdk installed (that should be a problem though). You may try building rtabmap without it:
```
colcon build --cmake-args -DWITH_ZED=OFF ...
```