introlab/rtabmap_ros #966: problem with insatallation of rtabmap_ros

azizOthmani, 2023-05-09:
Hello everyone, when i try to install rtabmp following readme.md of https://github.com/introlab/rtabmap_ros/tree/noetic-devel for the last step when i try catkin_make -j4 command this error appears 
```
[ 82%] Linking CXX executable /home/aziz/catkin_ws/devel/lib/rtabmap_demos/wifi_signal_sub
[ 82%] Linking CXX executable /home/aziz/catkin_ws/devel/lib/rtabmap_demos/save_objects_example
[ 82%] Linking CXX executable /home/aziz/catkin_ws/devel/lib/rtabmap_examples/external_loop_detection_example
/usr/bin/ld: /opt/ros/noetic/lib/x86_64-linux-gnu/librtabmap_core.so.0.21.1: undefined reference to `octomath::Pose6D::Pose6D(octomath::Pose6D const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [rtabmap_ros/rtabmap_demos/CMakeFiles/rtabmap_wifi_signal_pub.dir/build.make:352: /home/aziz/catkin_ws/devel/lib/rtabmap_demos/wifi_signal_pub] Error 1
make[1]: *** [CMakeFiles/Makefile2:8570: rtabmap_ros/rtabmap_demos/CMakeFiles/rtabmap_wifi_signal_pub.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: /opt/ros/noetic/lib/x86_64-linux-gnu/librtabmap_core.so.0.21.1: undefined reference to `octomath::Pose6D::Pose6D(octomath::Pose6D const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [rtabmap_ros/rtabmap_demos/CMakeFiles/rtabmap_wifi_signal_sub.dir/build.make:352: /home/aziz/catkin_ws/devel/lib/rtabmap_demos/wifi_signal_sub] Error 1
make[1]: *** [CMakeFiles/Makefile2:8624: rtabmap_ros/rtabmap_demos/CMakeFiles/rtabmap_wifi_signal_sub.dir/all] Error 2
/usr/bin/ld: /opt/ros/noetic/lib/x86_64-linux-gnu/librtabmap_core.so.0.21.1: undefined reference to `octomath::Pose6D::Pose6D(octomath::Pose6D const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [rtabmap_ros/rtabmap_examples/CMakeFiles/rtabmap_external_loop_detection_example.dir/build.make:352: /home/aziz/catkin_ws/devel/lib/rtabmap_examples/external_loop_detection_example] Error 1
make[1]: *** [CMakeFiles/Makefile2:8678: rtabmap_ros/rtabmap_examples/CMakeFiles/rtabmap_external_loop_detection_example.dir/all] Error 2
/usr/bin/ld: /opt/ros/noetic/lib/x86_64-linux-gnu/librtabmap_core.so.0.21.1: undefined reference to `octomath::Pose6D::Pose6D(octomath::Pose6D const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [rtabmap_ros/rtabmap_demos/CMakeFiles/rtabmap_save_objects_example.dir/build.make:559: /home/aziz/catkin_ws/devel/lib/rtabmap_demos/save_objects_example] Error 1
make[1]: *** [CMakeFiles/Makefile2:8651: rtabmap_ros/rtabmap_demos/CMakeFiles/rtabmap_save_objects_example.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j4" failed

```

matlabbe, 2023-05-15:
I would try to rebuild rtabmap from source when you build rtabmap_ros from source. Uninstall
```
sudo apt remove ros-noetic-rtabmap
```
then follow step 2 from https://github.com/introlab/rtabmap_ros#build-from-source

chukhoi61, 2024-01-12:
I did as you said but still got the error

matlabbe, 2024-01-13:
See that comment if it could help https://github.com/introlab/rtabmap_ros/issues/737#issuecomment-1407157456